esp8266/Arduino #3897: Can't connect to web server.

czucker, 2017-11-30:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

devyte, 2017-11-30:
@czucker no basic info, no IDE settings, only a basic referral to a sketch. You pretty much ignored the entire issue template, which exists for good reasons.
Please provide more details about the hardware, about the build settings, and about the specific sketch to try.
Try erasing the entire flash first before flashing the sketch.

devyte, 2018-02-07:
Closing due to lack of feedback.

closed 2018-02-07 by devyte