esp8266/Arduino #3899: [OSX] ESP12- E Flashing with esptool doesn't work - FT232RL

Max123459, 2017-11-30:
#### Hardware
Hardware:		ESP-12E; OSX High Sierra; FT232RL USB Connector

### Description

Hi, 

I hope you can understand my English ;-).

I have problems flashing my ESP12E with the ArduinoIDE. To rule out problems I've tried flashing it with Windows, this works. 

But flashing under OSX doesn't work, the drivers are installed and the device is recognized (flashing the esp software with the terminal works!).

I get following error:
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.usbserial-A50285BI
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/cu.usbserial-A50285BI at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
Uploading 226080 bytes from /var/folders/zj/9t2dfhl17f7cs0__nj42w8jh0000gp/T/arduino_build_902901/sketch_nov30a.ino.bin to flash at 0x00000000
	erasing flash
	size: 037320 address: 000000
	first_sector_index: 0
	total_sector_count: 56
	head_sector_count: 16
	adjusted_sector_count: 40
	erase_size: 028000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting timeout 15000
	setting timeout 100
	espcomm_send_command: receiving 2 bytes of data
	writing flash
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04
closing bootloader
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

I hope there is a fix for that. Please answer CLEARLY that i understand what to do ;-)

Thank you very much!
### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512K/64k, other values also tested
CPU Frequency:	80Mhz
Flash Mode:            DIO, QIO also tested!
Flash Frequency:	40Mhz
Upload Using:	Serial
Reset Method:	ck / nodemcu, both tested

### Sketch (empty, that no problems are caused by the sketch)

void setup() {

}

void loop() {

}

Max123459, 2017-12-01:
No one a idea?

Max123459, 2017-12-01:
Problem solved

closed 2017-12-02 by devyte

sabag, 2018-01-12:
how it was solved, I have the same problem after upgrading to high sierra

Max123459, 2018-01-12:
You have to Update the esptool

sabag, 2018-01-14:
i already have the most updated Arduino IDE (1.8.5) installed, still the problem exists.
it was working fine for me, un until two days ago when I upgraded to High Sierra.

Max123459, 2018-01-14:
Have you updated the esptool?

sabag, 2018-01-14:
I have esptool 0.4.9. 
how do I update it ?

Max123459, 2018-01-14:
Updating to esptool 0.4.12 Maybe solve the Problem...