esp8266/Arduino #3900: verifyCertChain with ESP8266httpUpdate using https?

whyameye, 2017-11-30:
### Basic Infos

#### Hardware
Hardware:			ESP-12F
Core Version:      	github commit 7b09ae5f69f8dccb5152d128c8484ec9a18adb11 Nov 22 2017

### Description
ESP8266httpUpdate supports https, which is great, but it appears I have to hardcode the fingerprint of the certificate and that there is no way to verify the certificate chain. Hardcoding the fingerprint makes it hard to update the certificate and the lack of verification of the certificate chain seems like a security risk. Are the workarounds for this? Is there a way to use verifyCertChain with ESP8266httpUpdate? Am I missing something obvious?

One thought I had was to load the certificate with client.setCACert_P then verify it, then get it's fingerprint (how?) then use that fingerprint with ESP8266httpUpdate. Is this a good thought? What's a better thought?

bapidebabrata, 2017-12-01:
Currently there is no ESP8266httpUpdate, update  method overload to allow WiFiClientSecure. But it was available once!!!  Ref: https://github.com/esp8266/Arduino/commit/a2789df2a5294c3205230f5a76361f68debd19a1#diff-9ad4a3ba4f6c8f8a13325db8e248d952

@igrr Is there any option/alternative in the latest git version to pass WiFiClientSecure object to the update method after certification verification.
I also checked your post https://gist.github.com/igrr/24dd2138e9c8a7daa1b4 .

earlephilhower, 2018-11-06:
This looks like it was fixed via #4980 where you can create a `BearSSL::WiFiClientSecure` with any kind of validation options needed and pass it into `ESP8266httpUpdate()`.

closed 2018-11-06 by earlephilhower