esp8266/Arduino #3903: Linker error with GDBStub

chessweb01, 2017-12-02:
I added #include <GDBStub.h> to my sketch for NodeMCU v1.0 and got the following linker error:

.pioenvs\nodemcuv2\lib\GDBStub\internal\gdbstub.o: In function `gdbReadCommand$part$3':
gdbstub.c:(.iram.text+0x6f9): dangerous relocation: call0: misaligned call target: gdbstub_set_hw_breakpoint
gdbstub.c:(.iram.text+0x796): dangerous relocation: call0: misaligned call target: gdbstub_del_hw_watchpoint

igrr, 2017-12-27:
@chessweb01 I have tested a few of the examples (Blink, FSBrowser, SPISlave_Master) and wasn't able to reproduce this issue. Please attach a sketch which demonstrates the issue.

freeck, 2018-01-25:
I remember that using buildflag 'long calls' introduced this linker error. That is caused by the fact that file gdbstub_init.S  is word or byte-alligned. So I inserted 'align 4' before the functions. Hope this works.