esp8266/Arduino #3904: ESP (NODE MCU) responding to devices only connected in the same network??

shekhars271991, 2017-12-03:
#include<ESP8266WiFi.h>
#include<ESP8266WebServer.h>

ESP8266WebServer server;
char* ssid="skr";
char* pass="12345678";
uint8_t pin_led =16;

void setup() {
  // put your setup code here, to run once:
  pinMode(pin_led,OUTPUT);
  WiFi.begin(ssid,pass);
  Serial.begin(115200);
  while(WiFi.status()!=WL_CONNECTED)
  {
    Serial.print(".");
    delay(500);
  }
  Serial.println("");
  Serial.print("IP Address: ");
  Serial.print(WiFi.localIP());
//webserver code
  server.on("/",[](){server.send(200,"text/plain","Hello World!");});
  server.on("/toggle",toggleLED);
  server.begin();
}

void loop() {
  // put your main code here, to run repeatedly:

  server.handleClient();

}
void toggleLED()
{

  digitalWrite(pin_led,!digitalRead(pin_led));
  server.send(204,"");
 }

RudyFiero, 2017-12-03:
@shekhars271991 
Can you please add some information as to what you are asking? Your heading is insufficient and it is much more respectful  to provide more relevant words so that a reader does not have to try and guess what your problem is.

5chufti, 2017-12-03:
most likely he is in a private WiFi and experiences the fact he can't reach the esp from the internet ...

RudyFiero, 2017-12-03:
Yes most likely. But that is a guess. And nothing about the conditions of his network or the external network are provided. On top of this there is nothing here to indicate there is an issue with the Arduino environment. Just on the face of it it looks like "I don't know what i am doing and I want you guys to figure out this for me." And this is not a support forum. This is not a support forum. This is for dealing with code issues or improvements. Questions regarding lack of understanding should be directed to places like http://www.esp8266.com/ but even there, because you are asking for free help, make it easier on people to understand what the problem is.

devyte, 2017-12-05:
@shekhars271991 no basic info, no basic settings, description is too vague.
Closing due to lack of info.

closed 2017-12-05 by devyte

shekhars271991, 2017-12-05:
hello everyone, Sorry about not providing detailed information about the issue. 

the problem you guessed is right, i have not set port forwarding in my router and since I have been using my NODE MCU from an external network this was not working.

After port forwarding the issue is resolved.

But, the thing is i was to build a IoT device (a product of sorts) which can be just plugged and played i.e. i wanted to achieve this without setting up port forwarding specifically in the router of every user who uses this device. Im really not good in networking, wanted to know if it was possible to achieve this by any code in the NODE MCU, (by making it public somehow).

Thanks a lot for your help.

RudyFiero, 2017-12-05:
Routers prevent exposure from Public networks. It is easy to get out of a private network but it should be hard to get in. And that is what the router does. If you want a publicly accessible connection into a private network it must be enabled from the inside, to configure the router to allow access. 

The other option is to have a publicly available server on the net and the device inside the private network call out to it. This is how many MQTT based systems work. A user interacts with the Broker and not directly on the device.