esp8266/Arduino #3905: sntp broken in git commit

whyameye, 2017-12-03:
### Basic Infos

#### Hardware
Hardware:			ESP-12F
Core Version:      	git commit 7b09ae5f69f8dccb5152d128c8484ec9a18adb11

### Description
in example sketch `HTTPSRequestCACert` output from git commit 7b09ae5f69f8dccb5152d128c8484ec9a18adb11 pulled on Nov 22, 2017
```
Setting time using SNTP
Current time: Thu Jan  1 08:00:00 1970
```

same sketch, output from 2.4.0-rc2:
```
Setting time using SNTP.
Current time: Mon Dec  4 03:38:13 2017
```

5chufti, 2017-12-03:
try moving the sntp initialisation part [configTime()] before the WiFi initialisation, this should help.
new sntp implementation gets synced with the first dhcpd offer, then every 60min.
Or force a sync as explained here: #1679
So it is more likely a not adapted example than a broken sntp.

d-a-v, 2017-12-03:
In this sketch, the ```while (now < 1000)``` should be ```while (now < 30000)``` since ```now``` is initialized to 28800=8*3600 (0 with lwip1.4's sntp implementation), then jumps to far away in 1970's future once ntp timestamp is received.
That way it works with both version of sntp.

devyte, 2017-12-04:
Can someone please make a PR with the needed changes to the HTTPSRequestCACert.ino example?

d-a-v, 2017-12-04:
I have it [part](https://github.com/d-a-v/Arduino/commit/e53cece0f5bec9c4ff1e034b4988da5bae836c97) of a future PR if it is not urgent. Anyone feel free to fix it if this can't wait.

devyte, 2017-12-05:
@whyameye does @d-a-v 's code change solve your issue?

whyameye, 2017-12-05:
yes! Thank you. :-)

devyte, 2017-12-05:
@d-a-v if you make a PR with that, I'll merge it.

d-a-v, 2017-12-05:
What this sketch is doing is waiting for time to be set by NTP server before doing the good stuff.
What do you think of either:
* a user callback from ```settimeodday()``` (upon ntp timestamp received) as proposed in the #1679 ntp thread ([here](https://github.com/esp8266/Arduino/issues/1679#issuecomment-346848602) (1.b.) and [there](https://github.com/esp8266/Arduino/issues/1679#issuecomment-346935643)) 
* or a function that would return true once, once ntp is set, then false until next time,
* or a variable set to true on sync (that the user would have to reset once read)
* or nothing to not break the API :]

@devyte @igrr

We could also have an always working ```settimeofday()``` if we can patch lwip1.4 (there is a ```static``` to remove) but I don't know the pain level of this in the core integration process.

d-a-v, 2017-12-20:
@devyte #4000 is the simple fix

PR #4000 (2017-12-21): in example properly wait for NTP to be set
fix #3905
commit d5bb4a99 (2017-12-21): in example properly wait for NTP to be set (#4000)

fix #3905
closed 2017-12-21 by devyte