esp8266/Arduino #3909: fatal error: Arduino.h: No such file or directory

takiuddinahmed, 2017-12-04:
Hello, i am facing a problem. when I compile any esp8266 code by selecting 'Generic ESP8266' board manager from my arduino ide the error which are happening are as follows:

> C:\Users\ASUS\AppData\Local\Temp\arduino_build_855432\sketch\esp1st.ino.cpp:1:21: fatal error: Arduino.h: No such file or directory
>
>
>
 > #include <Arduino.h>
  >                   ^
>
> compilation terminated.
>
> exit status 1
> Error compiling for board Generic ESP8266 Module.

devyte, 2017-12-05:
@takiuddinahmed I suggest a clean install, both of the IDE and the core. Please read the instructions carefully. If you still need help, please seek assistance at a community forum first to figure out what you are doing wrong.
Closing as can't reproduce.

closed 2017-12-05 by devyte

takiuddinahmed, 2017-12-06:
It works. Thanks

mikelau1, 2018-02-11:
hi can you please let know how did you resolve it? thanks!