esp8266/Arduino #3911: "Rebuilding all" when no build options changed

whyameye, 2017-12-04:
### Basic Infos

#### Hardware
Hardware:			ESP-12F
Core Version:      	master Nov 2017
Arduino IDE 1.8.5 (OSX High Sierra)

### Description
I often get the message "Build options changed. Rebuilding all" when literally not a single line of code has changed. And, based on the speed of compile, it is definitely rebuilding all, as promised.

This is less than ideal for workflow. :-)

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB SPIFFS
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	    40Mhz
Upload Using:		SERIAL
Reset Method:		ck

devyte, 2017-12-05:
I believe this is an Arduino IDE thing. If you so much as change a space in the .ino, or change the file timestamp somehow, even without any changes, it rebuilds everything.
Closing as OT.

whyameye, 2017-12-05:
It could be an Arduino IDE thing, but it isn't me changing a space or a timestamp. I can click "Verify" then click on it immediately again as soon as it is finished and it happens. I had wondered if it could be related to `boards.txt` as I grabbed that file from #3722 and the problem showed up soon after that. I probably should have mentioned that in the OP. I've used the Arduino environment a fair amount and this is new behavior.

d-a-v, 2017-12-05:
@whyameye here is the link for the previous [boards.txt](https://raw.githubusercontent.com/esp8266/Arduino/0c897c37a6eab3eab34147219617945a32a9b155/boards.txt) if you want to try it and check if your problem appeared with the new version.
I also experienced this but not always, and by reading this [link](https://forum.pjrc.com/threads/34233-Build-options-changed-rebuilding-all) we may have the beginning of an explanation (I sometimes compile modified examples without saving them).

d-a-v, 2018-01-20:
@whyameye you are right. Same behaviour here. Using the old boards.txt make this problem disappear.

d-a-v, 2018-02-13:
@whyameye yes, it does, thanks!
Unfortunately, this beta version currently breaks our OOM debug checker...

devyte, 2018-02-15:
@d-a-v any idea why it breaks?

d-a-v, 2018-02-15:
The new architecture does not cope well with gcc cmdline ```-include``` option. This is fine now with #4357.
arduino-1.9 works far faster here than 1.8 when it comes to decide if something has to be rebuilt. I can't say if it does it right though :)

d-a-v, 2018-03-08:
@whyameye  if arduino-1.9-beta is OK for you, can we close here ?

closed 2018-03-08 by whyameye

marcmerlin, 2018-05-01:
If I may, yes I've confirmed that 1.9 beta fixes it, but it seems that 1.8 broke it.
My build times were getting out of hand with caching broken. Caching is not a new feature, it would be nice if 1.8 actually fixed whatever it did to break it.
Going to an unproven 1.9 beta as the only way to restore proper caching (or I suppose downgrading arduino far enough) isn't the best solution IMO.

marcmerlin, 2018-08-14:
@d-a-v I just lost 2 days of debugging a non problem due to bugs in 1.9 that I was forced to move to because of this bug in 1.8.5 that takes so long to rebuild everything every time that I was pushed towards 1.9.
It's not ok to close this bug when there is clearly no fix in any stable release.
See also https://github.com/arduino/arduino-builder/issues/251

d-a-v, 2018-08-14:
Sorry to hear that
#5043

d-a-v, 2018-08-14:
also, arduino nightly (pre-1.8.6) also seems to fix this.

karelv, 2018-08-14:
@d-a-v Where do I find this nightly (pre-1.8.6) release? I would like to try (also suffering from slow debug iterations...)

d-a-v, 2018-08-14:
http://arduino.cc > download > nightlies

karelv, 2018-08-14:
ok, thanks, I see, it is now called hourly...
[image]

So I tried it, and the message `Build options changed, rebuilding all` is gone in 1.8.6 hourly.
And my compile time goes from 30 seconds to 10 seconds (Verify button without any changes to the source).

marcmerlin, 2018-08-14:
Thanks @d-a-v  . Sure enough, the 1.8.6 pre-release does fix this without adding the new bugs that are in 1.9.0 beta.
https://www.arduino.cc/en/Main/Software sure does not make it obvious that the hourly builds (really nightlies) are of the next stable point release (i.e. 1.8.6) and therefore more stable than the beta builds.

Hopefully they get to fix the website as well as release 1.8.6 soon, it's been a long time with this bug already.

barsznica, 2018-09-16:
IDE 1.8.7 seems to be fine, too.

d-a-v, 2018-09-16:
This issue was fixed by #5043

woutput, 2018-10-09:
Mmmmhh,  I suffer from this bug as well in 1.8.5.
I also use the IDE for a digispark  because of another issue I found mentioned [here](https://videovind.com/video/sZGYrgvBDmM/installing-drivers-and-programming-the-digispark-attiny85-dev-boards--tutorial.html) 
> I couldn't get it to work using Arduino IDE v 1.8.7. It kept asking me to select a com port which was impossible as it was greyed out. I used an older version of the software 1.8.5 and this worked first time.

What is the latest stable that solves both? :)

d-a-v, 2018-10-09:
Current master should work with 1.8.5, have you tried it ?

woutput, 2018-10-11:
I used the Arduino (1.8.5) Board manager to install "**esp8266** by **ESP8266 Community** version **2.4.2**"
I am sorry, I was confusing the Arduino IDE and ESP8266 projects, so my question must come have across quite strangely ;)

Do I understand correctly that esp8266 version 2.4.2. has this "Rebuilding all" bug and that current github master solves it? If so, I'll try with Arduino 1.8.5.

devyte, 2018-10-11:
The correct combination is current git head and IDE 1.8.7.

woutput, 2018-10-11:
As indicated already, I use IDE 1.8.5 to work around another bug.

So: Do I understand correctly that esp8266 version 2.4.2. has this "Rebuilding all" bug and that current github master solves it & is compatible with IDE 1.8.5?

d-a-v, 2018-10-11:
There are two fixes:
- one in current git master that fixes pre-1.8.7 arduino IDE (core-2.4.2 and before do not have this fix)
- one in arduino-1.8.7 that does not need this fix, so 2.4.2 is ok with it (arduino 1.8.6 and before do not have this fix)

woutput, 2018-10-11:
Clear answer, thanks @d-a-v !

TyIsI, 2018-10-15:
Can confirm that this is fixed with Arduino release 1.8.7.

Burt-Silverman, 2018-12-11:
I just upgraded from 1.8.5 to 1.8.8. I can confirm that aggressive caching went from broken to working. Unbelievable amazing speed up.