esp8266/Arduino #3915: OTA: Invalid bootstrapping

jonatanolofsson, 2017-12-05:
### Basic Infos

#### Hardware
Hardware:			?Wemos D1 mini?
Core Version:      	?git/master?

### Description
After having succesfully built an appication (custom makefile, i.e. command line) and uploaded it successfully through UART, I tried uploading with espota.py:

```
python tools/espota.py -i 192.168.1.66 -f app.bin
```

### Debug Messages

```
[ERROR]: Bad Answer: ERR: ERROR[11]: Invalid bootstrapping
```

I then tried, with my own application running on the board, to upload through platformio, but this yielded the same error. If, however, I do the upload from platormio via uart first, I can once again use OTA to upload (though does not run without manual reset). The platformio uses the arduino framework version they call "1.20300.1", whereas again my application is built from the git master branch.

### Question:
Do I need to create a different binary for OTA uploads than serial uploads? 

Alternatively, how do I set boot mode correctly for OTA?

WereCatf, 2017-12-06:
The documentation already says that if you upload a sketch through serial, you'll have to reset the board first manually, before OTA will work, ie. upload via serial, reset the board, then use OTA.

jonatanolofsson, 2017-12-08:
That solved the issue! I  Thank you! Though may I ask where in the documentation you are referring to?

closed 2017-12-08 by jonatanolofsson

satblip, 2018-09-02:
It solved the issue here too (platformio environment), thanks!

ArtiomLK, 2019-07-04:
> That solved the issue! I Thank you! Though may I ask where in the documentation you are referring to?

<p><strong>Note:</strong> ESP module should be reset after serial upload. Otherwise subsequent steps will not work. Reset may be done automatically for you after opening serial monitor as visible on the screenshot above. It depends on how you have DTR and RTS wired from USB-Serial converter to the ESP. If reset is not done automatically, then do it by pressing reset button or manually cycling the power. For more details why this should be done please refer to FAQ regarding ESP.restart().</p>

from [ESP8266 Arduino Core - OTA UPDATES](https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html#application-example)