esp8266/Arduino #3917: Using SPI in an interrupt handler?

judge2005, 2017-12-05:
Hi,

I would like to use SPI from inside an interrupt handler (timer0), but I keep getting Illegal instruction resets. Is there anything I can do other than redefine all the methods on the SPI class with ICACHE _RAM_ATTR? How big is this ICACHE_RAM_ATTR anyway?

A (very) quick look seems to indicate that it doesn't call outside of the class.

Is it worth considering that any function that doesn't call system functions be declared with this tag? E.g. ESP.getCycleCount(), which I use to reset the interrupt timer from within the interrupt handler.

Thanks - Paul

devyte, 2017-12-07:
You can't call something as time consuming as SPI functions from within a hardware timer ISR due to tight timing requirements.
About ICACHE_RAM_ATTR, it is a tag that tells the compiler to put the code in ram instead of flash, so as to assure that there is no chance of a flash cache messup during execution (this explanation is simplified). The requirement is that the ISR, and all functions called from within the ISR, be in ram.
I suggest setting a flag from within the ISR, and check that flag from within the loop. When enabled, do your SPI stuff and disable the flag.
Having said the above, this is not the right place to ask such questions. This is an issue tracker.
Closing per issue policy doc.

closed 2017-12-07 by devyte

dalbert2, 2018-06-05:
A quick comment on this: the OP asks a reasonable question and I respectfully disagree that SPI functions should not be callable from within a hardware ISR.  Unlike I2C, SPI is typically a very fast interface with most devices allowing at least a 10MHz clock rate; even if SPI functionality were implemented via software bit-banging, it is plenty fast enough to do within an ISR, I have done this often on much slower processors.  For many applications, setting a flag in the ISR and checking it in the polling loop will dramatically increase interrupt latency and in many cases will make the ESP8266 an unsuitable platform.

A reasonable solution for the OP is to avoid using the Arduino SPI class at all and implement a bit-banged interface to his/her device within the ISR.  As @devyte noted, all code running in the interrupt context must have the ICACHE_RAM_ATTR decorator.