esp8266/Arduino #3918: Same code with 2.2.0 runs on Arduino IDE but not in Visual Micro. With 2.3.0 vise versa.

Schullebernd, 2017-12-05:
I have a very weired problem here. I created a sketch for a  WeMos D1 Mini.

If I compile and upload the sketch with the **Arduino IDE (1.8.5)** by using ESP8266 board package version **2.2.0**, everything is running perfekt. If I complie **exactly the same code** with **Visual Micro** (for VS2017) by using ESP8266 board package version **2.2.0**, the Wemos continously reboots because of wdt reset.

But thats not all!

If I switch to the ESP8266 board packge version **2.3.0 its vise versa**. If I compile and upload with Arduino IDE by using the **2.3.0** package, the Wemos continously resets. If I compile and upload with Visual Micro by using the 2.3.0 packe, everything works perfect.

Can someone help me or tell me the differences between Arduino IDE compiling and Visual Micro compiling?

Just in case someone wants to test it, the sketch can be downloaded here: https://github.com/Schullebernd/SBAquaControl/tree/master/Sources (For testing nothing else then the Wemos is needed, because the wdt reset problem occures before the first Serial.print() command gets executed)

Thanks!

devyte, 2017-12-07:
@Schullebernd akaik, Visual MIcro is not officially supported here, so questions related to that should be asked elsewhere.
The 2.3.0 package is now ancient. If you have issues with it, I suggest trying with latest git, or at least 2.4.0-rc2.

devyte, 2017-12-08:
After a quick look at your linked code, I don't see anything glaringly obvious. 
However, you are doing a whole lot of String manipulation, including long chains of operator+() and replace() and so on, which causes mem fragmentation. Also, all your const char * strings are not stored in flash (lack the F() or PROGMEM macros), including long segments of html, which causes out of mem issues, due to silent alloc/realloc failures.
In addition, you do a lot of lengthy operations at a time, including several SD card IO operations in sequence, . It is possible that you are taking the cpu away from servicing the wifi stack for too long. You may want to do some time measurements on how long callbacks take to execute from start to finish.
But overall, latest git with lwip2 is your best bet. If that still causes problems, I suggest breaking our code down into smaller pieces and testing them individually. If you uncover a problem, please open a new issue with the relevant details.
I don't see a specific issue with the core, so closing as OT (not a support forum for a particular project), see issue policy doc.

closed 2017-12-08 by devyte

Schullebernd, 2017-12-12:
Thanks a lit for your reply. I‘ll try the 2.4.0 Version on both plattforms.
I was wondering about, that the ESP8266 immediatly resets still befire the first code line gets executed. This should not be caused by bad coding style. But again, I‘ll try to clean the code and do a step by step analysis!