esp8266/Arduino #3921: HTTPClient not Working with HTTPS

botletics, 2017-12-05:
Hi there,

I am using a NodeMCU (ESP-12) and I'm trying to get the following code to work using the HTTPClient library:

`#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

const char* ssid = "........";
const char* password = "........";
String deviceID = "testing123abcxyz";

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  
  // Post data to dweet.io
  HTTPClient http;
  String URL = "http://dweet.io/dweet/for/" + deviceID + "?foo=bar&test=123"; // Works with HTTP
  http.begin(URL); // Works with HTTP
  
  int httpCode = http.GET();
  if (httpCode > 0) {
    String payload = http.getString();
    Serial.println(payload); // Print response
  }
  
  http.end();
}

void loop() {
  // Nothing here
}`

If I change line 25 to "https://" instead of "http://" then it doesn't work. I know that this library can work with HTTPS with a fingerprint, but what if I don't have one?

Thanks!

devyte, 2017-12-05:
@botletics not the right place to ask "how do I ..." questions, this is an issue tracker. Please refer to a community forum for assistance.
Closing  per issue policy doc.

closed 2017-12-05 by devyte

botletics, 2017-12-05:
Oh OK, sorry.

Vishalkargathara, 2019-08-09:
issues always arrive in the form of questions so can not define what is a question and what is an issue.

Vishalkargathara, 2019-08-09:
i am getting **invalid operands of types 'const char [36]' and 'const char [22]' to binary 'operator+'**
this error sir, can you please help me?
---------------------------------------------------------------------------------------------
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

const char* ssid = "vishal";
const char* password = "vishal1723";

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  
  // Post data to dweet.io
  HTTPClient http;
  
  String URL = "http://localhost/project/update.php" + "?units=300&amount=123"; // Works with HTTP
  http.begin(URL); // Works with HTTP
  
  int httpCode = http.GET();
  if (httpCode > 0) {
    String payload = http.getString();
    Serial.println(payload); // Print response
  }
  
  http.end();
}

void loop() {
  // Nothing here
}

devyte, 2019-08-09:
you're trying to use operator+() between two const char strings, you can't do that in C++. wrap one with String() instead.

Tanmay-901, 2021-07-14:
@devyte  I am getting error code -1 consistently, I've used the IPV4 address. Here is the chunk of code: 
```
postData = String("http://192.168.56.1/testcode/server.php?") + String("amount_charged=") + String(amount_charged) + String("&name=") + String(names);
    http.begin(postData);
    int httpCode = http.POST(postData);   //Send the request
    String payload = http.getString();    //Get the response payload

    Serial.println(httpCode);   //Print HTTP return code
    Serial.println(payload);    //Print request response payload
```
Kindly help, I cannot get it to work.

4meters, 2023-05-28:
Is there any fix for that? Tried to do get request on https and it's not working, but http works.

jonafriendj, 2023-05-28:
Try using WiFiClientSecure instead of HTTPClient