esp8266/Arduino #3923: [doc] Wrong links in documentation

Rotzbua, 2017-12-06:
### Basic Infos

Hardware:		n.a.
Core Version:      	since 2.4.0-rc1

### Description

With the change (https://github.com/esp8266/Arduino/commit/283eb97cd3f6dac691e8350a612021f1e1ccd8e3) from `.md` to `.rst` of the documentation file format many links are not updated and still link to `.md`.

### Related PR

#3907
#3875
#3874
#3873

devyte, 2017-12-18:
The first two PRs are merged, the last two are closed because the branch seems to have been deleted.
If there are additional link fixes, please open a new issue, and I'll review.

closed 2017-12-18 by devyte

Rotzbua, 2017-12-18:
@devyte Yes there are some more broken doc links. I do not have the time to fix them all, sorry. I guess fixing the links should be added to roadmap until final release 2.4.0.