esp8266/Arduino #3926: Error compile Windows 10

sabas1080, 2017-12-06:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP12
Core Version:      	2.3.0

### Description

Problem description

### Settings in IDE

Module:  			Wemos D1
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

Example Blink
```

### Debug Messages

```
unable rename

arduino.ar reason: File exists
```

![24882775_1039864912834193_1928425778_o](https://user-images.githubusercontent.com/2874292/33676995-e8f58e70-da7c-11e7-8dae-bbbb6f99fac4.png)

I already tried running as administrator, in Arduino IDE 1.8.4 and 1.8.5 installer and portable

#https://github.com/esp8266/Arduino/issues/2987

devyte, 2017-12-06:
@sabas1080 I suggest a clean install. That includes deleting the Arduino15 folder. Don't forget to back up your projects before deleting.
Closing as can't reproduce.

closed 2017-12-06 by devyte

sabas1080, 2017-12-06:
Also try cleaning and deleting Arduino15 and the same error

Spackstor, 2019-12-15:
I believe this problem is caused by not closing your file before exiting Arduino. The first time this happens running as administrator may work. The second time, it will be the administrator's working file and Windows will protect it. It is not the only file involved.
So find the file's Temp directory by doing a search on the file and delete the Temp directory.  This worked for me and so far so good.