esp8266/Arduino #3929: Errorr espcomm ESP8266

koltipelto, 2017-12-07:
I'm having problem to upload the program, my conections is:

[image]

### Basic Infos

#### Hardware
Hardware:			ESP8266  12E

C:\Users\*****\Documents\Arduino\sketch_dec04b\sketch_dec04b.ino:34:24: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

 char* subscribeTopic = "MK-SmartHouse/lights/MK-LEDStripControl1"; 

                        ^

O sketch usa 260015 bytes (59%) de espaço de armazenamento para programas. O máximo são 434160 bytes.
Variáveis globais usam 37300 bytes (45%) de memória dinâmica, deixando 44620 bytes para variáveis locais. O máximo são 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

devyte, 2017-12-07:
@koltipelto not the right place.to ask for help. This is an issue tracker for the core, not a support forum. Also, that error message is documented. You either have bad wiring or a dead module. From your drawing, I think it's bad wiring.
Closing per issue policy doc.

closed 2017-12-07 by devyte

koltipelto, 2017-12-07:
@devyte  Thanks a lot! Was a bad wiring! Sorry for the post!