esp8266/Arduino #3930: i am getting vcc as 3.54 rather than 3.25v with new sdk, the speed of boot increased. when i resart with rst pin making low then it shows right. 3.25v

pavankarney, 2017-12-07:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.4.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			ESP8266 12-f
Flash Size: 		1MB?
CPU Frequency:		160Mhz
Flash Mode:			?qio?
Flash Frequency:	80Mhz
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>
#define VCC_ADJ 0.939829
        
void setup() {
Serial.begin(115200);
  while (!Serial);
  Serial.println();
}

void loop() {
float battPWR = ((float)ESP.getVcc()*0.939829) / 1000.0f;
Serial.print(F("V = ")); Serial.print(battPWR); Serial.println(" volts");
delay(1000);
}
```

### Debug Messages

```
messages here
```

5chufti, 2017-12-07:
I can't see any error if you get correct voltage after reset, say in defined state of the system.
As you do not specify under which circumstances you get higher reading ... mybe wifi is off.
From experience I can tell you that the reading will be higher when wifi is off.

devyte, 2017-12-08:
@pavankarney After serial upload, you are supposed to always do a physical reset of the ESP. This is a known issue for which there is no fix. If you don't do that, all bets are off as to the ESP's behavior.
Also, the previous comment is correct. The wifi radio draws current, so when it's on voltage will be higher, and when it's on it will be lower. The same goes for all other hardware internals that draw significant current.
You have access to the code executed by ESP.Vcc(), so you can inspect it, find the issue, modify the code, test changes, and propose a solution. After a quick look, I don't see anything obvious in there.
Finally, you could just have an electrical bug, e.g.: some boards have a resistor divider on A0 to scale the 0-1V range up to 0-3.3V. This is just an example, I'd expect such a divider to drop the voltage, not increase it, but it could be something else.
At this point, I don't see anything that can be done within this repo. If you investigate and find a problem within the code of this repo, please report it here, and this issue can be reopened. Alternatively, open a new issue and *fill in all the requested info*, it is requested for a reason.
Closing as non-actionable.

closed 2017-12-08 by devyte

pavankarney, 2017-12-08:
I am using this for my Lora sensor node, where along with my sensor values
i take vcc readings while sending Lora packets.
Previously iwas compiling using older ask 2.3.0-rc2 and all was fine iwas
getting right voltage when working with deep sleep restart with wifi off.

But now with newer SDK i see this problem when I switch the power on or
even restart it it's giving higher voltage 3.54 v than giving right voltage
3.25v

I found the booting speed increased and my sketch size reduced by atleast 5
%. For that I am very happy.

But my primary concern is getting right value of battery connected tho
esp8266

If you please give me any suggestion or correction i will be thank full for
your support

Regards
Pavan

devyte, 2017-12-08:
2.3.0-rc2 is beyond ancient, it predates 2.3.0, as it was a release candidate.
I seem to remember something about changes in the ADC calibration in an Espressif SDK update, which would be included in the 2.4.0-rc2 and latest git, but I don't remember the details.
Whatever else, your approach is not wrong: you are supposed to calibrate the value read from the ADC. I usually do a linear fit, so I read at least 2 values within the range I expect to read, compare against a multimeter, and calculate a factor and an offset, which are then used to rectify the ADC value.
Alternatively, you can configure for reading the ADC instead of Vcc, and bridge VCC to pin A0. Make sure that the range for A0 is correct: e.g.: on bare ESP12 the range is 0-1V, while on e.g.: Wemos D1 mini or Nodemcus it's 0-3.3V thanks to an onboard resistor divider.

pavankarney, 2017-12-08:
Thank you for helping. I will calibrate my esp again for new SDK and check.

Thankyou
Regards
Pavan