esp8266/Arduino #3932: wificlientsecure fails to connect with smtp2go, but works with gmail

kapyaar, 2017-12-07:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	2.4.0-rc2

### Description

I am working on the ESP8266 gmail sender (http://www.instructables.com/id/ESP8266-GMail-Sender/) to send email using WiFiClientSecure. I can get it to work with gmail, but fail to get even connect to the server with mail.smtp2go.com. 

### Settings in IDE

Module:  			?nodeMCU 0.9?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?DOUT?
Flash Frequency:	?40Mhz?
Upload Using:		?SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp
#include <Gsender.h>
#include "ESP8266WiFi.h"

#pragma region Globals
const char* ssid = "MyWiFi"; // WIFI network name
const char* password = "Password"; // WIFI network password
uint8_t connection_state = 0; // Connected to WIFI or not
uint16_t reconnect_interval = 10000; // If not connected wait time to try again
#pragma endregion Globals

uint8_t WiFiConnect(const char* nSSID = nullptr, const char* nPassword = nullptr)
{
  static uint16_t attempt = 0;
  Serial.print("Connecting to ");
  if (nSSID) {
    WiFi.begin(nSSID, nPassword);
    Serial.println(nSSID);
  } else {
    WiFi.begin(ssid, password);
    Serial.println(ssid);
  }

  uint8_t i = 0;
  while (WiFi.status() != WL_CONNECTED && i++ < 50)
  {
    delay(200);
    Serial.print(".");
  }
  ++attempt;
  Serial.println("");
  if (i == 51) {
    Serial.print("Connection: TIMEOUT on attempt: ");
    Serial.println(attempt);
    if (attempt % 2 == 0)
      Serial.println("Check if access point available or SSID and Password\r\n");
    return false;
  }
  Serial.println("Connection: ESTABLISHED");
  Serial.print("Got IP address: ");
  Serial.println(WiFi.localIP());
  return true;
}

void Awaits()
{
  uint32_t ts = millis();
  while (!connection_state)
  {
    delay(50);
    if (millis() > (ts + reconnect_interval) && !connection_state) {
      connection_state = WiFiConnect();
      ts = millis();
    }
  }
}

void setup()
{
  Serial.begin(115200);
  connection_state = WiFiConnect();
  if (!connection_state) // if not connected to WIFI
    Awaits(); // constantly trying to connect

  Gsender *gsender = Gsender::Instance(); // Getting pointer to class instance
  const char* EMAILBASE64_LOGIN = "base64_encoded_loginemail";
  const char* EMAILBASE64_PASSWORD = "base64_encoded_password";

  const char* FROM = "myaddress@gmail.com";

  String subject = "Subject is optional";
  if (gsender->Subject(subject)->Send("secondEmail@gmail.com", "Alerts are working now")) {

    Serial.println("Message send.");
  } else {
    Serial.print("Error sending message: ");
    Serial.println(gsender->getError());
  }
}

void loop() {}
```

### Debug Messages

```
With gmail I get the following:

Connecting to :smtp.gmail.com:465
please start sntp first !
220 smtp.gmail.com ESMTP n3sm3479813qte.14 - gsmtp

HELO localhost:
250 smtp.gmail.com at your service

AUTH LOGIN:
334 VXNlcm5hbWU6

EMAILBASE64_LOGIN: d*********************************9t
334 UGFzc3dvcmQ6

EMAILBASE64_PASSWORD:b************E=
235 2.7.0 Accepted

MAIL FROM: <w**********e@gmail.com>
250 2.1.0 OK n3sm3479813qte.14 - gsmtp

RCPT TO: <a*******c@gmail.com>
250 2.1.5 OK n3sm3479813qte.14 - gsmtp
DATA:
354  Go ahead n3sm3479813qte.14 - gsmtp
250 2.0.0 OK 1512667183 n3sm3479813qte.14 - gsmtp
221 2.0.0 closing connection n3sm3479813qte.14 - gsmtp
============================================

with mail.smtp2go.com, I tried different port numbers and get teh following.
Connecting to :mail.smtp2go.com:2525
please start sntp first !
ssl->need_bytes=8301 > 6859

Connecting to :mail.smtp2go.com:465
please start sntp first !
Error: connection lost

Connecting to :mail.smtp2go.com:25
please start sntp first !
ssl->need_bytes=8301 > 6859
```

5chufti, 2017-12-08:
here is an example that says to work with smtp2go
https://playground.arduino.cc/Code/Email

kapyaar, 2017-12-08:
@5chufti Oh, I can get to work on smtp2go with non secure client. It is just the secure client part that is not working.

igrr, 2017-12-26:
Thanks for reporting this @kapyaar, looks like our implementation of max_fragment_length extension may have a bug. Working on a fix for this...

closed 2017-12-28 by devyte

kapyaar, 2018-07-18:
@devyte  I came back to this issue, and saw that [#4033](https://github.com/esp8266/Arduino/pull/4033) fixes this issue. I am currently using 2.4.1 latest release, and I checked my libraries/ESP8266WiFi/src/WiFiClientSecure.cpp. The file seems to have this fix but the issue persists. 

Connecting to :mail.smtp2go.com:25
ssl->need_bytes=8301 > 6859

Am I missing anything? Any other changes that I have to make?

devyte, 2018-07-18:
1. Relase 2.4.1 has several known issues that have since been fixed in latest master branch. Try latest git instead.
2. I'm not sure exactly how gsender works, but I see a returned pointer to an instance, and it never gets freed. Beware mem leaks.
3. The axtls secure lib has a long list of known issues. It is no longer maitained by the authors, so bearssl was integrated to replace it. The axtls is still the default, b ecause bearssI is still experimental, but I strongly suggest studying the bearssl examples to try and get gsender to use the bearssl::wificlientsecure instead of the normal wificlientsecure that uses axtls.

kapyaar, 2018-07-22:
@devyte got it working with bearSSL. Thanks