esp8266/Arduino #3935: trying to connect: espcomm_open failed

FrancisZamora, 2017-12-08:
### Basic Infos

#### Hardware
Hardware:			NOD EMCU ESP 12 E 
Core Version:      ??

### Description

Problem description

### Settings in IDE

Module:  			 ESP 12E MODULE
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			4M
Flash Frequency:	115200
Upload Using:		USBMODEM1421
Reset Method:		???

### Sketch

#include <WiFiClientSecure.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WiFiUdp.h>
#include <ESP8266WiFiType.h>
#include <ESP8266WiFiAP.h>
#include <WiFiClient.h>
#include <WiFiServer.h>
#include <ESP8266WiFiScan.h>
#include <ESP8266WiFiGeneric.h>
#include <ESP8266WiFiSTA.h>
#include "ESP8266WiFi.h"
/* TSL2591 Digital Light Sensor */
/* Dynamic Range: 600M:1 */
/* Maximum Lux: 88K */

#include <ArduinoJson.h>
#include <Wire.h>
#include <Adafruit_Sensor.h>
#include "Adafruit_TSL2591.h"
#include "Adafruit_Si7021.h"
#include <ESP8266WiFi.h>

const char* ssid = "traphouse3";
const char* password = "liquorstore420";

// connect SCL to analog 5
// connect SDA to analog 4
// connect Vin to 3.3-5V DC
// connect GROUND to common ground

Adafruit_TSL2591 tsl = Adafruit_TSL2591(2591); // pass in a number for the sensor identifier (for your use later)
Adafruit_Si7021 sensor = Adafruit_Si7021(); //  identifier for temp/humidity sensor
StaticJsonBuffer<200> jsonBuffer;

/**************************************************************************/
/*
    Displays some basic information on this sensor from the unified
    sensor API sensor_t type (see Adafruit_Sensor for more information)
*/
/**************************************************************************/
void displaySensorDetails(void)
{
  sensor_t sensor;
  tsl.getSensor(&sensor);
  Serial.println(F("------------------------------------"));
  Serial.print  (F("Sensor:       ")); Serial.println(sensor.name);
  Serial.print  (F("Driver Ver:   ")); Serial.println(sensor.version);
[... 270 lines cut ...]
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

devyte, 2017-12-08:
@FrancisZamora This message provides the hint:

> setting port from /dev/tty.usbserial to /dev/cu.usbmodem1421

I somehow doubt that you have the ESP connected over a usbmodem.
Also, your local static JsonBuffer inside getJSON() is shadowing the global one. A jsonBuffer should never be global, as it is not really meant to be reused. It should be constructed, used, destroyed, period.
Closing due to user error.

closed 2017-12-08 by devyte

FrancisZamora, 2017-12-08:
Im connected through a USB.. the code works when I upload to an arduino board and run the sensor code.

FrancisZamora, 2017-12-08:
It might be something to do with esptool.py

devyte, 2017-12-08:
setting port from /dev/tty.usbserial to /dev/cu.usbmodem1421
usbserial is a usb serial port. The other usbmodem sounds like something else. Check your serial Port setting in the menu.