esp8266/Arduino #3941: WPS fails with most new (802.11ac) access points

dalbert2, 2017-12-09:
### Basic Infos
WPS connection fails with most newer access points but work fine with older b/g/n APs.

#### Hardware
Hardware:			ESP-12F
Core Version:      	2.1.0-rc2

### Description

The same code works perfectly to connect via WPS with older (802.11b/g/n) access points but fails on newer (ac1900) access points even when the 5GHz radio is disabled.  WPS finds the access point SSID but the negotiation to get the PSK fails.  100% failure rate with Belkin AC1900 DB (Model F9K1124V1)

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp

extern "C" {
   #include "user_interface.h"
}

LOCAL void ICACHE_FLASH_ATTR
wps_status_cb(int status)
{
   wifi_wps_disable();
   Serial.printf("\n\rwps_status_cb: ");
   switch (status) {
   case WPS_CB_ST_SUCCESS:
      Serial.printf("success!");
      wifi_station_connect();
      break;
   case WPS_CB_ST_FAILED:
      Serial.printf("failed.");
      break;
   case WPS_CB_ST_TIMEOUT:
      Serial.printf("timeout.");
      break;
   case WPS_CB_ST_WEP:
      Serial.printf("WEP: not supported.");
      break;
   case 4:
      Serial.printf("scan error.");
      break;
   default: Serial.printf("%u", status);
   }
   Serial.println();
}

void setup() {
   Serial.begin(115200);
   system_set_os_print(1);
   Serial.setDebugOutput(true);
   wifi_set_opmode(STATION_MODE);
   wifi_station_set_auto_connect(0);
   delay(5000);
   wifi_wps_disable();
   wifi_wps_enable(WPS_TYPE_PBC);
   wifi_set_wps_cb(wps_status_cb);
   wifi_wps_start();
}

void loop() {
   yield();
}
```

### Debug Messages

```
wps scan
build public key start
build public key finish
f r0, wifi status: 0 phy_mode: 3
wps discover [belkin.150]
scandone
WPS: neg start
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt
wifi_wps_disable
state: 5 -> 0 (0)
rm 0

wps_status_cb: failed.
```

dalbert2, 2017-12-09:
Also fails with the example in 2.4.0 documentation (http://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/station-class.html#wps)

```
#include <ESP8266WiFi.h>

void setup(void)
{
  Serial.begin(115200);
  Serial.println();

  Serial.printf("Wi-Fi mode set to WIFI_STA %s\n", WiFi.mode(WIFI_STA) ? "" : "Failed!");
  Serial.print("Begin WPS (press WPS button on your router) ... ");
  Serial.println(WiFi.beginWPSConfig() ? "Success" : "Failed");

  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println();
  Serial.print("Connected, IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {}
```

dalbert2, 2017-12-11:
Another data-point: WPS works fine with an Asus TM-AC1900 access point, but fails with the Belkin AC1900

Rotzbua, 2017-12-11:
Can the esp connect to the Belkin with normal pw?

dalbert2, 2017-12-11:
Yes, this is strictly a WPS problem. The ESP receives the SSID but fails 
to process the key.
The same code works perfectly for WPS connection with an Asus TM-AC1900 
and for every 802.11b/g/n AP I've tried.  I had hoped it was a 
free-memory issue (the source of my original WPS problems with 
802.11ac), but I've freed up a lot of RAM (more than 20K) and it still 
does not work with the Belkin access point.  Any ideas?

Regards,
Dave

Rotzbua, 2017-12-12:
The esp do not support 802.11ac. Both router have a backwards compatibility for n-standard. It seems that the Belkin maybe have a different behaviour or implementation than the Asus.

First I would eliminate all link connections. So try to force the Beklin into 802.11n and 20 MHz mode, because the esp8266 do not support n-standard with 40Mhz and may have a problem with mixed mode of Belkin.
Best

dalbert2, 2017-12-12:
I should have been clearer.  The issue is that the ESP cannot connect to 
the router on 2.4GHz (I know the ESP does not support 5GHz).  I have 
tried every setting on the Belkin AP including turning off the 5GHz 
radio entirely and forcing g, n, 20MHz, and 20/40MHz modes.  When the 
ESP attempts a WPS pairing, it sees the Belkin SSID and attempts a 
connection but does not receive the key and the WPS process immediately 
fails. Other devices (e.g. my cell phone) are able to pair with the 
Belkin router using WPS.

shoemakersp, 2017-12-15:
You mentioned trying every setting, but I did not see security mode noted directly. Most new routers default to WPA2 only and I often have to set the 2.4 to WPA/WPA2 for "older" devices. Again, you likely tried this, but I am curious if this has been addressed. Thank you for noting this problem and sharing your results with this.

Rotzbua, 2017-12-19:
Actually there a two WPS versions available. WPS 1.0 and 2.0. The esp doc do not mention which version or compatibility is given... Maybe this is the source of problem?

dalbert2, 2017-12-19:
It might help if we had more information about which access points work 
and which don't.  So far, the Belkin AC1900 is the only AP I have not 
been able to get it to work with.  If anyone else is having WPS issues, 
I'd be grateful if you would add to this list which APs you have tested 
and which worked and which didn't.

igrr, 2017-12-27:
Keeping this issue open since it can help collect more data, but please be advised that WPS is fully implemented in ESP8266 non-OS SDK. As such, it is unlikely that anything will be done to solve this issue in this project.

devyte, 2018-07-13:
@dalbert2 I think the core has migrated to a new SDK since this was opened, and I think the new SDK includes some fixes for WPS. Please retry with latest git.
Be aware that you now have to set a menu option that disables a 4KB memory optimization in favor of WPS in the IDE.

dalbert2, 2018-08-29:
Hi devyte, I can't seem to figure out where that menu option is to disable the 4K memory optimization.  I use PlatformIO with VSCode, but I've also tried with the standard Arduino IDE and I don't see any menu option to enable WPS vs. the 4KB memory optimization.  Any guidance would be much appreciated!  I am building for the NodeMcu 1.0 (ESP12E) target and the linker chokes with error:
undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'

devyte, 2018-08-29:
With latest git the option was removed, because the optimization is disabled automagically when the core wps is used.

dalbert2, 2018-08-29:
OK, I'll try to figure out how to work with the latest git.  

FWIW, I'd like to offer some feedback to the devs who often make the very reasonable suggestion of doing things like rebuilding the boards with different options or using git latest.  

The documentation, while wonderful, often lacks sufficient detail for someone not involved in the development of the tools to actually use it.  For example (and to help the next person who finds this thread), the 2.4.2 FAQ indicates that to use WPS, you must use the board generator and suggests the command "./tools/boards.txt.py --allowWPS --boardsgen" which is great, except you don't know where that file is (e.g. when using platformio, it's in your home directory/.platformio/packages/framework-arduinoesspressif8266/tools for the Arduino platform it is even better hidden in AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\tools)  - these may be obvious to someone developing this framework, but it's not obvious to users (since the devs have done such a great job insulating us from all of this).  

Once you find it, it's still not obvious is how to make this work.  Running the command from  the Windows or cygwin CLI yields no output, but does create a new boards.txt and a boards.txt.orig with the added build option (e.g. nodemcuv2.build.noextra4kheap=-DNO_EXTRA_4K_HEAP.  Unfortunately, this doesn't do anything to solve the WPS failure to link.

Taking a guess, I added the build flag -D NO_EXTRA_4K_HEAP to my platformio.ini file and it linked!

No matter how bad this sounds like whining, it is really a suggestion.  I'll try to add some of this detail to the FAQ if I have permission to edit.

d-a-v, 2018-08-29:
@dalbert2 I wrote this tool and the doc. Thank you for your suggestions I will try to update the doc accordingly. I am using linux and the arduino IDE (only, or makefiles with the arduino builder), and we are, in the documentation, referring to the root directory of this arduino core. I have little knowledge with windows, and none with platformio. I was guessing users would use a locate/find/search tool to reach those files, but you prove me wrong. All combinations (linux, windows, arduino-IDE, platformio, VSCode, future arduino cli) cannot be listed everywhere in documentation when it comes to reach a particular file.
Would a mention like "use a search tool to locate `boards.txt` with your particular setup" have helped ?

About the boards generator not beeing talkative, you are right and some verbosity would help.

About the define you had to add by hand, you need to ask a platformio/esp8266/arduino maintainer (although if you are working with the git version you don't need it anymore like @devyte said above).

> Sorry for all of the whining, I've submitted the suggested details to the documentation on github under branch dalbert2/Arduino - perhaps you can merge them into the HEAD if you agree with them?

You need to push the branch to github, open github and press the PR button so we can see them.

dalbert2, 2018-08-29:
@d-a-v Thank you for writing both tool and doc!  The problem with this sort of thing is that it always seems obvious when you already know how to do it :-)  I did the pull requests so you can see the proposed documentation changes.  

The problem with find tools is that modern hard drives are so big that searching the entire drive can take a very long time (even on SSDs) so giving the user a hint about what they're looking for may be useful.  Unfortunately I also understand the problem with supporting a virtually unlimited number of build environments.  I think a common concession is to support popular windows environments because Windows users tend to be least sophisticated and most common (windows still owns 40-60% of developer desktops and probably more if you're only considering Arduino).  I use linux, windows, and cygwin depending on what I'm developing.  Arduino hides the core in a particularly ugly way, so documenting how to do things for Arduino IDE might be a reasonable minimal solution.  Examples are usually worth their weight in gold.

BTW, PlatformIO is cross-platform and *very* nice...you won't keep using Arduino IDE once you've tried it.

devyte, 2018-08-30:
@dalbert2 when you find something that isn't clear, or needs fixing, making a PR with improvements as you did is exactly what you're supposed to do. Thanks!
I've already merged them, I saw the PRs before this last comment. If there are any further changes, they can be handled with new PRs.

devyte, 2018-08-30:
@d-a-v you self assigned this. Please keep in mind that the original issue here isn't about the tool, or even about the stack placement for the 4K heap optim, but that wps has problems with newer ac routers. I suspect compatibility problems with specific router vendors, which would mean the problem is either in the sdk or in the router firmware, but that's just my opinion from following several threads about it over an extended period of time.

d-a-v, 2018-08-30:
@devyte right,
My self-assignment is to keep track with adding verbosity to the board generator and will remove this afterwards (less noise than adding a new specific issue).

I don't pretend to fix WPS with ac-enabled APs :)

earlephilhower, 2019-01-31:
This rather epic issue seems to have been fixed a while back including add'l work by @d-a-v on the boards generator.  Nothing to do here.

closed 2019-01-31 by earlephilhower