esp8266/Arduino #3942: OTA updaet fails with error code (numbers)

MrSliff, 2017-12-09:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	latest git version

### Description

Im trying OTA Update with latest git version. It worked a while ago, since it didnt work anymore, i upgraded to latest git version, but the problem still exists. Maybe i have too much flash used, but the compiler says this:

Sketch uses 327315 bytes (31%) of program storage space. Maximum is 1044464 bytes.
Global variables use 37496 bytes (45%) of dynamic memory, leaving 44424 bytes for local variables. Maximum is 81920 bytes.
Uploading...................................................................................................................................................................................................................................An error occurred while uploading the sketch
.

20:58:22 [ERROR]: 10721608536160853653653653653621

So the upload starts, but fails after some time. Dont know whats going on here -.-

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		OTA
Reset Method:		ck

### Sketch

#define min(a,b) ((a)<(b)?(a):(b))
#define max(a,b) ((a)>(b)?(a):(b))

/*************************************************/
/* Includes                                      */
/*************************************************/

#define SPIFFS_SINGLETON 1
#define DEBUG_MODE

#include "stdio.h"
#include <ESP8266WebServer.h>
#include <ESP8266HTTPUpdateServer.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <WiFiServer.h>
#include <WiFiUdp.h>
#include <ESP8266mDNS.h>
#include <ArduinoOTA.h>
#include <FS.h>

#include "RGBtoRGBW.h"

/*************************************************/
/* General                                       */
/*************************************************/

bool firstInit = true;

[... 499 lines cut ...]

unsigned long hexToDec(String hexString) {
  
  unsigned long decValue = 0;
  int nextInt;
  
  for (int i = 0; i < hexString.length(); i++) {
    
    nextInt = int(hexString.charAt(i));
    if (nextInt >= 48 && nextInt <= 57) nextInt = map(nextInt, 48, 57, 0, 9);
    if (nextInt >= 65 && nextInt <= 70) nextInt = map(nextInt, 65, 70, 10, 15);
    if (nextInt >= 97 && nextInt <= 102) nextInt = map(nextInt, 97, 102, 10, 15);
    nextInt = constrain(nextInt, 0, 15);
    
    decValue = (decValue * 16) + nextInt;
  }
  
  return decValue;
}

### Debug Messages

```
messages here
```

MarkyAD, 2017-12-09:
Seen the same error after uploading a sketch that had an infinite loop. Flash a known good version of your sketch using usb, reset the board with the hardware reset switch, then ota should work again.

MrSliff, 2017-12-10:
Hey, thanks for answerring, but i cant understand what you mean.

I dont think there is an infinite loop in my sketch. Its working properly. Serving Frontend HTML files nicely and the other functions are working too when connecting to the ESP via browser and sending data to the esp.

There is just the HTML update and the OTA update which make problems.

MrSliff, 2017-12-10:
so, upatding BasicOTA over previous flashed BasicOTA stops at 84% with an error.

MarkyAD, 2017-12-10:
I guess you did the hardware reset or power cycled it after flashing from usb.. bad flash chip maybe? Power supply voltage look ok while its uploading? If you've got any spare esp, now might be a good time to try one of them. Wish I could be of more help, but no problems with ota here other than when its my own fault (which is often).

MrSliff, 2017-12-10:
Yes, i power cycled and the chip sits on a pcb with voltage regulator, so there should not be a problem with power supply.

I maybe try another chip. But i have to say, OTA worked a while ago on the same chip.

MarkyAD, 2017-12-10:
I have 2 esp on latest git install, but both using the older version of lwip. 4 other esp I have on 2.4.0-rc2, all with ota... maybe its worthy to try going back to that and testing with basicOTA again?

MrSliff, 2017-12-10:
Thats it!

Thank you very much!

So latest git version has problems with ota, version 2.4.0-rc2 works.

closed 2017-12-10 by MrSliff

alirezaimi, 2019-11-02:
hi
There is no solution provided here about this problem, I have this problem right now with same board .
please help !

alirezaimi, 2019-11-03:
ANY idea ?!

devyte, 2019-11-03:
Yes: you can spend 5 minutes to search the repo source code for what updater error 4 means: space. 
Or if you can't manage to do that, you can seek peer help at a community forum like esp8266.com or stackoverflow, where if you ask nicely, maybe somebody will show you how to do the above.

egzumer, 2020-01-30:
> Yes: you can spend 5 minutes to search the repo source code for what updater error 4 means: space.

Maybe if you're developer of the mentioned source code then yeah. I have a similar problem and it's taking me hours to find out what is going on, and I'm close to giving up. I can't even reupload the BasicOTA example over the air. It stops at 1%. It worked before same day. Arduino IDE is not helping with substituting every possible compile error with "Multiple libraries were found".