esp8266/Arduino #3943: Please add support WizNet W5500 for Ethernet lib.

MirekFonCZ, 2017-12-10:


devyte, 2017-12-12:
@MirekFonCZ Are you willing to implement this yourself? You could port [this implementation](https://github.com/disnesquick/w5500), for example. 
Alternatively you could use that as an example to add support to the current Ethernet lib, e.g.: similar to how w5100.cpp|h are done. I suggest this approach.
If you port the above implementation, please make sure to ask about the license, as it is Apache, while this repo is LGPL.
If you are not willing to implement this yourself, please close the issue, as the current workload is way too high, and this won't be addressed in the foreseeable future.

d-a-v, 2018-05-07:
You can try https://github.com/d-a-v/W5500lwIP

d-a-v, 2018-09-05:
@MirekFonCZ can we close this issue ?

d-a-v, 2018-10-12:
Closing due to lack of feedback and precision.
Ethernet lib is arduino legacy, we have better with lwIP.
Wiznet is supported by third party libraries (still beta) which allows fully integration with all internal libraries.

devyte, 2018-10-30:
It's already implemented, and in a much better way with integration into lwip instead of a completely separate thing. It's not merged into core yet, still being tested.

mynameisbill2, 2019-11-15:
I find another way to use ESP8266 working with an ethernet board, from [here](https://esp8266hints.wordpress.com/2018/02/13/adding-an-ethernet-port-to-your-esp-revisited/), which Tony Onofrio "thoughts". 

> Tony Onofrio 
July 8, 2019 at 10:15 am 
Hi, thanks for this page and the tutorial. There is very little information around on how to get an ESP8266 working with an ethernet board, after all it already has wifi for connection. Unfortunately I have applications that can’t or won’t use wifi and it needs the larger programming space and ram, without going to the larger arduinos.
So I had been struggling with getting the ethernet board (W5500 with yellow connector) going using the esp8266 core for the arduino ide. The compiler was using the older(?) ethernet library that came with the esp core by default. This just would not work. I spent many hours going through the various library scripts, but could not get it to work. Also the newer ethernet library used for all the other arduinos (not esp) worked fine and had extra routines. I eventually found a solution and thought you and others might be interested. I may even try and get it implemented in respective gits.
The solution was to first delete or move the ethernet library in the esp core. If you move it you can restore it or reload the core through board manager, if this doesn’t work.
On my PC the esp ethernet library was in –
C:\Users\Tony\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ .
I moved the whole Ethernet folder. This will force the compiler to use the other arduino ethernet library.
A file in the arduino ethernet library will need a small modification to work correctly with the esp. You need to modify Ethernet.cpp , on my PC it’s located at 
C:\Program Files (x86)\Arduino\libraries\Ethernet\src\ .
You need to change after line 85 –
#if ARDUINO > 106 || TEENSYDUINO > 121
to
#if defined(ARDUINO_ARCH_ESP8266)
W5100.setIPAddress(ip.raw_address());
W5100.setGatewayIp(gateway.raw_address());
W5100.setSubnetMask(subnet.raw_address());
#elif ARDUINO > 106 || TEENSYDUINO > 121
Save it and it should all work. This will allow the ethernet library to work with the esp core and all the other cores. I’ve only tested with esp and arduino.
Unfortunately any updates to the esp core or arduino core will probably overwrite the changes.
This works with GPIO15 and the transistor buffer, or you can change the SCS pin to any other available pin using the Ethernet.init(15); instruction, seems the default is GPIO15 for an esp8266.

devyte, 2019-11-15:
Please be aware that use of Ethernet in that way doesn't integrate with the core, i. e. it's a completely different and stand-alone piece of code. As a result, there are a lot of things that can't be used, like the webserver/httpclient, among a lot of other things.
The current plan is to finalize @d-a-v 's implementation linked abive, that integrates with lwip and therefore the core. Merging is expected very soon.

d-a-v, 2019-11-15:
@mynameisbill2 @PuceBaboon

The newer Arduino official Ethernet library will be imported in this core when @PaulStoffregen accepts to merge [this very minor pull-request](https://github.com/arduino-libraries/Ethernet/pull/96) as a counterpart of [his requests](https://github.com/esp8266/Arduino/issues/2677#issuecomment-405744137) so we can submodule it to replace the older locally patched version we still have here.

Because they have much to do, maybe you can go there and gently remind this is needed.

PaulStoffregen, 2019-11-16:
> when @PaulStoffregen accepts to merge this very minor pull-request

I no longer have write access to Arduino's repository.  I can not merge it.  You must ask the Arduino developers to merge it.

mynameisbill2, 2019-11-17:
> Please be aware that use of Ethernet in that way doesn't integrate with the core, i. e. it's a completely different and stand-alone piece of code. As a result, there are a lot of things that can't [...]

I tested the libraries I mentioned above and @d-a-v 's W5500lwIP with the example -- httpclient, both worked well. Although I don't know why.

d-a-v, 2019-11-18:
Another PR is now ongoing on https://github.com/PaulStoffregen/Ethernet/pull/37

PR #6812 (2020-07-16): replace old Ethernet Arduino-library with v2, submoduled
supports W5500, W5200 and W5100

*edit*:
closes #2671 
fixes #3943
closed 2020-07-16 by d-a-v