esp8266/Arduino #3944: Configurable TLS timeout for mutual authentication

ztittle, 2017-12-11:
https://github.com/esp8266/Arduino/blob/7df28582858e5a935c721d131064e6e3082edfe8/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp#L356

I've been trying to connect to the AWS IOT MQTT endpoint, but it appears that the ESP 8266 chip takes longer than the hard-coded 5-second timeout to perform the TLS handshake. After increasing this value to a generous 30 seconds, I was able to successfully connect to AWS IOT. 

For reference, I was using the example code listed on https://github.com/copercini/esp8266-aws_iot/blob/master/examples/MQTT_x509_bin/MQTT_x509_bin.ino, but loading the certs from PROGMEM so I would not run out of memory.

Could the timeout be made configurable through a #DEFINE or an overload on connect()? This would greatly help adoption when integrating to a secure MQTT service such as AWS IoT. 

`Debug output`
```
.........wifi evt: 7
....................wifi evt: 3
.192.168.1.252Attempting MQTT connection...[hostByName] Host: 54.68.144.180 is a IP!
:ref 1
please start sntp first !
State:  sending Client Hello (1)
:wr 100 100 0
:wrc 100 100 0
:sent 100
:rn 1460
:rd 5, 1460, 0
:rdi 1460, 5
:rd 1455, 1460, 5
:rdi 1455, 1455
:c0 1455, 1460
:rn 1460
:rd 1460, 1460, 0
:rdi 1460, 1460
:c0 1460, 1460
:rn 118
:rd 118, 118, 0
:rdi 118, 118
:c0 118, 118
State:  receiving Server Hello (2)
State:  receiving Certificate (11)
=== CERTIFICATE ISSUED TO ===
Common Name (CN):               *.iot.us-west-2.amazonaws.com
Organization (O):               Amazon.com, Inc.
Location (L):                   Seattle
Country (C):                    US
State (ST):                     Washington
Basic Constraints:              CA:FALSE, pathlen:10000
Key Usage:                      critical, Digital Signature, Key Encipherment
Subject Alt Name:               iot.us-west-2.amazonaws.com *.iot.us-west-2.amazonaws.com
=== CERTIFICATE ISSUED BY ===
Common Name (CN):               Symantec Class 3 Secure Server CA - G4
Organization (O):               Symantec Corporation
Organizational Unit (OU):       Symantec Trust Network
Country (C):                    US
Not Before:                     Wed Aug 16 00:00:00 2017
Not After:                      Wed May 16 23:59:59 2018
RSA bitsize:                    2048
Sig Type:                       SHA256
=== CERTIFICATE ISSUED TO ===
Common Name (CN):               Symantec Class 3 Secure Server CA - G4
Organization (O):               Symantec Corporation
Organizational Unit (OU):       Symantec Trust Network
Country (C):                    US
Basic Constraints:              critical, CA:TRUE, pathlen:0
Key Usage:                      critical, Key Cert Sign, CRL Sign
Subject Alt Name:
=== CERTIFICATE ISSUED BY ===
Common Name (CN):               VeriSign Class 3 Public Primary Certification Authority - G5
Organization (O):               VeriSign, Inc.
Organizational Unit (OU):       VeriSign Trust Network
Country (C):                    US
Not Before:                     Thu Oct 31 00:00:00 2013
Not After:                      Mon Oct 30 23:59:59 2023
RSA bitsize:                    2048
Sig Type:                       SHA256
State:  receiving Certificate Request (13)
State:  receiving Server Hello Done (14)
State:  sending Certificate (11)
:wr 876 876 0
:wrc 256 876 0
:wrc 256 620 0
:wrc 256 364 0
:wrc 108 108 0
:sent 876
State:  sending Client Key Exchange (16)
:wr 267 267 0
:wrc 256 267 0
:wrc 11 11 0
:sent 267
wifi evt: 7
wifi evt: 7
wifi evt: 7
State:  sending Certificate Verify (15)
:wr 269 269 0
:wrc 256 269 0
:wrc 13 13 0
:wr 6 6 0
:wrc 6 6 0
State:  sending Finished (16)
:wr 85 85 0
:wrc 85 85 0
:wr 69 69 0
:wrc 69 69 0
Alert: close notify
failed, rc=-2 try again in 5 seconds
:sent 269
:sent 160
:rn 160
:rcl
:abort
Attempting MQTT connection...[hostByName] Host: 54.68.144.180 is a IP!
:ur 1
:del
:ref 1
```

torntrousers, 2017-12-18:
Just as an FYI, try this as its easier that than certificate/key in a byte array: https://github.com/HarringayMakerSpace/awsiot/blob/master/Esp8266AWSIoTExample/Esp8266AWSIoTExample.ino

igrr, 2017-12-26:
@ztittle @torntrousers Created https://github.com/esp8266/Arduino/pull/4027 to fix this, please have a look.

torntrousers, 2017-12-26:
@igrr the 15 sec default gets it working for me. Thanks.

closed 2017-12-26 by devyte