esp8266/Arduino #3948: When STA mode ESP is restarted, not reconnecting, instead WiFi.status() returns 4 (WL_CONNECT_FAILED)

konkav-js, 2017-12-11:
### Basic Infos

#### Hardware
Hardware:			ESP-12E (nodemcu clone)
Core Version:      	2.3.0
Arduino IDE:             1.8.5 and 1.8.2 (tried both)

### Description

I am using two ESPs to communicate with each other. One, the 'controller', is in AP mode, it has a WiFiServer started. The second, the 'robot' is in STA mode, it connects to the first. When I first power up the ESPs, after a few seconds, the connection is live between ESPs, and they are communicating fine. If I restart the controller (by the reset button or by power cycle), there is a drop in connection (logically), but few seconds later, the AP comes online, the STA connects, and everything is OK. But. If I restart the robot, it won't connect again, until I restart the controller again.

When the robot is restarted, the `Serial.println(WiFi.softAPgetStationNum());` value first drops from `1` to `0`, and when the robot ESP starts, in a second or two (I am watching it over Serial, side-by-side the two ESP), it goes back to `1`. This is all like it should be. But. 

During starting, the robot are looping in the `while (WiFi.status() != WL_CONNECTED)` loop forever. First, the `Serial.println(WiFi.status());` is returning `6` (WL_DISCONNECTED), even when the `WiFi.softAPgetStationNum());` on the controller has returnet to `1`. It gives `6` for some ten second, or a little more, and after that, is goes to `4` (WL_CONNECT_FAILED). The controller gives `1` all the time (except the two second, when the robot is restarting).

I did not touch these two programs in a half a year, so maybe my memory is not correct, but I think that before it worked fine. I was able to restart any of the ESP, and after a few second, it would reestablished the connection. I like to program robust solutions, so it is a high possibility, that it worked, and something happened since then in the development tools. I added just few lines of code, none which interfere with the communication. But I can't guarantee it.

In the effort to debug the problem, I added a lot of code that I don't think is necessary. There is static IP address on both side, I even made the channel static, to exclude as much of the parameters, as possible, to narrow down the problem. Am I doing something wrong? Or is it not going to work this way (having a direct connection between two ESP recover from the connection break)? I really not want to put an AP in between them. I tried to find solution, but didn't succeeded. Thanks in advance for the help.

### Settings in IDE
 
Module:			Generic ESP8266 Module
Flash Size:			4MB/(1MB SPIFFS)
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch on controller (server)

```cpp
#include <ESP8266WiFi.h>
IPAddress robotIP(10, 10, 10, 20);
IPAddress controllerIP(10, 10, 10, 10);
IPAddress subnet(255, 255, 255, 0);
IPAddress gateway(10, 10, 10, 1);
char APssid[20]= "...";
char APpass[20]= "...";
byte APchannel= 10;
WiFiServer wifiserver(23);
WiFiClient robot;

void setup() {
  Serial.begin ( 115200 );
  delay(100);
  wifiserver.stop();
  WiFi.persistent(false);
  WiFi.disconnect();
  WiFi.softAPdisconnect();
  WiFi.mode(WIFI_AP);
  WiFi.softAPConfig(controllerIP, gateway, subnet);
  WiFi.softAP(APssid, APpass, APchannel);
  wifiserver.begin();
  wifiserver.setNoDelay(true);
}

void loop() {
if (wifiserver.hasClient()) {
    if (robot) robot.stop();
    robot = wifiserver.available();
  }
   Serial.println(WiFi.softAPgetStationNum());
   delay(500);
}
```

### Sketch on robot (client)

```cpp
#include <ESP8266WiFi.h>
char APssid[20] = "...";
char APpass[20] = "...";
IPAddress controllerIP(10, 10, 10, 10);
IPAddress robotIP(10, 10, 10, 20);
IPAddress subnet(255, 255, 255, 0);
IPAddress gateway(10, 10, 10, 1);
byte port= 23;
WiFiClient controller;

void setup() {
  Serial.begin ( 115200 );
  delay(100);
  Serial.println(WiFi.status());
  controller.stop();
  WiFi.persistent(false);
  WiFi.disconnect();
  WiFi.mode(WIFI_OFF);
  delay(100);
  WiFi.mode(WIFI_STA);
  WiFi.config(robotIP, gateway, subnet);
  WiFi.begin(APssid, APpass);
  Serial.println(WiFi.status());

  while (WiFi.status() != WL_CONNECTED)
      {
        delay(500);
        Serial.println(WiFi.status());
      }

void loop() {
if (!controller.connected()) {
    Serial.println("Connecting to controller");
    controller.connect(controllerIP, port);
  }
}
```

devyte, 2017-12-12:
@konkav-js please retest with latest git, asitnhas an updated SDK. I also suggest building with lwjp2 (select from the IDE menu), as it fixes lots of socket issues (not related).
Remove the while() and try the WiFi event system. Set a global flag in the relevant event callback, check the flag in your loop.

konkav-js, 2017-12-12:
The above sketches are just part of the program. So yesterday I went, and tried to debug it a little more. I had an idea, that maybe something went wrong with a lot of uploading the programs, I changed a little, then upload, then change, then upload, so I searched for complete deleting the flash. I erased both ESPs (with esptool.exe). After the new upload, nothing has changed. I tried then to connect to the controller with my phone (android) and computer (win10). And the same has happened. After controller restart, the first connection was a success, but if I broke the connection, I couldn't connect back. Both phone and PC gave similar answer, authentication failed.
So I went and made two new sketches from the above code, uploaded them, and they worked. I could restart any of the ESPs, in any order, after few second, they reconnected. Clearly, something was wrong with the full program. Fortunately, I have just started to use git, so I have the version just before I started to add new things. As it was late yesterday, I didn't tried the previous version, I will try it today. But few thing crossed my mind. Is it possible, that I run out of RAM? I do have a lot of variables, and I just added a few more. Is there a possibility to print out the free memory in runtime? With the above sketches, I was able to connect with all 3 devices (ESP, phone and PC) at the same time, and connection was stable.
Today I will first try the previous version, and if it works, then I will add just the variable declarations. I am curious about this. 
I did tried removing the `while (WiFi.status() != WL_CONNECTED)` loop, but the ESPs never started communicating with each other. It never triggered `if (wifiserver.hasClient())` either. I presume that the problem is with the AP side.
I found the description to using the git version, but this lwjp2 is new to me. Is there some description about it?

devyte, 2017-12-12:
Closing, as a clean sketch upload works.

devyte, 2017-12-12:
I suggest readthedocs. The ESP class has a method for getting the free heap. Be careful of stack use, don't use large objects on the stack, as the stack size is tiny. Don't declare const char * strings, as they take up RAM, use PROGMEM steings instead.
It's more than likely that your full sketch has subtle mem corruption somewhere.

closed 2017-12-12 by devyte

konkav-js, 2017-12-12:
I kind of located the problem, and it wasn't the free heap. The both the compiler and the `ESP.getFreeHeap()` inside the main loop returns ~32k of free RAM, so the problem is somewhere else.
 I opened the two sketches (the previous "working", and the new, "broken"), and was adding the new code block-by-block, compiling, uploading, and testing. I it was weird, because the part that broke the re-connection was:
The non-working version:
```cpp
if ((robot.available() > 9) && (robocomm == 1 || robocomm == 4)) {
```
The working version:
```cpp
if (robot.available() && (robocomm == 1 || robocomm == 4)) {
```
There is no `>9` in the working version.
What is more interesting, is that the above line is working as expected (at least until the connection drops), because if I print out the value of the `robot.available()` after the `if`, in the non-working case, it prints out only numbers greater than 9. In the working version, there are some prints, where there is less than 9.

Next thing was to download the git version. With it, the "non-working" version is working great. I will leave this information here: **the version what is available today through Board Manager, has a weird bug, but the git version is working fine.**

devyte, 2017-12-13:
In that case, your cause most likely was one of the known issues in lwip1.4. Latest git currently defaults to lwip2 in the IDE.