esp8266/Arduino #3952: 'WPS_CB_ST_UNK' was not declared in this scope

ItsJasonn, 2017-12-12:
#### Hardware
Hardware:			?ESP8266?
Core Version:      	?No idea?

### Description

I am trying to send i2c commands using a ESP8266 web server. Everything works fine when trying to upload from my PC, but for some reason it doesn't work anymore and it gives me the following error message:

`D:\Program Files (x86)\Arduino\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp: In function 'void wifi_wps_status_cb(wps_cb_status)':

D:\Program Files (x86)\Arduino\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp:623:14: error: 'WPS_CB_ST_UNK' was not declared in this scope

         case WPS_CB_ST_UNK:

              ^

exit status 1
Error compiling for board Generic ESP8266 Module.`

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?512K?
CPU Frequency:		?80Mhz?
Flash Mode:			?DIO?
Flash Frequency:	?40Mhz?
Upload Using:		?SERIAL?
Reset Method:		?ck?

ItsJasonn, 2017-12-12:
## FIX:
I resolved the issue by completely uninstalling the ESP8266 board using the Boards Manager in Arduino IDE, reinstalling the ESP8266 board, wait 1-2 minutes, and tried uploading the sketch again.

devyte, 2017-12-12:
I was about to suggest the same: a clean install.
Closing.

closed 2017-12-12 by devyte