esp8266/Arduino #3954: ESP8266HTTPClient

cimba007, 2017-12-12:
### Basic Infos
http.getString() failes for large payloads > 8kb

Is 8kb the limit?

#### Hardware
Hardware:			WEMOS-MINI
Core Version:      	        2.4.0-rc2

### Description

### Settings in IDE

Module:  			Generic ESP8266 Module?
Flash Size: 		1MB
CPU Frequency:	80Mhz
Flash Mode:		dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		 nodemcu

### Sketch

Basically HTTPClient-Authorization example

### Debug Messages

```
[HTTP-Client][handleHeaderResponse] RX: 'transfer-encoding: chunked'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] Transfer-Encoding: chunked
[HTTP] GET... code: 200
[HTTP-Client] read chunk len: 8192
ssl->need_bytes=8256 > 6859
[HTTP-Client][writeToStreamDataBlock] connection closed or file end (written: 8192).
[HTTP-Client] read chunk len: 7656
[HTTP-Client][writeToStream] short write asked for 1460 but got 0 retry...
Fatal exception 3(LoadStoreErrorCause):
epc1=0x4010044e, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40248740, depc=0x00000000

Exception (3):
epc1=0x4010044e epc2=0x00000000 epc3=0x00000000 excvaddr=0x40248740 depc=0x00000000

ctx: cont 
sp: 3fff0c40 end: 3fff1030 offset: 01a0

>>>stack>>>
3fff0de0:  3ffeffd8 0000004b 3fff2d14 4010099c  
3fff0df0:  00001de8 3fff0f30 3fff2d1c 402060d0  
3fff0e00:  3ffeffd8 00000008 3fff2d1c 40205402  
3fff0e10:  5454485b 6c432d50 746e6569 72775b5d  
3fff0e20:  54657469 7274536f 5d6d6165 6f687320  
3fff0e30:  77207472 65746972 3fff896c 4010053a  
3fff0e40:  3120726f 20303634 3fff0f40 40100817  
3fff0e50:  3fff0e90 3fff0e60 00000008 00000e22  
3fff0e60:  40104f84 001ca765 3ffefdf8 00001de8  
3fff0e70:  000036e0 3fff0f40 3fff0f40 4020573b  
3fff0e80:  3fff5754 000005b4 3fff0f40 40205787  
3fff0e90:  4020137a 00000001 3fff0f30 4020569c  
3fff0ea0:  3fff0f30 00000ccc 0000111c 40204b59  
3fff0eb0:  3fff5754 000005b4 000005b4 00000001  
3fff0ec0:  3fff0f70 000005b4 3fff0ea0 3fff0ed0  
3fff0ed0:  3fff0f30 00002000 00002000 3fff0f70  
3fff0ee0:  3fff0f30 00003de8 00002000 40204d3d  
3fff0ef0:  3fff51d4 0000000f 00000004 40200a0d  
3fff0f00:  00001de8 3fff0f70 3fff0f40 40205806  
3fff0f10:  3ffe8a6c 3fff0f20 00000008 3ffe9940  
3fff0f20:  3ffe8a6c 3fff0f70 3fff0fe0 40204e44  
3fff0f30:  3ffe9940 00000000 000003e8 4020534d  
3fff0f40:  3fff8974 0000311f 0000311c 3fff0f50  
3fff0f50:  3fffdad0 00000000 3ffeffd8 3fff0004  
3fff0f60:  3fffdad0 000000c8 3ffeffd8 402022ac  
3fff0f70:  3fff2954 3fff2c34 3fff2934 0000000f  
3fff0f80:  0000000f 000020fb 00001388 3fff2c6c  
3fff0f90:  0000004f 00000042 3fff2974 0000000f  
3fff0fa0:  00000000 3fff2994 0000000f 00000000  
3fff0fb0:  3fff29b4 0000001f 00000011 3fff2c04  
3fff0fc0:  0000001f 00000014 00000000 00000000  
3fff0fd0:  000000c8 ffffffff 3ffefd00 00000001  
3fff0fe0:  00000000 00000000 00000000 00000000  
3fff0ff0:  00000000 00000000 00000000 00000000  
3fff1000:  00000000 3fff2a04 0000003f 0000003b  
3fff1010:  feefeffe 00000000 3ffefffc 402062d8  
3fff1020:  feefeffe feefeffe 3fff0010 40100a20  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v0c897c37
~ld
```

```
Decoding 21 results
0x4010099c: free at C:\Users\Marc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1737
0x4020609c: operator delete[](void*) at C:\Users\Marc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/abi.cpp line 84
0x402053f6: Print::printf(char const*, ...) at C:\Users\Marc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/Print.cpp line 205
0x4010053a: get_unpoisoned at C:\Users\Marc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 942
0x40104cf6: wdt_feed at ?? line ?
0x40100817: realloc at C:\Users\Marc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1700
0x40104db2: ets_timer_setfn at ?? line ?
0x40104f84: ets_timer_arm_new at ?? line ?
0x40205707: String::changeBuffer(unsigned int) at C:\Users\Marc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/WString.cpp line 518
0x40205753: String::reserve(unsigned int) at C:\Users\Marc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/WString.cpp line 518
0x4020137a: delay at C:\Users\Marc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/core_esp8266_wiring.c line 54
0x40205668: StreamString::write(unsigned char const*, unsigned int) at C:\Users\Marc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/StreamString.cpp line 28
0x40204b4d: HTTPClient::writeToStreamDataBlock(Stream*, int) at C:\Users\Marc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 176
0x40204d31: HTTPClient::writeToStream(Stream*) at C:\Users\Marc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 176
0x402057d2: String::String(char const*) at C:\Users\Marc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/WString.cpp line 518
0x40204e38: HTTPClient::getString() at C:\Users\Marc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 176
0x40205341: Print::write(char const*) at C:\Users\Marc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/Print.cpp line 205
0x402022a9: loop at D:\Users\Jinja\Documents\Arduino\esp8266_mycontroller_eink\esp8266_mycontroller_eink.ino/esp8266_mycontroller_eink.ino.ino line 78
0x402062a4: loop_wrapper at C:\Users\Marc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/core_esp8266_main.cpp line 57
0x40100a20: cont_norm at C:\Users\Marc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/cont.S line 109
```

d-a-v, 2017-12-19:
Can you try again with latest master ?

VjSky, 2017-12-22:
I can confirm it does not work with the latest master and that I have the same issue. 
When i get a 404 or 403 error for example with a custom 404 page, like the IIS default one it just crashes:

Exception (28):
epc1=0x40221356 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff2ab0 end: 3fff2ce0 offset: 01a0

>>>stack>>>
3fff2c50:  3fff2ca4 3fff2c8c 3fff2c8c 4021d132  
3fff2c60:  3ffe8b20 3fff1830 3fff1830 4021d22e  
3fff2c70:  00005f03 3fff1830 3ffe8538 4020d2bf  
3fff2c80:  00000000 00000000 00000000 3fff56dc  
3fff2c90:  0000000f 00000001 3fff56c4 0000000f  
3fff2ca0:  00000007 3fff511c 0000000f 00000007  
3fff2cb0:  00000000 00000000 00000001 3fff1cb8  
3fff2cc0:  3fffdad0 00000000 3fff1cb1 4021dd70  
3fff2cd0:  feefeffe feefeffe 3fff1cc0 4010070c  
<<<stack<<<

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v0c897c37
~ld

Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 7 results
0x40221356: strcpy at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strcpy.S line 103
0x4021d132: String::String(char const*) at C:\Users\stefan.stojanovic\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/WString.cpp line 687
0x4021d22e: String::operator=(String&&) at C:\Users\stefan.stojanovic\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/WString.cpp line 687
0x4020d2bf: loop at Q:\IT\EMBEDDED\Arduino\PROJECTS\esp-rfid-KONTROLA PRISTUPA 2018\Code\esp-rfid/esp-rfid.ino line 291
0x4021dd70: loop_wrapper at C:\Users\stefan.stojanovic\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/core_esp8266_main.cpp line 124
0x4010070c: cont_norm at C:\Users\stefan.stojanovic\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/cont.S line 109

cimba007, 2018-01-13:
HTTPClient doesn't support root CA and WifiClient doesn't support chunked .. well ..

closed 2018-01-13 by cimba007