esp8266/Arduino #3960: ESP8266 Webserver too slow

ItsJasonn, 2017-12-13:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	?

### Description

I have setup a UI for a robot arm. In this UI you can add multiple commands to a list and start this sequence and stop till all commands have been executed. It works perfect with 3 commands, but when having 4 or more commands in the list the ESP8266 board and website freezes and won't work anymore. I have a feeling this is because of the speed of the ESP8266, because whenever I try to list down the commands using a for loop to generate the HTML code it loads in very slow, and sometimes I get a 50% loaded page in my browser.
Is there a way I can make the board faster, or a method I can use to improve the speed of the board. Because now I can't control the robot arm with just 3 commands.

### Settings in IDE

Module:  			ESP12E
Flash Size: 		512K
CPU Frequency:	80Mhz
Flash Mode:		DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch
## This sketch mostly handles website generation
```
void auto_connect() {
  // show instruction on LCD
  lcd.setCursor(0,0);   lcd.print("Initialise WIFI....");
  lcd.setCursor(0,1);   lcd.print("1-connect MANUS AP");  
  lcd.setCursor(0,2);   lcd.print("2-Browse to:");  
  lcd.setCursor(0,3);   lcd.print("192.168.4.1");
  
  WiFiManager wifiManager;
  // Autoconnect to local wifi acces point. Only the first time, take your smartphone, look for open accespoint
  // connect en browse to 192.168.4.1. Find the prefered acces point and type the password 
  //reset saved settings, only used when changing pasword
  
  //wifiManager.resetSettings();
  
  //set custom ip for portal default is 192.168.4.1
  //wifiManager.setAPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

  //fetches ssid and pass from eeprom and tries to connect
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration
  wifiManager.autoConnect("AutoConnectAP");
}

void handleManus() {
  clear_display_on_first_acces();
  
  if(server.hasArg("BTN_NEW_COMMAND")) {
    server.sendHeader("Location", String("/newCommand"), true);
    server.send(302, "text/html", "");
  } else if(server.hasArg("BTN_START")) {
    server.sendHeader("Location", String("/runningPage"), true);
    server.send(302, "text/html", "");
  } else {
    if(server.hasArg("STAPPEN") && server.hasArg("SNELHEID") && server.hasArg("REMWEG") && server.hasArg("MOTOR") && server.hasArg("BTN_SAVE") && !server.hasArg("BTN_DONT_SAVE")) {
      int _motorId = 0;
[... 572 lines cut ...]
    delayMicroseconds(10000);
  }
  
  if(!reversed) {
    sequenceIndex += 1;
  } else {
    sequenceIndex -= 1;
  }
}

String getValueBySeperator(String data, char separator, int index) {
    int found = 0;
    int strIndex[] = { 0, -1 };
    int maxIndex = data.length() - 1;

    for (int i = 0; i <= maxIndex && found <= index; i++) {
        if (data.charAt(i) == separator || i == maxIndex) {
            found++;
            strIndex[0] = strIndex[1] + 1;
            strIndex[1] = (i == maxIndex) ? i+1 : i;
        }
    }
    return found > index ? data.substring(strIndex[0], strIndex[1]) : "";
}
```

igrr, 2017-12-13:
>     content += "<tr class='emptyRow'><td colspan='2'> </td></tr>";
>   content += "<tr class='emptyRow'><td colspan='2'> </td></tr>";

I suggest looking into using raw string literals for multiline strings
where possible, or at least using String.reserve to pre-allocate memory.

pipi61, 2017-12-13:
Hi!
Not a webserver slow, but string manipulate too much.
Reduce string operations.
Store webpage in flash, and use ajax to fill variable fields.
I use this: https://www.john-lassen.de/en/projects/esp-8266-arduino-ide-webconfig

devyte, 2017-12-16:
Doing sequenced concatenation ops like in this sketch is a Very Bad Idea on the ESP. It is very inefficient, makes for a huge number of allocs and reallocs, and it causes mem fragmentation.
Previous comments are correct.
Also, I assume you mean core version 2.3.0. Please use latest git, and build with lwip2 (option in IDE menu), as there are known slowness issues with lwip1.4.
Closing as user error.

closed 2017-12-16 by devyte