esp8266/Arduino #3961: OTA Update Web Browser - MobileFirst

MSergW, 2017-12-13:
libraries/ESP8266HTTPUpdateServer/src/ESP8266HTTPUpdateServer.cpp
replace:
```
R"(<html><body><form method='POST' action='' enctype='multipart/form-data'>
                  <input type='file' name='update'>
                  <input type='submit' value='Update'>
               </form>
         </body></html>)";
```
to
```
R"(<!DOCTYPE html><html><head><meta charset='utf-8'><meta name='viewport' content='width=device-width,initial-scale=1'/></head><body><form method='POST' action='' enctype='multipart/form-data'><input type='file' name='update'><input type='submit' value='Update'></form></body></html>)";
```
then when upgrade from mobile phone - the items will be of normal size

igrr, 2017-12-14:
Could you please open this as a Pull Request?

devyte, 2019-11-10:
The suggestion was merged a while back with #3732 .
Closing.

closed 2019-11-10 by devyte