esp8266/Arduino #3962: Binder function for encapsulated ESP8266 Web Server

rsegecin, 2017-12-14:
Hardware:			ESP8266-12F
Core Version:      	2.4.0-rc2

I'm trying to encapsulate a WebServer hosted on ESP8266 that function as Access Point as well. I'm using a binder to be able to reference the methods in my encapsulated class for the server targets, eg: if a client enters the root address for the website it runs a response code written in one of the methods in the encapsulated class. I've used this binder in many other occasions but it seems that the error that I'm getting points out to this binder that I'm sure it's correct. The full code is [here](https://github.com/rsegecin/MindTheGate), the relevant points are below:

Instance of the binder variable:

```
...
bind_member<WebServerAPClass, &WebServerAPClass::HandleRoot> handleRoot(this);
...
server->on("/", handleRoot);
...
``` 

The binder:
```
template<class T, void(T::*PTR)()>
struct bind_member
{
    typedef void(*fn_type)();
    explicit bind_member(T* _ptr)
    {
        ptr = _ptr;
    }
    static void func(void)
    {
        (ptr->*PTR)();
    }
    operator fn_type()
    {
        return &func;
    }
private:
    static T*  ptr;
};
template<class T, void(T::*PTR)()>
T* bind_member<T, PTR>::ptr = NULL;
```

The output:
```
Setting wifi AP mode ... Setting soft-AP configuration ... [APConfig] local_ip: 192.168.88.1 gateway: 192.168.88.1 subnet: 255.255.255.0
[APConfig] DHCP IP start: 192.168.88.100
[APConfig] DHCP IP end: 192.168.88.200
Ready
Setting soft-AP ... [AP] softap config unchanged
Ready
AP IP address: 192.168.88.1
...[stack]...
ets Jan  8 2013,rst cause:2, boot mode:(3,6)
```

The stack decoded:
```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 17 results
0x40206e65: std::function ::function(std::function  const&) at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2380
0x40206e76: std::function ::function(std::function  const&) at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2439
0x4020403e: ESP8266WebServer::on(String const&, HTTPMethod, std::function ) at C:\Users\Rinaldi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 214
0x40206a70: operator new(unsigned int) at C:\Users\Rinaldi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/abi.cpp line 84
0x40201ff0: std::_Function_base::_Base_manager   >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 1931
0x402021cc: bind_member ::func() at C:\Users\Rinaldi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\RTCTimer/SimpleBinders.h line 14
:  (inlined by) std::_Function_handler   >::_M_invoke(std::_Any_data const&) at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2071
0x40206e76: std::function ::function(std::function  const&) at C:\Users\Rinaldi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 214
0x40204085: ~function at C:\Users\Rinaldi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 214
:  (inlined by) ESP8266WebServer::on(String const&, std::function ) at C:\Users\Rinaldi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 233
0x40201ff0: std::_Function_base::_Base_manager   >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 1931
0x402021cc: bind_member ::func() at C:\Users\Rinaldi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\RTCTimer/SimpleBinders.h line 14
:  (inlined by) std::_Function_handler   >::_M_invoke(std::_Any_data const&) at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2071
0x4020215f: ~_Function_base at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2029
:  (inlined by) ~function at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2174
:  (inlined by) WebServerAPClass::Init() at C:\Users\Rinaldi\AppData\Local\Temp\arduino_build_415025\sketch/WebServerAP.cpp line 27
0x40201ff0: std::_Function_base::_Base_manager   >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 1931
0x402021cc: bind_member ::func() at C:\Users\Rinaldi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\RTCTimer/SimpleBinders.h line 14
:  (inlined by) std::_Function_handler   >::_M_invoke(std::_Any_data const&) at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2071
0x40205c00: HardwareSerial::flush() at C:\Users\Rinaldi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/HardwareSerial.cpp line 180
0x40201f91: setup at C:\Users\Rinaldi\Documents\Arduino\MindTheGate/MindTheGate.ino line 11
0x40206ba3: loop_wrapper at C:\Users\Rinaldi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/core_esp8266_main.cpp line 57
```

igrr, 2017-12-14:
Without the first part of the exception message (before the stack dump) it's pretty hard to say what is happening here. Looks like in invalid pointer is dereferenced somewhere.

It looks like [std::bind](http://en.cppreference.com/w/cpp/utility/functional/bind) should serve you purpose, and it is known to work. Would you mind trying that instead of a custom binder?

rsegecin, 2017-12-14:
I wasn't aware of std::bind I'll check that out. Mean while this's the all output:

```
Setting wifi AP mode ... Setting soft-AP configuration ... [APConfig] local_ip: 192.168.88.1 gateway: 192.168.88.1 subnet: 255.255.255.0
[APConfig] DHCP IP start: 192.168.88.100
[APConfig] DHCP IP end: 192.168.88.200
Ready
Setting soft-AP ... [AP] softap config unchanged
Ready
AP IP address: 192.168.88.1

Exception (28):
epc1=0x40206e65 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000007c depc=0x00000000
sp: 3fff06c0 end: 3fff0950 offset: 01a0
3fff0860:  0000000d 0000000b 00000000 40206e76  
3fff0870:  00000000 00000000 0000000d 4020403e  
3fff0880:  00000010 3fff08f0 00000000 40206a70  
3fff0890:  3fff1f6c 00000001 40201ff0 402021cc  
3fff08a0:  00000000 3ffef678 3fff08f0 40206e76  
3fff08b0:  00000000 00000000 3fff08f0 40204085  
3fff08c0:  3fff1f54 00ffffff 40201ff0 402021cc  
3fff08d0:  3ffef690 3ffef678 3ffef8dc 4020215f  
3fff08e0:  3fff1f3c 0158a8c0 40201ff0 402021cc  
3fff08f0:  3fff1f1c 0000000f 00000001 3ffe98a0  
3fff0900:  0158a8c0 3ffe98a0 0158a8c0 40205c00  
3fff0910:  0001c200 0000001c 3ffef8dc 3ffef924  
3fff0920:  3fffdad0 00000000 3ffef8dc 40201f91  
3fff0930:  feefeffe feefeffe 3ffef91c 40206ba3  
3fff0940:  feefeffe feefeffe 3ffef930 40100a20  

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v0c897c37
~ld

```

rsegecin, 2017-12-14:
I've just tested using std::bind and now I'm getting the error on the line `server->on(...)`, but now I'm more convinced that it's a cpp programming problem than of the ESP, unless there's an exception that I'm not aware of.  I've committed this change as well on the project mentioned above. 

Line altered:

```
std::function<void()> handleRoot = std::bind(&WebServerAPClass::HandleRoot, this);
...
server->on("/", handleRoot);
```

Here's the output:

```
Setting wifi AP mode ... Setting soft-AP configuration ... [APConfig] local_ip: 192.168.88.1 gateway: 192.168.88.1 subnet: 255.255.255.0
[APConfig] DHCP IP start: 192.168.88.100
[APConfig] DHCP IP end: 192.168.88.200
Ready
Setting soft-AP ... [AP] softap config unchanged
Ready
AP IP address: 192.168.88.1

Exception (28):
epc1=0x40206db1 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000007c depc=0x00000000
sp: 3fff0670 end: 3fff0910 offset: 01a0

...[stack decoded below]...

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v0c897c37
~ld
```

```
Decoding 18 results
0x40206dc2: std::function ::function(std::function  const&) at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2439
0x40204016: ESP8266WebServer::on(String const&, HTTPMethod, std::function ) at C:\Users\Rinaldi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 214
0x40206994: operator new(unsigned int) at C:\Users\Rinaldi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/abi.cpp line 84
0x40201fe8: std::_Function_base::_Base_manager    (WebServerAPClass*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 1931
0x40206d68: std::_Function_base::_Base_manager    (WebServerAPClass*)> >::_M_get_pointer(std::_Any_data const&) at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 1893
:  (inlined by) std::_Function_handler    (WebServerAPClass*)> >::_M_invoke(std::_Any_data const&) at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2071
0x40206dc2: std::function ::function(std::function  const&) at C:\Users\Rinaldi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 214
0x4020405d: ~function at C:\Users\Rinaldi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 214
:  (inlined by) ESP8266WebServer::on(String const&, std::function ) at C:\Users\Rinaldi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 233
0x40201fe8: std::_Function_base::_Base_manager    (WebServerAPClass*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 1931
0x40206d68: std::_Function_base::_Base_manager    (WebServerAPClass*)> >::_M_get_pointer(std::_Any_data const&) at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 1893
:  (inlined by) std::_Function_handler    (WebServerAPClass*)> >::_M_invoke(std::_Any_data const&) at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2071
0x40202185: ~function at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2174 (discriminator 3)
:  (inlined by) WebServerAPClass::Init() at C:\Users\Rinaldi\AppData\Local\Temp\arduino_build_782052\sketch/WebServerAP.cpp line 29 (discriminator 3)
0x40201fe8: std::_Function_base::_Base_manager    (WebServerAPClass*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 1931
0x40206d68: std::_Function_base::_Base_manager    (WebServerAPClass*)> >::_M_get_pointer(std::_Any_data const&) at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 1893
:  (inlined by) std::_Function_handler    (WebServerAPClass*)> >::_M_invoke(std::_Any_data const&) at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2071
0x40201fe8: std::_Function_base::_Base_manager    (WebServerAPClass*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\rinaldi\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\includ [... cut ...]
0x40205b00: HardwareSerial::flush() at C:\Users\Rinaldi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/HardwareSerial.cpp line 180
0x40201f51: setup at C:\Users\Rinaldi\Documents\Arduino\MindTheGate/MindTheGate.ino line 11
0x40206ac7: loop_wrapper at C:\Users\Rinaldi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/core_esp8266_main.cpp line 57
0x40100a20: cont_norm at C:\Users\Rinaldi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/cont.S line 109
```

devyte, 2017-12-14:
@rsegecin I agree about language. The std::bind is standard, so there are plenty of use examples out there. Also, take a look at C++ lambdas. I believe that the ::on() methods are pretty permissive about the callback argument.
Closing this, per issue policy doc.

closed 2017-12-14 by devyte

rsegecin, 2017-12-14:
Thank you @devyte,  thank you @igrr

igrr, 2017-12-14:
@rsegecin I'm probably missing something, but i can't find where you set `server` member for your WebServerAPClass...

rsegecin, 2017-12-14:
@igrr You're right I'm so ashamed. The stack pointer points to various lines that in majority are at the end of methods on closing curly braces, the only places that were not at the end was at the line "server.on" that's not wrong because it wasn't declared but as the stack was also displaying the binder method I thought the problem was the binder.  Now it works =). Thank you very much for taking your time and I'm sorry if anything.