esp8266/Arduino #3964: How to use ESP-01 Wi-Fi as fast as Bluetooth?

PeterPeterPan, 2017-12-14:
(my english is poor, I try my best to express my question.)

I want to use two ESP-01 to communicate as bluetooth. 
(one is AP another is STA, no need to connect to the same AP)

I try to use ESP-01 instead of arduino nano and HC-05.
But there are some problems.
I need to send data once by client in 20ms (or lower).
But it's working too slowly by ESP-01 Wi-Fi, from time to time will pause.
What I did wrong? What can I do?

Here is my code.
<STA>
https://gist.github.com/anonymous/8c2d070ed0c7dd20902c2e1b5910a06b
<AP>
https://gist.github.com/anonymous/d1639972a4e0d179d34164f96a1b4a18

devyte, 2017-12-14:
@PeterPeterPan no basic info, no IDE settinngs. You pretty much ignored the entire issuee template.
I'm going to assume that you used core 2.3.0. Try latest git, and build with lwip2 (menu option).

closed 2017-12-14 by devyte

PeterPeterPan, 2017-12-14:
I'm sorry...
and thank you very much.