esp8266/Arduino #3965: How to use ESP-01 Wi-Fi as fast as Bluetooth?

PeterPeterPan, 2017-12-14:
### Basic Infos
(my english is poor, I try my best to express my question.)

#### Hardware
Hardware:	ESP-01
Core Version:   2.3.0

### Description

I want to use two ESP-01 to communicate as bluetooth.
(one is AP another is STA, no need to connect to the same AP)

I try to use ESP-01 instead of arduino nano and HC-05.
But there are some problems.
I need to send data once by client in 20ms (or lower).
But it's working too slowly by ESP-01 Wi-Fi, from time to time will pause. (Seems to be random?)
What I did wrong? What can I do?

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512k
CPU Frequency:       80Mhz
Flash Mode:		dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

Here is my code.

STA code:
https://gist.github.com/anonymous/8c2d070ed0c7dd20902c2e1b5910a06b
AP code:
https://gist.github.com/anonymous/d1639972a4e0d179d34164f96a1b4a18

devyte, 2017-12-16:
@PeterPeterPan This is duplicate of #3964 . I already provided some advice there:
Try latest git, and build with lwip2 (menu option), as there are known issues with lwip1.4.
Again, this is not the right place to ask such questions. Please ask for assistance at a community forum.
Closing per issue policy doc.

closed 2017-12-16 by devyte

PeterPeterPan, 2017-12-17:
Thank you for your patient reply and suggestions.
I'm sorry I did not notice the nature of the discussion board.
thank!