esp8266/Arduino #3966: Large File transfer from sd card to webserver using server.streamFile(), taking too much time

Jagatheeswaran-Senthilvelan, 2017-12-14:
### Basic Infos

#### Hardware
Hardware:	        ESP8266 -12e
Core Version:      	?2.1.0-rc2?

### Description

Sending Large File from sd card to web server using the server.streamFile(), taking too much time. Example for a 1MB file it took more than 5 minutes of time, sometimes it crashes in between.

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			?qio
Flash Frequency:	80Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch
#include <SD.h>
#include <SPI.h>

#include <ESP8266WiFi.h> 
#include <ESP8266WebServer.h> 

const char* ssid = "---------";
const char* password = "----------";

File dataFile;

IPAddress ip(192, 168, 1, 200); // where xx is the desired IP Address
IPAddress gateway(192, 168, 1, 1); // set gateway to match your network
IPAddress subnet(255, 255, 255, 0); // set subnet mask to match your network

ESP8266WebServer server(80);

void handleRoot() 
{
  char temp[400];

  snprintf(temp, 400,
    "<html>\
  <head>\
  </head>\
    <body>\
    <h1>How to html page really good</h1>\
    <a href =\"/getdata\">getdata</a>\
  </body>\
</html>");
  server.send(200, "text/html", temp);

}

void getdata() {
    File dataFile = SD.open("datalog.txt");
    int fsizeDisk = dataFile.size();
    Serial.print("fsizeDisk: ");
    Serial.println(fsizeDisk);

    size_t fsizeSent = server.streamFile(dataFile,"text/plain");

    Serial.print("fsizeSent: ");
    Serial.println(fsizeSent);

    dataFile.close();

     delay(200);
  }

void setup(void) {

  Serial.begin(115200);
  Serial.print("Connecting to network");
  WiFi.mode(WIFI_STA);
  WiFi.config(ip, gateway, subnet);
  WiFi.begin(ssid, password);   // Connect to WiFi network

  while (WiFi.status() != WL_CONNECTED) {    // Wait for connection
    delay(500);
    Serial.print(".");
  }

  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  server.on("/", handleRoot);
  server.on("/getdata", getdata);
  
  server.begin();
  Serial.println("HTTP server started");
}

void loop(void)
{
  server.handleClient();
}

Jagatheeswaran-Senthilvelan, 2017-12-14:
Please suggest me some idea to transfer large files from sd card to web server. I want to transfer 10MB file,

d-a-v, 2017-12-14:
Are you using master ?

Jagatheeswaran-Senthilvelan, 2017-12-14:
You mean SD Card Master Library for ESP8266, Yes I'm using that only.

d-a-v, 2017-12-14:
I meant latest git version (=master) of esp8266/Arduino core ([doc](https://github.com/esp8266/Arduino#using-git-version))

Jagatheeswaran-Senthilvelan, 2017-12-14:
Yeah, I'm using that only. @d-a-v

d-a-v, 2017-12-14:
I don't have the hardware yet (sd on esp). I may try to simulate a 10MB ```File``` to use with ```::streamfile()``` and try to reproduce.

Can you describe the crash you sometimes have ?

Jagatheeswaran-Senthilvelan, 2017-12-14:
Sure, that would be a great help. Regarding crash, it stops sending data at a  certain point but that happens rarely not all the time. @d-a-v

d-a-v, 2017-12-14:
I tried with a character-based FakeStream, ~10.4MiB and it takes 69s @ 149KiB/s / 1.22Mibits/s.
Bandwidth is regular with lwip2, less regular with lwip1 (but a bit faster due to bigger MSS).
It does not crash and takes never more than 75s.

Can you try this sketch ?
```
//#include <SD.h>
//#include <SPI.h>

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

//#include <FakeStream.h>
#ifndef FAKESTREAM_H
#define FAKESTREAM_H

class FakeStream: public Stream
{
  protected:
    size_t fakesize;

  public:
    FakeStream (size_t size): fakesize(size) { }

    virtual int available() {
      return fakesize;
    }
    virtual int read() {
      if (fakesize) {
        --fakesize;
        return 0x5a;
      }
      return -1;
    }
    virtual int peek() {
      return 0x5a;
    }
    virtual void flush() {
      fakesize = 0;
    };
    virtual size_t write (uint8_t data) {
      (void)data;
      return 1;
    }
    size_t size () {
      return fakesize;
    }
    const char* name () {
      return "5a";
    }
};

#endif // FAKESTREAM_H

const char* ssid = "open";
const char* password = "";

ESP8266WebServer server(80);

void handleRoot()
{
  server.send(200, "text/html", "hello");
}

void getdata() {
  FakeStream dataFile(11000007);
  int fsizeDisk = dataFile.size();
  Serial.print("fsizeDisk: ");
  Serial.println(fsizeDisk);

  size_t fsizeSent = server.streamFile(dataFile, "text/plain");

  Serial.print("fsizeSent: ");
  Serial.println(fsizeSent);

  delay(200);
}

void setup(void) {

  Serial.begin(115200);
  Serial.print("Connecting to network");
  WiFi.mode(WIFI_STA);
  //WiFi.config(ip, gateway, subnet);
  WiFi.begin(ssid, password); // Connect to WiFi network

  while (WiFi.status() != WL_CONNECTED) { // Wait for connection
    delay(500);
    Serial.print(".");
  }

  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  server.on("/", handleRoot);
  server.on("/getdata", getdata);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void)
{
  server.handleClient();
}
```

earlephilhower, 2019-01-31:
@d-a-v provided a test and sketch over a year ago and there's been up response from original bug author.  Closing.

closed 2019-01-31 by earlephilhower

80avin, 2019-07-02:
You should also send the `Content-Length` header with value of file size. Without content-length header, the client keeps waiting even after the file has finished.
Further, you should also send a header for Caching the file if the file isn't expected to change content too frequently.
I'm not sure if it is the best solution, but it surely improves the speed.

```c++
void getdata() {
  File dataFile = SD.open("datalog.txt");
  int fsizeDisk = dataFile.size();
  Serial.print("fsizeDisk: ");
  Serial.println(fsizeDisk);

  server.sendHeader("Content-Length", (String)(fsizeDisk));
  server.sendHeader("Cache-Control", "max-age=2628000, public"); // cache for 30 days

  size_t fsizeSent = server.streamFile(dataFile, "text/plain");
  Serial.print("fsizeSent: ");
  Serial.println(fsizeSent);
  dataFile.close();
  delay(200);
}
```