esp8266/Arduino #3969: Arduino not displaying Library version correctly

dagnall53, 2017-12-14:
My arduino (ver 1.8.5)
displays: 
[image]
but I has 2.3 installed in my App directory..
is there something I need to do to get this to show the correct version number?

Pablo2048, 2017-12-14:
You have posted image from Library Manager, not the Boards Manager...

dagnall53, 2017-12-14:
Thanks... !
[image]

What is the difference-- if that is not a stupid question... I thought the Library showed code stuff and boards showed hardware.. 
So just what is the "1.0.0 Library".. and yes, nothing happens for me when I press more info.

devyte, 2017-12-16:
One is a library, the other is the entire core.
Closing.

closed 2017-12-16 by devyte