esp8266/Arduino #3970: dhcp hostname not set in SDK 2.1.0(deb1901)

dragondaud, 2017-12-14:
### Basic Infos

#### Hardware
Hardware:		ESP-12/NodeMCU v1.0
Core Version:      	2.4.0-github

### Description

When using github version of Arduino ESP8266, the dhcp client is not sending hostname in request, resulting in a blank name in the DHCP server, and no DNS entry created. If I recompile with 2.3.0, then the hostname is sent, and assigned as expected.

I noticed blank hostnames previously, but wasn't sure where they came from. It turns out that jumping between 2.3.0 and 2.4.0-github was masking the issue, as once the server has the name set, it maintains it thru renewals, even when the name is not specified subsequently. And something is causing my modules running 2.4.0-github to occasionally lose their lease (localIP() shows 0.0.0.0).

ALSO, the default generated hostname (ESP_xxxx) is invalid, as the underscore character is not allowed in a valid hostname. A valid hostname may only contain letters, numbers and the dash "-", though a valid hostname cannot begin or end with a dash. [https://en.wikipedia.org/wiki/Hostname](url)

### Settings in IDE

Arduino IDE:		1.8.5
Module:  			NodeMCU 1.0(ESP-12E)
Flash Size: 		4M(1M)
CPU Frequency:	80 Mhz
lwip:				v2
Upload Using:		Serial

### Sketch

```
#include <ESP8266WiFi.h>

#define HOST "test-hostname"
#define WIFI_SSID "SSID"
#define WIFI_PASS "PASS"

void setup() {
  Serial.begin(115200);
  while (!Serial);
  delay(100);
  Serial.println();
  Serial.print(F("setup: WiFi connecting to "));
  Serial.print(WIFI_SSID);
  Serial.print(F("..."));
  WiFi.hostname(HOST);
  WiFi.mode(WIFI_STA);
  WiFi.begin(WIFI_SSID, WIFI_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(F("."));
    delay(500);
  }
  Serial.println(F(" OK"));
  Serial.print(F("Last reset reason: "));
  Serial.println(ESP.getResetReason());
}

void loop() {
  String ip = WiFi.localIP().toString();
  int heap = ESP.getFreeHeap();
  Serial.printf("%s: %6d %6d \r\n", ip.c_str(), millis(), heap);
  delay(2000);
}
```

### Debug Messages

No related errors or other messages observed, even with full debug enabled.

d-a-v, 2017-12-14:
I think this is related to the dhcp client in lwip2.
Can you try with lwip1.4 (select in menu) and report if this is working with it ?

dragondaud, 2017-12-14:
Selecting "v1.4 Prebuilt(gcc)" does indeed fix THIS problem, but as you know I had other troubles using the old lwip in my sketches, specifically with my telnet server. Thanks for the quick response, let me know if you want me to test a fix or anything.

d-a-v, 2017-12-14:
Can you please try to update lwip2 like this:
```
cd tools/sdk/lwip2
make
cd builder
git checkout master
git pull origin master
cd ..
make install
```
then retry your sketch and check for hostname set by dhcp client ?

dragondaud, 2017-12-15:
apps/sntp/sntp.c:728:1: error: stray '\' in program
 \n\n\n#if LWIP_UDP\nvoid sntp_force_request
(void)\n{\n\tsntp_request(NULL);\n}\n#endif\n
 ^

dragondaud, 2017-12-15:
> 719-821-8900 <(719)%20821-8900>

after manually fixing the src error, it compiles and installs fine. And it
does fix the dhcp hostname issue. I'll let you know if I see anything else,
thanks!

d-a-v, 2017-12-26:
@devyte will be included in the next lwip2 update

dragondaud, 2018-01-15:
DHCP hostname stopped working, again. Did this change not make it into master? I recently sync'd, but am not sure exactly when it stopped working. I tried the instructions above, again, and it didn't help.

dragondaud, 2018-01-15:
Noticed that dhcp hostname works normally if I enable WiFi debugging in the build. Stops working again when I disable debugging.

d-a-v, 2018-01-15:
It is currently working with master and you sketch above, no debug option enabled.
Please double check then open a new issue (or reopen this one and update the title).

dragondaud, 2018-01-15:
I will try to isolate the problem further, but here are some packet captures that demonstrate the problem. The first one is successful, with debugging enabled, and you can see the requested hostname. The second capture, running the same exact code, but with debugging disabled, and it either doesn't have a hostname request, at all, or it is garbage as in this sample. Something is getting overwritten...

15:44:23.280937 60:01:94:34:52:20 > ff:ff:ff:ff:ff:ff, ethertype IPv4 (0x0800), length 350: (tos 0x0, ttl 255, id 1, offset 0, flags [none], proto UDP (17), length 336)
    0.0.0.0.68 > 255.255.255.255.67: [udp sum ok] BOOTP/DHCP, Request from 60:01:94:34:52:20, length 308, xid 0xfce0d625, Flags [none] (0x0000)
	  Client-Ethernet-Address 60:01:94:34:52:20
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: Request
	    MSZ Option 57, length 2: 1500
	    Requested-IP Option 50, length 4: 10.10.10.197
	    Server-ID Option 54, length 4: 10.10.10.1
	    Parameter-Request Option 55, length 5: 
	      Subnet-Mask, Default-Gateway, BR, Domain-Name-Server
	      NTP
	    Hostname Option 12, length 16: "TankGuard-345220"

15:50:36.939437 60:01:94:34:52:20 > ff:ff:ff:ff:ff:ff, ethertype IPv4 (0x0800), length 350: (tos 0x0, ttl 255, id 1, offset 0, flags [none], proto UDP (17), length 336)
    0.0.0.0.68 > 255.255.255.255.67: [udp sum ok] BOOTP/DHCP, Request from 60:01:94:34:52:20, length 308, xid 0x2ce43abb, Flags [none] (0x0000)
	  Client-Ethernet-Address 60:01:94:34:52:20
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: Request
	    MSZ Option 57, length 2: 1500
	    Requested-IP Option 50, length 4: 10.10.10.197
	    Server-ID Option 54, length 4: 10.10.10.1
	    Parameter-Request Option 55, length 5: 
	      Subnet-Mask, Default-Gateway, BR, Domain-Name-Server
	      NTP
	    Hostname Option 12, length 5: "G^A0^B^G"

d-a-v, 2018-01-15:
Thanks for this finding!

dragondaud, 2018-01-16:
FYI, switching to lwip1.4 makes the problem disappear.
Also of note, lwip2 is sending multiple dhcp requests every boot. Sometimes the first one has the correct hostname, but the second request overwrites it with garbage. I can make traces and upload if you need.

caipifrosch, 2018-02-04:
Thanks a lot for highlighting this issue here. Same experience at my end. 

I am quite happy with stability of my lwip v2 webinterface by now, but I am desperately  missing hostname feature. 

Any idea when this is going to be tackled?

d-a-v, 2018-02-04:
@caipifrosch @dragondaud So the first dhcp request is always good and the second one (after the delay defined by the dhcp server) can be (or is always) garbage ?

caipifrosch, 2018-02-04:
As dragondaud's description perfectly matched my experience, there were no further investigations into this at my end. 

Just tested again with lwip 1.4prebuild and V2prebuild MCC=536.... Hostname just wont work on lwipV2. V1.4 hostname seems to be fine, but mobile access with my Android7 is very slow, no matter what browser I use.

Nevertheless, I can **not** confirm that wifi debug option sets the hostname correctly.

d-a-v, 2018-02-04:
So I need help with a live patch to try. Can you
```
cd tools/sdk/lwip2
make download
```
edit ```builder/glue-lwip/lwip-git.c``` L313 and comment the line out:
```
//netif->hostname = wifi_station_get_hostname();
```
then recompile
```
make install
```
and relink your sketch (by rebuilding it).

Reason: ```netif (lwip2) ->hostname``` used by dhcp client is set twice, one coming from lwip1's netif and the other one above. This may be a mistake. I commented out the one from lwip1 and I had no more hostname in my dhcp requests, so the bug may be coming from this line above. Calling orders sometimes vary and I have not yet been able to control this since they are called from espressif binaries.

I have the DHCP hostname with the commented line above, do you ? @dragondaud @caipifrosch

caipifrosch, 2018-02-04:
OK, tried it without real background knowledge here... no success. Manually downloaded lwip2 from git as make install was not working. Make install then gave another error message.

Any idea how to get around this? A link to a guide how to compile lwip would be highly appreciated.

Error message:

C:\develop\Arduino\hardware\ESP8266com\ESP8266\tools\sdk\lwip2>make install
MAKE Version 5.41  Copyright (c) 1987, 2014 Embarcadero Technologies, Inc.
        make -C builder -f Makefile.arduino install
MAKE Version 5.41  Copyright (c) 1987, 2014 Embarcadero Technologies, Inc.
Incorrect command line argument: -C

d-a-v, 2018-02-05:
Can you replace your ```liblwip2.a``` and ```liblwip2_1460.a``` with these ([link](https://github.com/d-a-v/Arduino/tree/hostname/tools/sdk/lib)) and verify you get the right name in your dhcp server ?
(just select the file and click on the download button)

dragondaud, 2018-02-05:
@caipifrosch you would need to use GNU make and gcc to build, looks like you have a commercial tool in your path. If you install GNU Make, and put it in your path before Embarcadero's then it will work.

@d-a-v I'm testing the change now.

d-a-v, 2018-02-05:
@dragondaud I updated the test as follow.
(remote blind debugging...)
```
--- a/glue-lwip/lwip-git.c
+++ b/glue-lwip/lwip-git.c
@@ -166,6 +166,7 @@ err_glue_t esp2glue_dhcp_start (int netif_idx)
        netif_set_link_up(&netif_git[netif_idx]);
        //netif_set_up(&netif_git[netif_idx]); // unwanted call to netif_sta_status_callback()
        netif_git[netif_idx].flags |= NETIF_FLAG_UP;
+ netif_git[netif_idx].hostname = wifi_station_get_hostname();
        err_t err = dhcp_start(&netif_git[netif_idx]);
        uprint(DBG "new_dhcp_start returns %d\n", (int)err);
        return git2glue_err(err);
@@ -354,7 +355,7 @@ void esp2glue_netif_update (int netif_idx, uint32_t ip, uint32_t mask, uint32_t
        }
        
        netif->mtu = mtu;
-   netif->hostname = hostname;
+ netif->hostname = wifi_station_get_hostname(); (void)hostname;
        ip4_addr_t aip = { ip }, amask = { mask }, agw = { gw };
        netif_set_addr(&netif_git[netif_idx], &aip, &amask, &agw);
        esp2glue_netif_set_up1down0(netif_idx, 1);
```

caipifrosch, 2018-02-05:
My device can be accessed by hostname now. Thanks a lot! 

Webpages seem to load ok, but without any excessive testing my feeling is that it got somehow slower on android 7 device when using hostname. A website Logo of just 2kB does appear pretty late. Speed is good on android 7 IP access or win10 notebook.

Have there been any other changes in the stack?

@dragondaud: yes, I am indeed using a commercial tool here at my end for other stuff. Funny, when seeing this error I thought these make tool guays would use same toolchain... hahaha - Will have a look into the GNU make tool though...

dragondaud, 2018-02-05:
DHCPREQUEST for 10.10.10.195 (10.10.10.1) from 60:01:94:34:4f:0c (Hostname Unsuitable for Printing) via em1

Still seeing problem in lwip2 here.

d-a-v, 2018-02-05:
@caipifrosch thanks for reporting !
@dragondaud did you try the first proposal, or the second one ?

dragondaud, 2018-02-05:
I didn't see the second one until now, and I will try it now, too.

dragondaud, 2018-02-05:
Though, I will note that I still believe something in lwip2 is clobbering memory, because with the latest build my DHT22 sensor stopped working reliably, but going back to lwip1.4 and it works fine again.

d-a-v, 2018-02-05:
@dragondaud can you confirm the hostname problem is solved by the second proposal ?
If so, and about your DHT22 sensor, this is indeed weird. I have such sensor, so If you can provide a minimal sketch triggering the issue (by opening a new repo-issue), I will be able to have a look.

dragondaud, 2018-02-05:
Proposal 2 does fix the dhcp hostname issue, and the DHT22 seems to be happy, too. I'm betting something was referencing the non-initialized hostname and clobbering stuff. I'll close this issue once the fix is merged into master, or you can, whenever. Thanks!

d-a-v, 2018-02-05:
Thanks for testing !

closed 2018-02-07 by devyte

cloeffler, 2019-01-02:
I have just updated my esp libraries in Atom and now the hostname is missing from the DHCP. This stops my esp's from getting DHCP on my network. Im not shure what changed in the last update or two, but why would you take the hostname out of DHCP?

cloeffler, 2019-01-02:
Also is there a command to put the hostname back in , or is the only option to recompile the sdk?

RAVURIVENKAT, 2019-06-17:
WiFi.hostname(WiFi_hostname);

PR #4039 (2017-12-29): multi-mss menus for lwip2, remove lwip xcc variant, lwip2 readme, updates and fixes
lwip2 updates:
  > multi-mss makefile
  > forwardported espconn (no multicast yet)
  > restore max 3 ntp servers for configTime() coherency
  > unchain seldom chained pbufs
  > dns cache name length back to (256->48->) 128
  > use sntp_stop/start() when dhcp address got
  > fix netif's hostname glue-handling
  > forwardported ping from lwip1.4

fix #3970
fix maybe #3963
commit 370e75cb (2017-12-29): multi-mss menus for lwip2, remove lwip xcc variant, lwip2 readme, updates and fixes (#4039)

lwip2 updates:
  > multi-mss makefile
  > forwardported espconn (no multicast yet)
  > restore max 3 ntp servers for configTime() coherency
  > unchain seldom chained pbufs
  > dns cache name length back to (256->48->) 128
  > use sntp_stop/start() when dhcp address got
  > fix netif's hostname glue-handling
  > forwardported ping from lwip1.4
fix #3970
fix maybe #3963
PR #4299 (2018-02-07): lwip2: use only wifi_station_get_hostname() to set netif hostname
fix #3970
commit 0339bbb1 (2018-02-07): lwip2: use only wifi_station_get_hostname() to set netif hostname (#4299)

* lwip2: use only wifi_station_get_hostname() to set netif hostname
fix #3970

* lwip2 comments on hostname