esp8266/Arduino #3972: Login Page can be bypassed easily

AlvinChrist, 2017-12-15:
I found that this kind of login page can easily be bypassed.. Just by manipulating the cookie I can bypass the login form without entering any username / password.. Any fix for this? ( Currently this script will allow me to access root webpage if my ESPSESSIONID=1 , I just open my Console in Firefox and change my ESPSESSIONID from 0 to 1 , and voila.. I can access the root webpage :) )

dragondaud, 2017-12-15:
> this kind of login page

What login page? What script? This is the issue board for the ESP8266 Arduino core. The only reference to ESPSESSIONID in this repository is in an example sketch called SimpleAuthentification.ino, which is only an example, not a secure production system. There are plenty of resources elsewhere to learn more about programming, cookies, and security.

devyte, 2017-12-16:
@AlvinChrist no basic info, no IDE settings, no example sketch, vague description that talks about your browser instead of the code in this repo.
Closing due to lack of info.

closed 2017-12-16 by devyte