esp8266/Arduino #3975: connecting NodeMCU via USB to PC causes erratic mouse movements

ringooelbereth, 2017-12-15:
### Basic Infos

The NodeMCU was working fine. I might have killed it by shorting INPUT_PULLUP pins to ground and applying 5V to INPUT pins. 

Is it even possible that a damaged NodeMCU will cause mouse movements on a desktop PC?
When connecting it, it shows as a CH340 device on COM6 in the device manager (as before).

#### Hardware
Hardware:			?ESP-12f?
Core Version:      	?2.1.0-rc2?

NodeMCU Lolin V3 (ESP8266MOD)

### Description

When I connect my NodeMCU via USB cable to my windows 10 PC, my mouse starts moving wildly and buttons are clicked. I tried to observe the behaviour more closely but it is moving files on my desktop, opening them, starting programs, etc. so I had to unplug it quickly. It looks like the mouse is only moving right and down and there is a pattern in the movements and clicks.

This started with a frozen Arduino IDE. Shortly before this incident, I wrote a program that would receive input on two pins and had four interrupts to the raising and falling flanks of each io. The voltage that was applied was possibly 5V from a SDS011, I did not check because I read that the ESP8266 is 5V resistant.

The board was working before. This started after the Arduino IDE was frozen for a while and crashed when I was running my program for a while.

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4M 3M SPIFFS
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

Programmer:            AVRISP mkll

### Sketch

const byte p25_pin = D7; // CHOOSE YOUR PIN
volatile byte p25_rising_counter = 0;
volatile byte p25_falling_counter = 0;
int p25_numberOfInterrupts = 0;
volatile int p25_pwm_value = 0;
volatile int p25_prev_time = 0;

const byte p10_pin = D6;  // CHOOSE YOUR PIN
volatile byte p10_rising_counter = 0;
volatile byte p10_falling_counter = 0;
int p10_numberOfInterrupts = 0;
volatile int p10_pwm_value = 0;
volatile int p10_prev_time = 0;
 
void setup() {
 
  Serial.begin(115200);
  pinMode(p25_pin, INPUT);
  pinMode(p10_pin, INPUT);
  attachInterrupt(digitalPinToInterrupt(p25_pin), handle_p25_falling, FALLING);
  attachInterrupt(digitalPinToInterrupt(p25_pin), handle_p25_rising, RISING);
  attachInterrupt(digitalPinToInterrupt(p10_pin), handle_p10_falling, FALLING);
  attachInterrupt(digitalPinToInterrupt(p10_pin), handle_p10_rising, RISING);
 
}

void handle_p25_falling() {
  int now = micros();
  if (p25_rising_counter != p25_falling_counter){
    p25_pwm_value = p25_prev_time - now;
    p25_falling_counter++;
  }
  else{
    Serial.println("Error: P25 Falling-Interrupt detected where it should not be. \n  Debug Info:");
    Serial.println("  p25_rising_counter" + String(p25_rising_counter));
    Serial.println("  p25_falling_counter" + String(p25_falling_counter));
    Serial.println("  p25_pwm_value" + String(p25_pwm_value));
    Serial.println("  p25_prev_time" + String(p25_prev_time));
    Serial.println("  now" + String(now));
  }
}

void handle_p25_rising() {
  int prev = micros();
  if (p25_rising_counter == p25_falling_counter){
    p25_prev_time = prev;
    p25_rising_counter++;
  }
  else{
    Serial.println("Error: P25 Rising-Interrupt detected where it should not be. \n  Debug Info:");
    Serial.println("    p25_rising_counter" + String(p25_rising_counter));
    Serial.println("    p25_falling_counter" + String(p25_falling_counter));
    Serial.println("    p25_pwm_value" + String(p25_pwm_value));
    Serial.println("    p25_prev_time" + String(p25_prev_time));
    Serial.println("    now" + String(prev));
  }
  
}

void handle_p10_falling() {
  p10_falling_counter++;
}

void handle_p10_rising() {
  p10_rising_counter++;
}
 
void loop() {
  Serial.println("p25_pwm_value" + String(p25_pwm_value));
  Serial.println("    p25_rising_counter" + String(p25_rising_counter));
  Serial.println("    p25_falling_counter" + String(p25_falling_counter));
  Serial.println("    p25_prev_time" + String(p25_prev_time));
}

### Debug Messages
.....

Error: P25 Rising-Interrupt detected where it should not be. 
  Debug Info:
    p25_rising_counter1
    p25_falling_counter0
    p25_pwm_value0
    p25_prev_time123456
    now123456789
p25_pwm_value0
    p25_rising_counter1
    p25_falling_counter0
    p25_prev_time123456

this message was flooding the console because obviously there is no delay in the loop()

dragondaud, 2017-12-15:
Your issue has nothing to do with ESP8266 Arduino, and is a hardware issue more appropriate for other websites, not this project's Issue board. Try http://www.esp8266.com/ for general assistance. That said, Windows has been known to recognize any activity on a serial port as a mouse, because people used to use serial mice. You can disable this behavior with the following command from an admin cmd window:
REG ADD "HKLM\SYSTEM\CurrentControlSet\Services\sermouse" /V Start /T REG_DWORD /F /D 4
Then reboot.

devyte, 2017-12-15:
It's also possible that you have a virus, or a driver issue.
Closing.

closed 2017-12-15 by devyte

lrmoreno007, 2017-12-15:
I solved that problem by installing updated drivers. But it happened to me in an ESP32 with chip CP2102.

kapollo, 2018-11-15:
Should google send a lost soul here, I just had the same problem @dragondaud was right I overflowed the serial and couldn't do anything until I disabled the serial mouse in windows (10 x64) including reflashing.
cheers

gmdimitriz, 2019-01-18:
i got the same issue. its not a virus. the problem stops if i press "reset" button of the esp8266