esp8266/Arduino #3977: Increase I2C clock stretching value?

icucode, 2017-12-15:
### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Slow sensors doesn't work with default clock stretching value. Users or library authors have to adjust by calling e.g. Wire.setClockStretchLimit( 500L ); on ESP8266 for it to work. Can the default clock stretching (230 uS) be increased to make the code more user friendly?  

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

devyte, 2017-12-16:
@icucode Only some devices require clockstretching, and of those only the really slow ones would make use of this. The whole purpose of Wire.setClockStretchLimit() is for the user to call it with the value required by the specs of the device being used.
In other words, if you have a slow sensor, you are _supposed_ to call this. Doing so makes the handling of the device slowness explicit.
Closing as won't fix. If a reason better than skipping a simple 1-line function call is presented, I'll reopen.

closed 2017-12-16 by devyte

icucode, 2017-12-16:
I understand that the explicit part can be good to have to protect the I2C bus from slow devices. But the I2C specification doesn't explicitly have a timeout for clock stretching:

"4.2.2 ... I2C can be a ‘DC’ bus, meaning that a slave device stretches the master clock when
performing some routine while the master is accessing it. This notifies the master that the
slave is busy but does not want to lose the communication. The slave device will allow
continuation after its task is complete. There is no limit in the I2C-bus protocol as to how
long this delay can be."

See:https://www.nxp.com/docs/en/user-guide/UM10204.pdf

I may also have been unlucky with my I2C device since it does not explicitly tell in the datasheet how slow it is. Instead I have had to rely on time consuming trial and error to locate the problem since I don't own an oscilloscope.

Egyras, 2018-03-17:
Hi, I have SDP6xx-500Pa and min/max readings only -136Pa/136Pa with scale factor 60. Was checking on NodeMCU ESP8266. Any ideas what can be wrong ?