esp8266/Arduino #3978: boards.txt DebugLevel should have NDEBUG option

earlephilhower, 2017-12-15:
An option to use "-DNDEBUG" should be added to all boards.txt debug levels to disable ASSERT macros (and save the string memory they use).

I sent in a merged pull request that takes the constant string of the filename in the assert() macro and moves them to PROGMEM.  Unfortunately, the same can not be done with the function name part of the macro, __PRETTY_FUNCTION__, because it is defined as a char* by GCC and added to the function locals a "static const char[]="xxxx";"  In a nutshell, you can't convert that to PROGMEM w/o changing GCC internals.  (See https://stackoverflow.com/questions/4384765/whats-the-difference-between-pretty-function-function-func for more info)

In some cases, like ESP8266WiFiClient, those __PRETTY_FUNCTION__ strings can take *100s* of bytes due to C++ template expansion.  My ESP8266 WebRadio sketch has *~500 bytes* of template expansion strings stuck in RAM in .rodata because of this.  That's a significant amount of total usable RAM for me and others pushing the ESP8266's capabilities.

In boards.txt the change would look like:
+  *.menu.DebugLevel.NDEBUG=NDEBUG

@d-a-v - I'd submit a patch myself, but I see some stuff about generating boards.txt from a python script in the file, but the referenced generator doesn't seem to be there.  If there's some other repo I should look at and submit a PR, just point me in the right direction.

Thx
-EFP3

d-a-v, 2017-12-16:
The python script is in #3722.
Since it is not yet merged (will happen after 2.4), and since current ```boards.txt``` was however generated from this script, and since the script was not perfect at that time, I sometimes make fixing PRs with only ```boards.txt``` for 2.4.
Current one is #3974 and includes your suggestion. It indeed saved 2Kflash and 350Bram on the currently random opened sketch I have.

earlephilhower, 2017-12-16:
Thanks for the super-fast response!

commit 355232ff (2017-12-18): boards.txt: missing -DESP8266 + debug menu for -DNDEBUG (#3974)

* boards.txt: add missing -DESP8266 - fix #3973

* + debug menu NoAssert-NDEBUG => -DNDEBUG
fix #3978 and saves ram+flash
on behalf of @earlephilhower
closed 2017-12-16 by earlephilhower