esp8266/Arduino #3979: ESP8266 - High power consumption during deep sleep ( ESP.deepSleep(0) ). 200uA instead of 15~20uA

immajor, 2017-12-15:
Hi there!

I was looking for a topic like this, but I only found one at [nodemcu ](https://github.com/nodemcu/nodemcu-firmware/issues/1625)and found none at Arduino. If there is  any, please point me to that direction! :)

### Basic Infos

Some times when I call `ESP.deepSleep(0);`, the board is consuming 200uA instad of 20uA.

#### Hardware
Hardware:			ESP-12-E
Core Version:      	unknown

### Description

When I call `ESP.deepSleep(0);` and I measure the current consumed my ESP8266, I get 2 different values. In some cases (70% of the time),  the current consumed is 17~19uA, so everything is fine! If I unplug the +3.3V and plug it back, it stays in deepSleep mode, the current goes back to 17~19uA.

But in some other cases I measure ~200uA current consumed by it. And if I unplug the +3.3V and plug it back, it restarts.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Flash Mode:		qio0 - high
Flash Frequency:	256000
Upload Using:		SERIAL
Reset Method:		---

### Sketch

```cpp

#include <Arduino.h>

void setup() {
//simply connecting to WiFi
//sending an UDP packet
}

void loop() {
      //receiving an UDP packet
      analogWrite(LEDpin,1024);
      delay ( 1000 );
      analogWrite(LEDpin,0);
      delay ( 25 );
      
      **ESP.deepSleep(0);**

}
```

### Debug Messages
none

immajor, 2017-12-15:
Is there something I can do to avoid the higher power consumption?
If you need I can send some photos or a video about the issue.
Thank you in advance for your reply! :)

immajor, 2017-12-16:
Hi there!

I found the source of the problem :)

When there was no other load on the voltage source, only the ESP and it went to deepSleep, then the current went down, so the input voltage went up and this increased 3.9V input voltage caused the high deep sleep current.

closed 2017-12-16 by immajor

immajor, 2018-02-14:
Hi,

I did nothing, as the voltage will go lower, the power consumption will normalize. :) But this is a waist of energy and most likely not helping the ESP either. But lithium battery should not be charged above 4.0V and the biggest capacity of the battery is around 3.7-3.3V where the power consumption is not so off scale.

But I will use a HT7333 linear regulator in the future. This is a super regulator, based on the specs. In most cases it should help allot and it is quite cheap also (if the ebay and aliexpress sellers are sending legit parts). Check these for more details:
https://www.youtube.com/watch?v=IYuYTfO6iOs
https://www.youtube.com/watch?v=J66_8P043ko

for ultra long battery life, ceramic capacitors are better because electrolytic capacitor van eat some uA while storing voltage. (if im right)

immajor, 2018-02-15:
@Pietfro 
Or use LIFEPO4 batteries, they need no voltage regulating because they are ~3.3V until they are discharged. But they need special charging and discharge protection circuit.

LIFEPO4 discharge is like this:
http://lygte-info.dk/review/batteries2012/Soshine%2018650%20LiFePO4%201800mAh%20(Black)%20UK.html

Regular lithium batteries are like this:
http://lygte-info.dk/review/batteries2012/Samsung%20ICR18650-26F%202600mAh%20(Pink)%20UK.html

Pietfro, 2018-02-15:
Thank you @immajor for such a thorough answer. Currently I am using mcp1700t-3302 and it seems to make the job. However, I had to use a bigger capacitor to cope with higher current during ESP8266 startup.
I have to play with the input voltage...:)

immajor, 2018-02-16:
As I see mcp1700 has a bit better specs then HT7333, but also bit more expensive. This is also a good choice!

Gadam123, 2018-03-03:
Yeah, i suffer the same issue. I have 2 x ESP8266 both with MCP1700. One consumes 15-20uA the other 180uA in DeepSleep. I doubt this is an issue with overvoltage. I would say its rather luck that you got it lower. I suspect a faulty ESP rather. Nevetheless i am still digging...

WJR1989, 2018-03-19:
Any updates on why there is such a disparity between the performance of two units?

5chufti, 2018-08-24:
NodeMCU is very bad example when need for extra low power.
The 3.3V regulator usually is not low current and the usb-serial chip also consumes power during deepsleep of esp8266. 2mA is not too bad, have seen worse!

jdelrio810, 2018-08-25:
thanks for your answer

Gadam123, 2018-08-26:
i had a faulty ESP. I detached the top metal cover to look for any issues but couldnt find any. Maybe a capacitor that was leaking, maybe flash...donno. Besides the high current consumption ESP worked well.
Anyway when i replaced the ESP to other the deep sleep current went down do 20-30uA.