esp8266/Arduino #3980: Is the boardmanager referenced 2.3.0 Zip file correctly updated?

dagnall53, 2017-12-16:
I just re-updated to 2.3, but it looks like my libraries are out of date compared to those showing on the github. 

The Json `http://arduino.esp8266.com/versions/2.3.0/package_esp8266com_index.json`  calls up 
` "category": "ESP8266", 
          "name": "esp8266", 
          "url": "https://github.com/esp8266/Arduino/releases/download/2.3.0/esp8266-2.3.0.zip", 
          "checksum": "SHA-256:884188a54bc75f1f2e7f827619f8cb1980dc3f430f92f9b07186ed7f4db8c738", 
          "help": {
            "online": "http://esp8266.github.com/Arduino/versions/2.3.0/"
          }, 
          "version": "2.3.0", 
          "architecture": "esp8266", 
          "archiveFileName": "esp8266-2.3.0.zip", `

Checking the files I now have in my Libraries shows that they are out of date compared to those on  the Github.. it looks like they are 2.2 versions, dated 2016. 
Looking at the zip suggests it was not yet updated with the latest 2.3.0?

devyte, 2017-12-16:
Sounds like you have a bad installation. I suggest a clean install from scratch, of both the IDE and the core. That includes deleting the relevant dirs. Don't forget to backup your code beforehand.
You can't have both the board manager installation and the git installation simultaneously. I suggest sticking to the git.
Closing as can't reproduce.

closed 2017-12-16 by devyte

dagnall53, 2017-12-16:
tried a clean install just before you responded, still inspection shows library content is 2016, not 2017.. 
I deleted the whole program(86) arduino contents and the app data folder...
(on my computer this was C:\Users\Richard\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries

It has now been repopulated via boards manager, but the files seem out of date compared to the github. 

Example, on the github, 
ArduinoOTA/arduinoOTA.cpp was updated 25 days ago, 
but the version I have just had installed by board manager is dated 21/06/2016
Looking at the json and downloading 2.3.0zip suggests that the zip is not fully up to date..
 
Is the github showing what will be 2.4.0 ?

devyte, 2017-12-16:
2.3.0 is over a year old. We're slowly moving towards 2.4.0. Latest git has the most recent.