esp8266/Arduino #3982: please add to boards.txt DTR for Arduino 'Uno WiFi'

jandrassy, 2017-12-17:
On Uno WiFi the on-board esp8266 is accessible only thru Atmega. The new Serial1 library for Uno WiFi, enables uploading esp sketches from IDE into esp. And it can bridge the DTR signal to reset esp into flashing mode. The original Uno WiFi library from Arduino.org was to slow to bridge flashing sync and didn't use the on-board connections to CH_EN and GPIO0 for reset esp to flashing mode.

In boards.txt for the 'Arduino' group of boards the disableDTR and disableRTS is true. I propose adding two lines for Uno WiFi to enable DTR and RTS.
```
arduino-esp8266.menu.BoardModel.unowifideved.serial.disableDTR=false
arduino-esp8266.menu.BoardModel.unowifideved.serial.disableRTS=false
```

d-a-v, 2018-09-04:
Would you propose a PR adding these two lines in the board generator ?

jandrassy, 2018-10-07:
after a year? I don't use Uno WiFi anymore. I would need to setup everything. And I am sure other Uno WiFi owners who use this package to program the on-board esp8266 are used to the workaround after 3 core versions

closed 2018-10-07 by jandrassy