esp8266/Arduino #3987: Connecting issue

Atexor, 2017-12-17:
### Basic Infos

#### Hardware
Hardware:			ESP 12E on devkit module (nodemcu)
Core Version:      	latest stable - 2.3.0

### Description

Hi, I've problems with connecting to my WiFi. It takes infinite time without success. Only one time (flash) it succeed. I've entered my SSID & password corrently and I tried with other WiF device (access point on my phone).
I had flashed latest ESP8266 NONOS SDK V2.1.0 20170505, hovewer I'm not sure if it's overwritten by Arduino IDE while flashing.

### Settings in IDE

Module:  			NodeMCU V1.0 (ESP12-E Module)4
Flash Size: 		4MB (3M SPIFFS)
CPU Frequency:		80Mhz
Flash Mode:			?
Flash Frequency:	?
Upload Using:		SERIAL (just USB)
Reset Method:		nodemcu

### Sketch

```
#include <ESP8266WiFi.h>

#define LED_RED D0

const char* ssid = "Atex";
const char* password = "strongpassword";

WiFiClient client;

byte led_status = 0;

void setup()
{
  Serial.begin(115200);
  pinMode(LED_RED, OUTPUT);
  //yieldDelay(1000);
  delay(1000);
  
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print("Connecting to WiFi: ");
    Serial.println(ssid);
    WiFi.begin(ssid, password);
    delay(2000);
    //yieldDelay(2000);
  }
  Serial.println("Connected!");  
}

/*
void yieldDelay(unsigned long fTime)
{
  unsigned long endTime = millis() + fTime;
  while (millis() < endTime) {
     yield();
  }
}*/

void loop()
{

  //yieldDelay(500);
  delay(500);
  if(led_status == 0)
  {
    led_status = 1;
    digitalWrite(LED_RED, LOW);
  }
  else
  {
    led_status = 0;
    digitalWrite(LED_RED, HIGH);
  }
}
```

### Debug Messages
Just every 2 secs:
```
⸮Connecting to WiFi: Atexor
Connecting to WiFi: Atexor
Connecting to WiFi: Atexor
Connecting to WiFi: Atexor

```
phone:
```
Connecting to WiFi: Atex
Connecting to WiFi: Atex
Connecting to WiFi: Atex
```

devyte, 2017-12-17:
@Atexor you're calling WiFi.begin() inside the while loop. You need to call it only once, then react once a connection has been established.
I suggest using the WiFi event system instead of checking with a while loop.
Closing as user error.

closed 2017-12-17 by devyte

Atexor, 2017-12-17:
I found one example where there was that "solution". I didn't expect it was wrong and hot it worked for its author. Thank you very much.
Now it connects.