esp8266/Arduino #3991: get.py fails for 2.3.0

Trospector, 2017-12-18:
Trying to manually install 2.3.0 on Lubuntu 16.10. 

'python get.py' fails because one of the files to be downloaded is a .tgz file while the others are all .atar.gz files. Looked into get.py, it has no support for .tgz files, only for .tar.gz and .zip files (in the unpack() routine). Pulling my hair out ..... just way too many issues with esp8266, I think I am beginning to realize what the "cheap" attribute of esp8266 means -- you pay cheap upfront and then later you pay with precious your time debugging all the issues surrounding it!

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

Trospector, 2017-12-18:
oh btw get.py also fails complaining it cannot handle https URLs (on Lubuntu 16.10). Wonder whether this was tested at all before being put out..

vdeconinck, 2017-12-18:
This is a volunteer project and there are many known issues that need addressing by a small team, so fully testing each release against each OS and variant is not possible. If you have time to submit improvements to the manual install scripts, that is very welcome.
That being said, I'm not sure why you're going the hard - manual - route to install such an old version. 2.3.0 can all be installed using the IDE's board manager - https://github.com/esp8266/Arduino#installing-with-boards-manager
FYI, 2.3.0 has been superceded by 2.4.0-rc1 and 2.4.0-rc2 - see https://github.com/esp8266/Arduino/releases

vdeconinck, 2017-12-18:
By the way, although having one url ending in .tgz while all others end in .tar.gz is a bit weird, I cannot see how it could be a problem because unpack() is passed "local_path", which is based upon the "archiveFileName" entry of the json file, not the "url" entry, and all "archiveFileName" values end in .tar.gz ...
@Trospector Regarding https, can you check which version of Python is installed on your system ? It seems urllib support https since v2.7.9 - see https://docs.python.org/2/library/urllib.html

devyte, 2017-12-21:
@Trospector core 2.3.0 is very old. Doing a manual install, and on a Linux variant that is atypical, has a lot of implications to consider. Given that latest git install on Ubuntu works without issues, I suspect you have a problem specific to your environment. 
While we do try to have widespread compatibility, we can't handle testing and assuring all variants. If you think there is some fix possible that will make it work on your system, without disrupting other users, please open a new issue or PR and include details, so that it can be considered for merging.
If you need help getting the install to work on your system, please refer to a community forum.
Closing.

herrold, 2017-12-21:
@devyte 

The top page at GH shows 2.3.0 as the most recent release today

https://github.com/esp8266/Arduino

Perhaps there is a problem with the release process, that this is not getting updated ... having something like:
   stable
   LTS
and
   bleeding
cuts comes to mind

devyte, 2017-12-21:
@herrold 2.3.0 was the last release, but that doesn't reduce its age. There hasn't been a release since because the resources haven't been available. And we're not a commercial support repo, but rather a repo based on volunteers who freely contribute from our free time, so there is no "LTS" or anything like that.
The point of my  reason for closing is that it doesn't make sense to use the few developer resources currently available to support a manual install procedure on an atypical platform for an old version of the code.
It's a free repo. If you need this, you're welcome to fix it and contribute your solution.

Trospector, 2017-12-23:
Sorry if I sounded harsh in the first post. But, here is what I was dealing with, which I have now fixed for myself manually:
- needed to upgrade my ESP01 NONOS_SDK from 1.5.3 to 2.3.0 (or atleast 2.0+)
  -- couldn't figure out how to do that via Arduino IDE 1.8.3 even after looking around quite a bit
  -- decided to use esptool.py instead
     -- wasted a bunch of time on this because on my Lubuntu 16.10, "sudo apt-get install esptool" gets me the __other__ tool named "esptool", and for a good amount of time I had no clue there were two esptools. (of course the -h option and the documentation should have been give aways..)
     -- managed to finally use esptool.py to upgrade esp8266 to NONOS_SDK_2.3.0
  
  -- Next, I wanted to flash my user program to the upgraded esp8266, but the program still reported the old SDK! I figured I need to upgrade the ESP8266 libraries in my Arduino IDE 1.8.3.
     -- checked the Board Manager in the IDE, but it said esp8266 version 2.3.0 was already installed!
     -- so I decided to try the manual install route, to .../.arduino15/packages/esp8266/hardware/esp8266/tools
     -- that is when I ran into trouble with the https and the get.py, with both Python 2.7 and Python 3.5
     -- finally, from the error messages of get.py, figured out what to download/extract/rename, used wget to manually download the required tools.

There you go...!

igrr, 2017-12-23:
Just to understand the issue better, which articles/resources gave you the impression that ESP8266 itself can be "updated"? I.e. that there is some code running on the chip which is not part of the sketch flashed via Arduino IDE? Thanks.

devyte, 2017-12-23:
@Trospector omg you seem to have a severe misunderstanding of how things work. As a one time thing, I'm going to take the time to explain a bit.
The NONOS_SDK is a lib distributed by Espressif. It is distributed precompiled in binary form, and we have no access to to changing it. Latest version is 2.1.0, plus some fixes.
The core in this repo is built on top of an adaptation of the above, so when you install this core from the board manager or via git, you get both the sdk and our code. Latest release is 2.3.0,  which is old, so you should use git instead. We're working towardsa 2.4.0 release.
Your app is built on top of this core. When you build it, it links the core and the sdk together with your user app to create the binary file that gets loaded into the ESP.
You can't update just the nonos sdk on the esp. You have to rebuild your app with a different version of this core that contains what you want.
If you have Arduino IDE on ubuntu, installing 2.3.0 via board manager is documented, and installing latest git via command line is also documented.

Trospector, 2017-12-24:
@igrr and @devyte: indeed I am likely missing the bigger picture but here is how I got my limited esp8266 world view:

From the Espressif document "ESP8266 SDK Getting Started Guide", which shows the Flash Maps and how to download a particular SDK onto the ESP8266, I got the impression I could upgrade the SDK directly on the chip. Especially because of the RTOS SDK option, I assumed an RTOS should reside on the chip independent of any app, right? So I assumed the NONOS_SDK also resided on chip. The remaining question was: how does the app link to the SDK? I assumed the compiler did it somehow..
 
I did figure out later (after flashing the SDK to the chip via esptool.py) that the Arduino IDE is not just linking to an on-chip library but is compiling and flashing everything (the user app and the SDK ..) to the chip each time. This I figured out based on my app reporting SDK 1.5.3 though I had previously flashed SDK 2.1.0 to the chip, and based on the long compile time for a simple sketch in the IDE.

The part that was really not clear (maybe I didn't look in the right places) is that the Arduino IDE esp8266 version 2.3.0 is old, and that if I need to use Espressif's SDK 2.1.0 then it is not sufficient to use the IDE esp8266 version 2.3.0. I mistook that to be a bug in the IDE's esp8266 support and then went the git route.. (which happened to work for me, so far..)

BTW I needed to use SDK 2.1.0 because I wanted to try out  wifi_enable_gpio_wakeup(), which I don't think is part of SDK 1.5.3.

@devyte: thanks for your explanation. Could you point me to a doc that makes all this clear? Thanks.

igrr, 2017-12-26:
Just tried this on Ubuntu 16.04:

```
root@tmp:/build/Arduino# cd tools
root@tmp:/build/Arduino/tools# python get.py
Platform: x86_64-pc-linux-gnu
Downloading esptool-0.4.12-linux64.tar.gz
Done
Extracting dist/esptool-0.4.12-linux64.tar.gz
Renaming esptool-0.4.12-linux64 to esptool
Downloading linux64-xtensa-lx106-elf-gb404fb9.tar.gz
Done
Extracting dist/linux64-xtensa-lx106-elf-gb404fb9.tar.gz
Downloading mkspiffs-0.2.0-no_magic_length-linux64.tar.gz
Done
Extracting dist/mkspiffs-0.2.0-no_magic_length-linux64.tar.gz
Renaming mkspiffs-0.2.0-no_magic_length-linux64 to mkspiffs
root@tmp:/build/Arduino/tools# cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.3 LTS"
root@tmp:/build/Arduino/tools# python --version
Python 2.7.12
```
Now with python3:
```
root@tmp:/build/Arduino/tools# rm -rf dist/ esptool/ mkspiffs/ xtensa*
root@tmp:/build/Arduino/tools# python3 get.py
Platform: x86_64-pc-linux-gnu
Downloading esptool-0.4.12-linux64.tar.gz
Done
Extracting dist/esptool-0.4.12-linux64.tar.gz
Renaming esptool-0.4.12-linux64 to esptool
Downloading linux64-xtensa-lx106-elf-gb404fb9.tar.gz
Done
Extracting dist/linux64-xtensa-lx106-elf-gb404fb9.tar.gz
Downloading mkspiffs-0.2.0-no_magic_length-linux64.tar.gz
Done
Extracting dist/mkspiffs-0.2.0-no_magic_length-linux64.tar.gz
Renaming mkspiffs-0.2.0-no_magic_length-linux64 to mkspiffs
root@tmp:/build/Arduino/tools# python3 --version
Python 3.5.2
```

So it seems to be working...

closed 2017-12-26 by igrr