esp8266/Arduino #3993: Wifi hangs if no DHCP Server is available

manholl, 2017-12-18:
ESP12F
SDK Version 2.3

I use Wifi.begin(SSID, key) to connect to my WLAN - if the DHCP-Server is available- everythink works ok and the "onStationModeGotIP" Event is fired - see last line..
```
scandone
.f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with TML4, channel 1
dhcp client start...
ip:192.168.0.112,mask:255.255.255.0,gw:192.168.0.1
12334 DEBUG  | GOT IP!!!!: 192.168.0.112
```

But if the DHCP Server is not available the Programm hangs until the Server is available again.
I tried to use the onStationModeDHCPTimeout but it never gets fired..
```
scandone
.f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with TML4, channel 1
dhcp client start...
pm open,type:2 0
```

I have expected that after some time the "onStationModeDHCPTimeout" Event is fired, or i get a wifi.status so that i can handle that i have no IP. But now my Program hangs - this is a big problem for me, because my ESP is battery powered.. and will not enter deep sleep mode.

devyte, 2017-12-18:
@hotchpotch please retest with latest git, and build with lwip2 (option in menu).

manholl, 2017-12-19:
Hello again..
Now i checked it with the latest git build, and after some debugging i found that not Wifi.begin is the problem. 
The problem is that i also use WiFiMulti 

ESP8266WiFiMulti.cpp Line 112

> WiFi.begin(bestNetwork.ssid, bestNetwork.passphrase, bestChannel, bestBSSID);
>                 status = WiFi.status();
> 
>                 // wait for connection or fail
>                 while(status != WL_CONNECTED && status != WL_NO_SSID_AVAIL && status != WL_CONNECT_FAILED) {
>                     delay(10);
>                     status = WiFi.status();
>                 }

The while loop hangs here because "status" is WL_DISCONNECTED until a IP Adress is assigned...
WIFI.status() calls wifi_station_get_connect_status() and this function retuns "CONNECTING" but  WIFI.status() returns "WL_DISCONNECTED"  (switch - default)

devyte, 2018-01-03:
WiFiMulti i meant for networks with a working dhcp server. I think implementing a timeout for that while loop should be enough.

devyte, 2018-01-12:
@manholl @gavanfantom @denisn73 please test PR #4146 . It implements a timeout in the WiFi connection loop inside WiFiMulti. It addresses cases of no DHCP, wrong password, etc.
Most importantly, I have hardcoded a timeout of 5s.

closed 2018-01-20 by devyte

dachshund-digital, 2019-03-08:
Anyone have a code example of how to use the following handler?  onStationModeDHCPTimeout?  I can't seem to find any examples on this specific handler, illustrating its use.

PR #4146 (2018-01-20): Add timeout to WiFi connection loop in WiFiMulti
Fixes #3993 .