esp8266/Arduino #3998: ESP8266 disconnects after a while

elbrodark, 2017-12-19:
Hi greetings from germany (and my poor english...),
i´m using my Arduino Nano with a ESP8266.
amazon link: [AMAZON LINK](https://www.amazon.de/gp/product/B01L1PH1PE/ref=oh_aui_search_detailpage?ie=UTF8&psc=1)

I use it as a Garagedoor opener, but after i used it, it do not function anymore, I must power it off/on and then it works again. I´m a real noob at programming and i dont get it why it does not function.

here "my" code:
```
/*
 *  This sketch demonstrates how to set up a simple HTTP-like server.
 *  The server will set a GPIO pin depending on the request
 *    http://server_ip/gpio/0 will set the GPIO2 low,
 *    http://server_ip/gpio/1 will set the GPIO2 high
 *  server_ip is the IP address of the ESP8266 module, will be 
 *  printed to Serial when the module is connected.
 */

#include <ESP8266WiFi.h>

const char* ssid = "LXXXX";
const char* password = "BXXXXXXXXXX";

// Create an instance of the server
// specify the port to listen on as an argument
WiFiServer server(80);

void setup() {
  Serial.begin(115200);
  delay(10);

  // prepare GPIO2
  pinMode(2, OUTPUT);
  digitalWrite(2, 1);
  
  // Connect to WiFi network
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  
  // Start the server
  server.begin();
  Serial.println("Server started");

  // Print the IP address
  Serial.println(WiFi.localIP());
}

void loop() {
  // Check if a client has connected
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
  
  // Wait until the client sends some data
  Serial.println("new client");
  while(!client.available()){
    delay(1);
  }
  
  // Read the first line of the request
  String req = client.readStringUntil('\r');
  Serial.println(req);
  client.flush();
  
  // Match the request
  int val;
  if (req.indexOf("/gpio/0") != -1)
    val = 0;
  else if (req.indexOf("/gpio/1") != -1)
    val = 1;
  else {
    Serial.println("invalid request");
    client.stop();
    return;
  }

  // Set GPIO2 according to the request
  digitalWrite(2, val);
  
  client.flush();

  // Prepare the response
  String s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n<!DOCTYPE HTML>\r\n<html>\r\nGPIO is now ";
  s += (val)?"high":"low";
  s += "</html>\n";

  // Send the response to the client
  client.print(s);
  delay(1);
  Serial.println("Client disonnected");

  // The client will actually be disconnected 
  // when the function returns and 'client' object is detroyed
}

```

d-a-v, 2017-12-20:
Can you try with latest git version ? [doc](https://github.com/esp8266/Arduino#using-git-version)

pfeerick, 2018-01-01:
Nothing jumps out at me from the code as being a problem. I'm guessing if you're using it with a Nano as a garage door opener that there is a relay board in the mix as well? Maybe it is draining too much power when it triggers, and the ESP8266 is crashing? 

Anyway, this is not the best place to discuss this, this is the issue tracker for the ESP8266 core, [not a discussion forum](https://github.com/esp8266/Arduino/blob/master/POLICY.md). Best bet is to ask over at http://www.esp8266.com/ or https://stackoverflow.com/

closed 2019-09-15 by elbrodark