esp8266/Arduino #3999: nodemcu8266 crashes on put

pkishino, 2017-12-20:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			NodeMCU 1 (ESP-12E)
Core Version:      	Latest git

### Description

Problem description
When performing a simple put to a rest interface, after 2 requests it always crashes

### Settings in IDE

Module:  			NodeMCU 1 (ESP-12E)
Flash Size: 		4MM
CPU Frequency:		80Mhz
Upload Using:		USB

### Sketch

```cpp

#include <ESP8266HTTPClient.h>
#include <ESP8266WiFi.h>

const char* ssid = "XXXXX";
const char* password = "XXXXX";
HTTPClient http;
String url = "http://192.168.2.1:3000/tags/uid=123456";
 
void setup() {
  // Connect to WiFi network
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
 
  WiFi.begin(ssid, password);
 
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.print(WiFi.localIP());
  Serial.println();
 
}
 
void loop() {
  http.begin(url);
  int httpCode = http.PUT("");
  if(httpCode > 0) {
      Serial.printf("[HTTP] PUT... code: %d\n", httpCode);
  } else {
      Serial.printf("[HTTP] PUT... failed, error: %s\n", http.errorToString(httpCode).c_str());
  }
  http.end();
  delay(10000);
}
```

### Debug Messages

```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 16 results
0x402037f1: ClientContext::connect(ip_addr*, unsigned short) at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 92
:  (inlined by) WiFiClient::connect(IPAddress, unsigned short) at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 135
0x402037ea: WiFiClient::connect(IPAddress, unsigned short) at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 92
0x4010020c: _umm_free at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1291
0x402033ac: WiFiClient::connect(char const*, unsigned short) at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 92
0x40205d92: std::default_delete ::operator()(WiFiClient*) const at /Users/Patrick/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/unique_ptr.h line 68
0x402027e4: HTTPClient::connect() at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 51
0x4010053d: _umm_realloc at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1495
:  (inlined by) realloc at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1713
0x40202cc8: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 51
0x4020503f: String::changeBuffer(unsigned int) at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/cores/esp8266/WString.cpp line 746
0x4020508b: String::reserve(unsigned int) at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/cores/esp8266/WString.cpp line 746
0x40202d78: HTTPClient::PUT(unsigned char*, unsigned int) at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 51
0x40202d96: HTTPClient::PUT(String) at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 51
0x402021f1: loop at /Users/Patrick/Desktop/gitrepos/mwcapp/wifi/wifi.ino line 96
0x40205b01: esp_schedule at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/cores/esp8266/core_esp8266_main.cpp line 57
0x40205b2c: loop_wrapper at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/cores/esp8266/core_esp8266_main.cpp line 57
0x40100710: cont_norm at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/cores/esp8266/cont.S line 109

```

pkishino, 2017-12-20:
Testing with using the basic examples in the library. keeps crashing on all the examples in the httpclient examples.. here is output from basichttpclient example:
```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
Decoding 24 results
0x40224680: pbuf_free at core/pbuf.c line 723
0x40228588: ip_output at core/ipv4/ip.c line 817
0x40224fc0: tcp_pcb_purge at core/tcp.c line 1425
0x4022683c: tcpip_tcp_timer at core/timers.c line 81
0x402252f7: tcp_pcb_remove at core/tcp.c line 1468 (discriminator 1)
0x402254d4: tcp_abandon at core/tcp.c line 372
0x4022679d: sys_timeout at core/timers.c line 296
0x40225532: tcp_abort at core/tcp.c line 406
0x4020137a: delay at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/cores/esp8266/core_esp8266_wiring.c line 54
0x40202f18: ClientContext::abort() at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 92
0x402031e2: ClientContext::connect(ip_addr*, unsigned short) at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 92
:  (inlined by) WiFiClient::connect(IPAddress, unsigned short) at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 135
0x40204d2d: String::operator=(char const*) at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/cores/esp8266/WString.cpp line 545
0x40202d88: WiFiClient::connect(char const*, unsigned short) at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 92
0x4020362f: TransportTraits::create() at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 51
0x40203b50: HTTPClient::connect() at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 51
0x40204b3b: String::changeBuffer(unsigned int) at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/cores/esp8266/WString.cpp line 545
0x40204320: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 51
0x40205ab4: Print::write(unsigned char const*, unsigned int) at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/cores/esp8266/Print.cpp line 99
0x4020474d: Print::write(char const*) at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/cores/esp8266/Print.cpp line 99
0x402043ce: HTTPClient::GET() at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 51
0x402021ab: loop at /var/folders/0k/px4pdd8d23vcjgcthpwh46vw0000gn/T/arduino_modified_sketch_957689/BasicHttpClient.ino line 51
0x4020213a: setup at /var/folders/0k/px4pdd8d23vcjgcthpwh46vw0000gn/T/arduino_modified_sketch_957689/BasicHttpClient.ino line 36
0x40205638: loop_wrapper at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/cores/esp8266/core_esp8266_main.cpp line 57
0x40100710: cont_norm at /Users/Patrick/Library/Arduino15/packages/esp8266/hardware/esp8266/cores/esp8266/cont.S line 109

```

d-a-v, 2017-12-20:
Can you retry using lwip-v1.4 in Tools menu ?

pkishino, 2017-12-20:
lwip? I dont see anything like that in the arduino ide tools menu

d-a-v, 2017-12-20:
If you are using latest git as stated, have you restarted the arduino IDE ?
If you still don't see the menu, can you retry with a clean install with latest git ?

closed 2017-12-21 by pkishino