esp8266/Arduino #4002: Boundary condition in cbuf

boblemaire, 2017-12-20:
While trying to figure out how data is being lost in webserver responses, I went through the code for cbuf.  I discovered a boundary condition in the resize function.  If you were to issue the function:
`cbuf:resize(cbuf.available());`
to reduce the size of a buffer to it's current contents, the new cbuf will be the exact size of the current contents, but cbuf logic requires that the buffer always be at least one byte larger than it's current contents. Here is a sequence of calls to cbuf that illustrates the problem with a cbuf(10):
```
size=10, available=0, room=9, full=false
Write one byte, returned write count=1, cbuf:size=10, available=1, room=8, full=false
Write one byte, returned write count=1, cbuf:size=10, available=2, room=7, full=false
Write one byte, returned write count=1, cbuf:size=10, available=3, room=6, full=false
Write one byte, returned write count=1, cbuf:size=10, available=4, room=5, full=false
Write one byte, returned write count=1, cbuf:size=10, available=5, room=4, full=false
Write one byte, returned write count=1, cbuf:size=10, available=6, room=3, full=false
Write one byte, returned write count=1, cbuf:size=10, available=7, room=2, full=false
Write one byte, returned write count=1, cbuf:size=10, available=8, room=1, full=false
Resize=8, returned size=8, cbuf:size=8, available=8, room=-1, full=false
Write one byte, returned write count=1, cbuf:size=8, available=1, room=6, full=false
Write one byte, returned write count=1, cbuf:size=8, available=2, room=5, full=false
Write one byte, returned write count=1, cbuf:size=8, available=3, room=4, full=false
Write one byte, returned write count=1, cbuf:size=8, available=4, room=3, full=false
Write one byte, returned write count=1, cbuf:size=8, available=5, room=2, full=false
Write one byte, returned write count=1, cbuf:size=8, available=6, room=1, full=false
Write one byte, returned write count=1, cbuf:size=8, available=7, room=0, full=true
Write one byte, returned write count=0, cbuf:size=8, available=7, room=0, full=true
```
As you can see, seven bytes are lost as a result of this.

The solution is to change cbuf line 40 from:
`if((newSize < bytes_available) || (newSize == _size)) {`
 to:
`if((newSize <= bytes_available) || (newSize == _size)) {`
  
I have tried this fix and it seems to work, in that the buffer is not resized.   I haven't yet tried this to see if it solves my missing data problem (I  have a workaround in place), but thought someone would like to know about this possible problem in what is a widely used core class.

As an afterthought, invoking the resize as cbuf:resize(cbuf.available()+1); would avoid the problem.  Of course recoding cbuf to be able to utilize all of it's size would be the best long term solution.

devyte, 2017-12-21:
@boblemaire can you please prepare a PR?

boblemaire, 2017-12-21:
OK, may take a day or two.

boblemaire, 2017-12-22:
Submitted PR #4016.

closed 2017-12-22 by boblemaire