esp8266/Arduino #4004: espClient.loadCertificate(cert)

thomas-rode, 2017-12-20:
### Basic Infos

#### Hardware
Hardware:		ESP-12 or Wemos mini
Core Version:      	2.4.0-rc2

### Description
i use a standard certificate from OU = Amazon Web Services O=Amazon.com Inc. L=Seattle ST=Washington C=US vid valid Dates from 31.08.2016 till 01.01.**2050**.
Using espClient.loadCertificate(cert); or espClient.setCertificate(binaryCert, len); the dubug output is

Organizational Unit (OU):	Amazon Web Services O=Amazon.com Inc. L=Seattle ST=Washington C=US
Not Before:			Wed Aug 31 20:25:41 2016
Not After:			Tue Nov 25 17:31:43 **1913**

there must be an problem in the time calculation (2050 --> 1913). 

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz?
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

    // Load certificate file
    File cert = SPIFFS.open("/cert.der", "r"); //replace cert.crt eith your uploaded file name
    if (!cert) {            Serial.println("Failed to open cert file");  
    } else                  Serial.println("Success to open cert file");
    delay(1000);
    if (espClient.loadCertificate(cert))        Serial.println("cert loaded\n");
    else                                        Serial.println("cert not loaded\n");

### Debug Messages

Heap: 34424
=== CERTIFICATE ISSUED TO ===
Common Name (CN):		AWS IoT Certificate
Organization (O):		<Not Part Of Certificate>
Basic Constraints:		critical, CA:FALSE, pathlen:10000
Key Usage:			critical, Digital Signature
=== CERTIFICATE ISSUED BY ===
Common Name (CN):		<Not Part Of Certificate>
Organization (O):		<Not Part Of Certificate>
Organizational Unit (OU):	Amazon Web Services O=Amazon.com Inc. L=Seattle ST=Washington C=US
Not Before:			Wed Aug 31 20:25:41 2016
Not After:			Tue Nov 25 17:31:43 1913
RSA bitsize:			2048
Sig Type:			SHA256

messages here
any ideas ?

d-a-v, 2017-12-20:
You might have rediscovered the year 2038's end of [32bits] world [more info](https://en.wikipedia.org/wiki/Year_2038_problem)
hint: 1900+2050-2038=

thomas-rode, 2017-12-21:
may be, but can that be the problem that AWS rejects the connection?

State:	receiving Certificate Request (13)
State:	receiving Server Hello Done (14)
State:	sending Certificate (11)
:wr 12 12 0
:wrc 12 12 0
:sent 12
:rn 7
:rcl
:abort
State:	sending Client Key Exchange (16)
Error: connection lost
Alert: close notify
failed, rc=-2 try again in 5 seconds

devyte, 2018-07-13:
Closing due to bearssl merged and migration from axtls. Also, lack of feedback in several months.

closed 2018-07-13 by devyte