esp8266/Arduino #4005: NodeMCU ESP8266 12E - gibberish after bootloader output in serial monitor

gigmaps, 2017-12-20:
### Basic Info
Connecting Amica NodeMCU 8266 to Arduino IDE gives gibberish after (what appears to be) correct bootloader output on reset. 

#### Hardware
Hardware:		Amica ESP8266 CP2102 NodeMCU LUA ESP-12E WIFI Serial Wireless Module
Chip:			ESP8266EX
Purchased From:	https://www.amazon.com/gp/product/B010O1G1ES/ref=ask_ql_qh_dp_hza
Core Version:      	2.4.0-rc2

### Description
Arduino IDE v1.8.5 installed with latest rc2 version of esp8266/Arduino installed - with the Silicon Labs CP2102 drivers installed correctly. Using the latest version of esp8266/Arduino and can't get the module to respond correctly (only outputs gibberish characters) - except for on bootloader reset.  Baud rate is

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP- 12E Module)
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Flash Mode:		n/a
Flash Frequency:	?? not sure where this is set in Arduino IDE ??
Upload Speed:	115200
Upload Using:	USB
Reset Method:	nodemcu

### Serial Output
````
ets Jan  8 2013,rst cause:2, boot mode:(3,7)
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v355232ff
~ld
UJj,⸮⸮⸮T⸮⸮⸮-+Z⸮孲⸮⸮U⸮+⸮⸮⸮⸮⸮T⸮⸮⸮⸮⸮⸮UUUVU⸮T⸮⸮⸮-+Z⸮孢⸮⸮UJ+⸮⸮⸮⸮⸮T⸮⸮⸮⸮⸮⸮UUUVUE⸮
````

### Sketch
Only testing with the WifiScan example for now

5chufti, 2017-12-21:
set the baudrate in your ser.mon. window to 74880 and use Serial.begin(74880); in  your sketch, then everything should be ok.

gigmaps, 2017-12-21:
@5chufti legend !

On my system ser.mon window doesn't even need to be at 74880 - as long as the sketch has Serial.begin(74880) it seems to work fine.

thank you

closed 2017-12-21 by gigmaps

pauledd, 2017-12-29:
Oh man, it took half a day to stumble upon this issue. 
This is even more weird. 
I also need to set Serial.begin(74880); in my sketches but in the serial window
I can ONLY use 115200 to see correct output, if I select anything else (even 74880) I get totally
useless output.

gigmaps, 2017-12-31:
@pauledd yes highly frustrating - and probably the card - I repurchased two Makerfocus NodeMCU 12E cards from amazon and don't have the same issues (though some sketches are a little fussy on the baud rate selected in the SM - but not the sketch).
An interesting read to shed some light on all the different NodeMCU card versions / manufacturers available:
https://frightanic.com/iot/comparison-of-esp8266-nodemcu-development-boards/

brucebuxkemper, 2018-01-13:
You are a hero, I've been searching sites and posts for a day now and this is the one that fixed it for me. 

They must have a clock divisor set wrong some place!

Thank you!!!!

brucebuxkemper, 2018-01-14:
If the crystal is running 1.5 times too fast, could this be why my wifi is not working?

5chufti, 2018-01-14:
Depend if this frequency is properly reflected in the esp_init_data block.

js9742, 2018-01-15:
Same here!
I got a Lolin ESP8266 nodeMCU v3 from aliexpress..
I'm testing on Hellow Server example code.
it only works fine at sketch 74880, ser.mon 115200.
But the wifi is not working..

brucebuxkemper, 2018-01-15:
Hi,
I still have not found a solution. I'll email you if I do.

I'm not sure the serial issue is a fundamental problem with the board and
the bad clock rate is keeping the wifi from working.

I'm still working on it and I ordered a different board.

FYI, my email is BruceBuxkemper@gmail.com

pauledd, 2018-01-15:
I got my money back for the faulty unit and ordered another one here in germany. The board looks quite the same like the Chinese one except one part: the faulty china unit has on that wifi metal case printed "Vendor AI-THINKER". The good one I got from germany has printed HYSIRY and this "pacman" next to the Wifi-logo.

brucebuxkemper, 2018-01-15:
I"m not sure where the esp_init_data block is or how to debug it.

In Arduino IDE the board is set to 80MHz.

brucebuxkemper, 2018-01-15:
Did you have the same baud rate issue/no wifi on the Chinese board and then
have it work on the German board?

If so I will quit trying on the HiLetgo board I have and wait for a new
board.

pauledd, 2018-01-15:
Yes, 74880 baud and no wifi. But be warned, dont try to identify "good" boards on the shop images of the board. I noticed that with the china seller. He sent me the "AI-THINKER" board while in his shop the images did not have that logo. Its a bit trial and error...

surak, 2018-01-15:
I'm on the same boat. If I try the wifiscan example, I only get some readable output when I set to 74880 and keep reading at 115200, but I don't get any wifi.

surak, 2018-01-15:
And the Blink examples don't work with the builtin led either

5chufti, 2018-01-16:
unfortunately there is no "standard" for the builtin led for esp "generic boards", so your mileage with blink example _will_ vary. The latest github should have some led selection menu I heared...

brucebuxkemper, 2018-01-20:
Just wanted to update you.

I got a new HiLetgo board and it works fine.

My conclusion is that if you have to set baudrate to 74880 to get a good
output 115200, the board is bad and wifi will not work. Throw it away and
get a new one.

For $6 it is not worth wasting your time on. I'm still waiting to see if
HiLetgo atleast does the right thing and replaces the two bad boards that I
have now.

m-m-0-0, 2018-01-20:
I have a board that has the same issue, looks like the problem is it has a 40MHz crystal instead of the standard 26MHz one, what I did to solve this was to flash esp_init_data_default.bin with ESP8266 Download Tool changing CrystalFreq from 26M to 40M (I flashed it at 0x3fc000, having a 32mb board).
Let me know if this solves it for you.

m-m-0-0, 2018-01-20:
ok, turns out that doesn't work with arduino, so what I did now was commenting line 91,93,94 and 95 from cores\esp8266\core_esp8266_phy.c and it seems to be working correctly, i'm not sure wifi works as intended, i'll have to test that.
It is not a good solution, but it seems to work.

5chufti, 2018-01-20:
Did you try the 40MHz crystal setting in the IDE?

js9742, 2018-01-21:
Nice! It works at 40MHz! 
[48]=0 on the core_esp8266_phy.c
Now it works with same baud rate (sketch 115200, ser.mon 115200).

and 1 AP scanned with the example code of wifi scan, but it looks like not on my router..

at this time my iPhone can scan 7 AP's.. but the esp8266 can't scan all AP's now. 

if I grasp the usb cable, 2 AP's could be scanned.. weird..

m-m-0-0, 2018-01-21:
@5chufti I don't have any crystal setting in the IDE, is it just me? if not, it could surely be added.
@js9742 I too am having problems with the wifi scan, it's just seeing 1 AP, while my phone can see at least 5, I'll test and let you know.

surak, 2018-01-21:
@js9742 @m-m-0-0 which board are you using on arduino? I'm trying with the nodeMCU 1.0 or the WeMos D1 & R2. Your frequency trick worked for the serial output, but I still don't have wifi.

m-m-0-0, 2018-01-21:
I have a LoLin NodeMCU v3, looks like the wifi works, although really really badly. try changing location to see if you can find at least one AP if you can.

surak, 2018-01-21:
@m-m-0-0 I'm literally touching the router with the same board as you. Was the frequency the only thing you changed on arduino? Which board do you set on arduino IDE?

m-m-0-0, 2018-01-21:
[image]
This is my usual output, without moving the board (note: that is not my router).

@surak I set NodeMCU 1.0, with all the other settings unchanged.

5chufti, 2018-01-21:
@m-m-0-0 if you don't see the crystal setting, try the latest stable release of the arduino esp core, there they are for sure.

surak, 2018-01-21:
@5chufti where? I just downloaded 1.8.5

5chufti, 2018-01-22:
@surak I was talking about the **core** not the Arduino **IDE**.
Go to the board manager and see what version you find for esp8266.
See https://github.com/esp8266/Arduino#installing-with-boards-manager for how to install.

m-m-0-0, 2018-01-22:
For the crystal option to appear, you have to select generic 8266 module as
the board. I was using the nodemcu option.

m-m-0-0, 2018-01-22:
@5chufti Unfortunately, even after erasing the whole flash the situation is the same.

m-m-0-0, 2018-01-22:
I've tried the WiFiAccessPoint example and it is working correctly, but it still can't find the majority of the AP. Could it be a software problem? I don't see what it could be though.

brucebuxkemper, 2018-01-24:
That makes sense.

I just trashed mine away and got new ones.  HiLetgo has said that they will
replace them.

surak, 2018-02-25:
I removed the ESP8266 and used this configuration, with arduino fully updated:

[image]

Everything works now: it reads at 115200, and finds my APs

mjs513, 2018-03-21:
Ok.  Still confused on what the solution to this issue.  I just downloaded the latest version of the core and tried the same dev board as in post 1.   The only think that seems to work is setting the baud rate to 74880?  I did notice that v1.0 of the board defaults to the 40Mz crystal.  I tried the solution in the above post but it did not work either - I do get a warning "warning: espcomm_send_command: wrong direction/command: 0x01 0x00, expected 0x01 0x08".

Or is the solution ditch this board for something else?

brucebuxkemper, 2018-03-21:
The above did not work for me either.  However, I did order two new boards
and had no problems.

I then emailed HiLetGo and they sent me replacement boards at no charge. I
also ordered several more with no problem.

I highly recommend that you throw the board you have away and order another
one.

5chufti, 2018-03-21:
> My conclusion is that if you have to set baudrate to 74880 to get a good output 115200, the board is bad and wifi will not work. Throw it away and get a new one. 

Exactly for this case it is possible to set crystal to 40MHz. With this setting baudrates and WiFi will be ok again.

Please send all "defective" boards in my direction, I can allways use some spares ...

mjs513, 2018-03-24:
Thanks Guys for the responses.  Sorry it took me so long to respond.  Been tied up since I posted.  I email them and see if they send me a replacement.

Thanks
Mike

brucebuxkemper, 2018-04-02:
I now have another board that does not work unless I set the baud rate to
74880.   I have flashed the same firmware that I flashed into the working
boards.

Is it possible that this board has a different crystal and needs to a
different esp_init_data block?

If so, how do I change this block?

devyte, 2018-04-03:
@brucebuxkemper yes, it depends on the board manufacturer. It should be enough to select the crystal freq from the IDE menu.

kabasher, 2018-05-03:
All I had to do is - stooped using GPIO 15 ( I had an external LED connected to GPIO Pin 15 with pull down resistor. Not sure why this solved my problem but I will be investigating this soon.

tnavda, 2018-07-06:
Guys thanks for the hints, and this got my (pos?) working finally.  I created a new boards.txt config which works for mine.  My vendor was AI-THINKER.  Access point works now at least...

```
##############################################################
nodemcuv3.name=NodeMCU 1.0 (ESP-12E Module ChopShop)
nodemcuv3.build.board=ESP8266_NODEMCU
nodemcuv3.build.variant=nodemcu
nodemcuv3.upload.tool=esptool
nodemcuv3.upload.maximum_data_size=81920
nodemcuv3.upload.wait_for_upload_port=true
nodemcuv3.upload.erase_cmd=
nodemcuv3.serial.disableDTR=true
nodemcuv3.serial.disableRTS=true
nodemcuv3.build.mcu=esp8266
nodemcuv3.build.core=esp8266
nodemcuv3.build.spiffs_pagesize=256
nodemcuv3.build.debug_port=
nodemcuv3.build.debug_level=
nodemcuv3.menu.CrystalFreq.40=40 MHz
nodemcuv3.menu.CrystalFreq.40.build.extra_flags=-DF_CRYSTAL=40000000 -DESP8266
nodemcuv3.menu.FlashFreq.40=40MHz
nodemcuv3.menu.FlashFreq.40.build.flash_freq=40
nodemcuv3.menu.CpuFrequency.80=80 MHz
nodemcuv3.menu.CpuFrequency.80.build.f_cpu=80000000L
nodemcuv3.menu.CpuFrequency.160=160 MHz
nodemcuv3.menu.CpuFrequency.160.build.f_cpu=160000000L
nodemcuv3.upload.resetmethod=nodemcu
nodemcuv3.build.flash_mode=dio
nodemcuv3.build.flash_freq=40
nodemcuv3.menu.FlashSize.4M1M=4M (1M SPIFFS)
nodemcuv3.menu.FlashSize.4M1M.build.flash_size=4M
nodemcuv3.menu.FlashSize.4M1M.build.flash_size_bytes=0x400000
nodemcuv3.menu.FlashSize.4M1M.build.flash_ld=eagle.flash.4m1m.ld
nodemcuv3.menu.FlashSize.4M1M.build.spiffs_pagesize=256
nodemcuv3.menu.FlashSize.4M1M.upload.maximum_size=1044464
nodemcuv3.menu.FlashSize.4M1M.build.rfcal_addr=0x3FC000
nodemcuv3.menu.FlashSize.4M1M.build.spiffs_start=0x300000
nodemcuv3.menu.FlashSize.4M1M.build.spiffs_end=0x3FB000
nodemcuv3.menu.FlashSize.4M1M.build.spiffs_blocksize=8192
nodemcuv3.menu.FlashSize.4M3M=4M (3M SPIFFS)
nodemcuv3.menu.FlashSize.4M3M.build.flash_size=4M
nodemcuv3.menu.FlashSize.4M3M.build.flash_size_bytes=0x400000
nodemcuv3.menu.FlashSize.4M3M.build.flash_ld=eagle.flash.4m.ld
nodemcuv3.menu.FlashSize.4M3M.build.spiffs_pagesize=256
nodemcuv3.menu.FlashSize.4M3M.upload.maximum_size=1044464
nodemcuv3.menu.FlashSize.4M3M.build.rfcal_addr=0x3FC000
nodemcuv3.menu.FlashSize.4M3M.build.spiffs_start=0x100000
nodemcuv3.menu.FlashSize.4M3M.build.spiffs_end=0x3FB000
nodemcuv3.menu.FlashSize.4M3M.build.spiffs_blocksize=8192
nodemcuv3.menu.LwIPVariant.v2mss536=v2 Lower Memory
nodemcuv3.menu.LwIPVariant.v2mss536.build.lwip_include=lwip2/include
nodemcuv3.menu.LwIPVariant.v2mss536.build.lwip_lib=-llwip2
nodemcuv3.menu.LwIPVariant.v2mss536.build.lwip_flags=-DLWIP_OPEN_SRC -DTCP_MSS=536
nodemcuv3.menu.LwIPVariant.v2mss1460=v2 Higher Bandwidth
nodemcuv3.menu.LwIPVariant.v2mss1460.build.lwip_include=lwip2/include
nodemcuv3.menu.LwIPVariant.v2mss1460.build.lwip_lib=-llwip2_1460
nodemcuv3.menu.LwIPVariant.v2mss1460.build.lwip_flags=-DLWIP_OPEN_SRC -DTCP_MSS=1460
nodemcuv3.menu.LwIPVariant.Prebuilt=v1.4 Higher Bandwidth
nodemcuv3.menu.LwIPVariant.Prebuilt.build.lwip_lib=-llwip_gcc
nodemcuv3.menu.LwIPVariant.Prebuilt.build.lwip_flags=-DLWIP_OPEN_SRC
nodemcuv3.menu.LwIPVariant.OpenSource=v1.4 Compile from source
nodemcuv3.menu.LwIPVariant.OpenSource.build.lwip_lib=-llwip_src
nodemcuv3.menu.LwIPVariant.OpenSource.build.lwip_flags=-DLWIP_OPEN_SRC
nodemcuv3.menu.LwIPVariant.OpenSource.recipe.hooks.sketch.prebuild.1.pattern=make -C "{runtime.platform.path}/tools/sdk/lwip/src" install TOOLS_PATH="{runtime.tools.xtensa-lx106-elf-gcc.path}/bin/xtensa-lx106-elf-"
nodemcuv3.menu.Debug.Disabled=Disabled
nodemcuv3.menu.Debug.Disabled.build.debug_port=
nodemcuv3.menu.Debug.Serial=Serial
nodemcuv3.menu.Debug.Serial.build.debug_port=-DDEBUG_ESP_PORT=Serial
nodemcuv3.menu.Debug.Serial1=Serial1
nodemcuv3.menu.Debug.Serial1.build.debug_port=-DDEBUG_ESP_PORT=Serial1
[... 52 lines cut ...]
nodemcuv3.menu.FlashErase.sdk=Sketch + WiFi Settings
nodemcuv3.menu.FlashErase.sdk.upload.erase_cmd=-ca "{build.rfcal_addr}" -cz 0x4000
nodemcuv3.menu.FlashErase.all=All Flash Contents
nodemcuv3.menu.FlashErase.all.upload.erase_cmd=-ca 0x0 -cz "{build.flash_size_bytes}"
nodemcuv3.menu.UploadSpeed.921600=921600
nodemcuv3.menu.UploadSpeed.921600.upload.speed=921600
nodemcuv3.menu.UploadSpeed.115200=115200
nodemcuv3.menu.UploadSpeed.115200.upload.speed=115200
nodemcuv3.menu.UploadSpeed.9600=9600
nodemcuv3.menu.UploadSpeed.9600.upload.speed=9600
nodemcuv3.menu.UploadSpeed.57600=57600
nodemcuv3.menu.UploadSpeed.57600.upload.speed=57600
nodemcuv3.menu.UploadSpeed.230400.linux=230400
nodemcuv3.menu.UploadSpeed.230400.macosx=230400
nodemcuv3.menu.UploadSpeed.230400.upload.speed=230400
nodemcuv3.menu.UploadSpeed.256000.windows=256000
nodemcuv3.menu.UploadSpeed.256000.upload.speed=256000
nodemcuv3.menu.UploadSpeed.460800.linux=460800
nodemcuv3.menu.UploadSpeed.460800.macosx=460800
nodemcuv3.menu.UploadSpeed.460800.upload.speed=460800
nodemcuv3.menu.UploadSpeed.512000.windows=512000
nodemcuv3.menu.UploadSpeed.512000.upload.speed=512000
nodemcuv3.menu.led.1=1
nodemcuv3.menu.led.1.build.led=-DLED_BUILTIN=1
```

tnavda, 2018-07-07:
I am still not able to get Wifi module to work correctly when building with Arduino.  Upon further investigation, I am selecting a 4MB flash which I have.  I found a post on epressif's forum where someone else had trouble.  They said, flash it with the default blank.bin, and see if the AP works, if it does then the chip is fine.   So I did that, and sure enough the AP works without issue.  Since I have a 4MB flash, if I dump the address for 0x3FC000 for 128 bytes, I should get the correct esp_init_data, which I do when I flash with the defaults.  However when I flash a Sketch with Arduino IDE I get some weird looking block where the esp_init_data should be.  Is this just how Arudino does it, or is something broken?

How I flash with defaults
```
esptool.py --port /dev/tty.SLAB_USBtoUART --baud 115200  write_flash -fm dio -fs 4MB 0x00000 boot_v1.7.bin 0x01000 user1.2048.new.5.bin 0x3fc000 esp_init_data_default_v08_40mhz.bin 0xfe000 blank.bin 0x3fe000 blank.bin 
````

0x3FC000 with defaults flashed
```
0000h: 05 08 04 02 05 05 05 02 05 00 04 05 05 04 05 05  ................ 
0010h: 04 FE FD FF F0 F0 F0 E0 E0 E0 E1 0A FF FF F8 00  .þýÿðððàààá.ÿÿø. 
0020h: F8 F8 4E 4A 46 40 3C 38 00 00 01 01 02 03 04 05  øøNJF@<8........ 
0030h: 00 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00  ................ 
0040h: E1 0A 00 00 00 00 00 00 00 00 01 93 43 00 00 00  á..........“C... 
0050h: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................ 
0060h: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................ 
0070h: 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00  ................ 
```

0x3FC000 after Arduino flashes
```
0000h: 00 00 FF 00 32 3F 33 39 37 39 38 35 05 00 06 00  ..ÿ.2?397985.... 
0010h: 08 00 08 00 08 02 3E 00 FD 07 BD 07 BD 07 BD 07  ......>.ý.½.½.½. 
0020h: 0F 03 02 0B 02 01 07 01 01 07 01 01 0A 10 80 FE  ..............€þ 
0030h: 0E 80 4E 4A 46 40 3C 38 02 07 0C 12 17 1B 02 06  .€NJF@<8........ 
0040h: 0C 12 16 1B 03 07 0C 13 17 1B 02 06 0B 12 16 19  ................ 
0050h: 00 00 00 00 00 00 26 0D 80 FE F0 03 00 00 0F 03  ......&.€þð..... 
0060h: F5 70 00 B2 94 01 60 61 04 C1 EB 65 00 00 00 00  õp.²”.`a.Áëe.... 
0070h: 00 00 01 00 00 00 00 00 00 00 00 00 00 00 70 04  ..............p. 
```

devyte, 2018-07-07:
I believe that we don't use the esp_init_data.bin, and instead we build the init data from a C array defined in a .c file. 
You shouldn't need to flash that, building your app builds a single .bin (spiffs not included) that you flash, and that's it. You may need to wipe the flash. All of that is available from the IDE menu.

tnavda, 2018-07-07:
Devyte,

What’s the best way to troubleshoot the Wifi AP not working when building and deploying via Arduino IDE?  I’ve tried with Core and Wifi debugging but it doesn’t tell any useful information.

Additionally while it’s not a big deal if you aren’t using the bin, but is the CPU not looking specifically at 0x3fc000 for my flash size at the esp init data?  So when I write the sketch to the cpu, even when erased, the values at that offset don’t match the esp init data.  So this to me concludes that arduino isnt writing the bin at the write locations.  Or can the esp init address be configures during boot process which is taken care of in the build process?

skulko, 2018-08-11:
I have a cheap ali esp8266(esp-12e) and this worked best for me: (wifi is working and serial monitor of arduino is almost always showing correct symbols):

Serial.begin(921600);

UploadSpeed: 921600

Serial Monitor: 1000000 baud

Lep230, 2019-03-08:
whit cristal Frequency = 26 Mhz works all right at mine esp-01 black!!

2xSamurai, 2019-03-23:
> I have a cheap ali esp8266(esp-12e) and this worked best for me: (wifi is working and serial monitor of arduino is almost always showing correct symbols):
> 
> Serial.begin(921600);
> 
> UploadSpeed: 921600
> 
> Serial Monitor: 1000000 baud

I was able to see the serial output when I used this setting. But I was unable to connect to the wifi AP.

MaxiGabriel, 2019-08-10:
The Serial output show some errors with Serial.begin(921600); in my Arduino Weros D1 (Wifi).
I try this configuration and desapear all errors in Serial output:

Serial.begin(1000000);

UploadSpeed: 921600

Serial Monitor: 1000000 bauds

Using NodeMCU 1.0 (ESP-12E Module)

diyelexvidyadhar, 2019-08-10:
I get my IP by following these steps maybe it work for you
[image]

 Arduino IDE 1.8.9 , windows 10 
settings
[image]

code  
#include <ESP8266WiFi.h>
 
  const char* wifiName = "wifi name";
  const char* wifiPass = "your password";
 
// the setup function runs once when you press reset or power the board
void setup() {
  Serial.begin(115200);
  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("ESP8266 MAC: ");
  Serial.println(WiFi.macAddress());
  Serial.print("Connecting to ");
  Serial.println(wifiName);
 
  WiFi.begin(wifiName, wifiPass);
 
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}
 
// the loop function runs over and over again forever
void loop() {
  
}

JDK-Dulangi, 2019-10-14:
Hey i think I also brought a cheap Chinese board though it was sold at a reputed shop, it prints gibberish at baud rates of 9600, and 115200, but at 74880 it started printing some English
and I got a check sum error, then i tried Erasing all Flash contents and now I m getting compiling errors as undefined reference to vPortFree
Does anyone know how to fix this?

JavanPoirier, 2019-12-06:
I was also having this issue in my own code but not in the samples. I was able to isolate it to a single pinMode() I was calling. This was a result of assigning a GPIO that was not what I thought it was. On a vanilla Arduino I am used to assigning the pins based on what number is on the board. So when I set pin 7 for this board it is not equivalent to D7...? Anyways, referring to D#, changing my pinout and reading up on the GPIO pinout got me past this.

sbabadag, 2020-05-18:
> I get my IP by following these steps maybe it work for you
> [image]
>
> Arduino IDE 1.8.9 , windows 10
> settings
> [image]
>
> code
> #include <ESP8266WiFi.h>
>
> const char* wifiName = "wifi name";
> const char* [...]

Absolulty after many trials ... your configuration worked for me.... but dont forget to reset by chpd pin to gnd after downloading to board many thanks...

sbabadag, 2020-05-18:
sorry rst pin to gnd

tralves, 2020-11-14:
My problem was that I was setting GPIO1 (TXD0) as OUTPUT and writing LOW... After I stopped doing that, it started working (74880 baud).

bhamilton1co, 2021-02-04:
I was having a similar issue as others with a knock off ESP8266EX D1 Mini Pro; garbage in serial monitor, even though the board would flash successfully. Things started looking better using Serial.begin(74880);  and changing various IDE options like the Crystal Freq, but WiFi was not working. Ultimately I figured out that it had a different size Flash chip than what was indicated on the product listing and purchase order. So when I set everything back to normal including Serial.begin(115200); Crystal Frequency 26MHz, etc, but choosing **Flash Size 4MB (FS: 1MB** my sketch starting working properly, WiFi started working, everything is good to go. Here are my Arduino settings

[image]

d-a-v, 2021-02-04:
@bhamilton1co what's the flash chip reference ?