esp8266/Arduino #4006: Memory leak when using Adafruit_NeoPixel library and UDP

EricMiddleton, 2017-12-21:
### Basic Infos
Receiving large amount of UDP datagrams while using Adafruit_NeoPixel library causes the ESP to start leaking memory until it runs out and reboots.

#### Hardware
Hardware:			NodeMCU
Core Version:			2.3 and latest (12/20/17) git
OS:					Linux (Debian Stretch)
Arduino:				1.8.4

### Description
I have a large application that involves updating Neopixel lights from data received in UDP datagrams. I had noticed that the ESP will often get into a memory leak "loop" in which the device will slowly start running out of memory before dying. I made the following observations:

- The problem occurs when using the stock WiFiUDP library and the ESPAsyncUDP library

- Usually, the ESP will disconnect from the WiFi AP once the memory leak starts. Other times, it will stay connected until running out of RAM

- Once the memory leak begins, it cannot be stopped. I've tried stopping NeoPixel updates and even deleting the entire Adafruit_NeoPixel strip instance once the leak starts, but the leak always continues

I understand that the Adafruit_NeoPixel library is not recommended for use with the ESP (because it stops interrupts while outputting LED data), but the fact that anything can cause the ESP (which is otherwise running properly) to start continuously leaking data seems like evidence of a larger issue.

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4M (1M SPIFFS)
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu
Debug Level:		CORE+WIFI+HTTP_UPDATE+UPDATER+OTA

### Sketch

```cpp

#include <Adafruit_NeoPixel.h>

#include <ESP8266WiFi.h>
#include <ESPAsyncUDP.h>

#include <string>

const uint16_t PORT = 5492;

const char* SSID = "SSID";
const char* PSK = "PSK";

AsyncUDP socket;

int lightCount = 300;

Adafruit_NeoPixel strip(lightCount, 2, NEO_RGB | NEO_KHZ800);

void setup() {
  Serial.begin(115200);
  
  Serial.print("\nConnecting to AP");
  WiFi.begin(SSID, PSK);
  while(WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print('.');
  }
  Serial.println("done");

  strip.begin();

  if(socket.listen(PORT)) {
    socket.onPacket([](AsyncUDPPacket p) {
      Serial.print(".");
      strip.show();
    });
  }
  else {
    Serial.println("[Error] Failed to listen on socket");
  }
  
}

unsigned long nextTime = 0;
void loop() {
  auto curTime = millis();
  if(curTime >= nextTime) {
    nextTime = curTime + 1000;

    Serial.println(ESP.getFreeHeap());
  }
}

```
### Python Script

This script can be used to start the memory leak:
```python

import socket;
import time;

ESP_IP = "192.168.1.181"
PORT = 5492

sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
bigData = bytearray(905);
bigData[1] = 9;

while True:
	sock.sendto(bigData, (ESP_IP, PORT))
	time.sleep(0.01)

```
### Debug Messages

```
[MemoryLeak.log](https://github.com/esp8266/Arduino/files/1577894/MemoryLeak.log)
```

raheelh, 2018-01-07:
There was a leak fixed about 8 days ago i think. Did you try out latest and still seeing this?

moose4lord, 2018-01-13:
I'm having a problem with an application that uses Neopixels and WiFi, and I thought your memory leak experience might be related to what I'm seeing. However, when I run your test case with my ESP setup everything works fine. The memory reported by getFreeHeap() fluctuates a little, but I don't see the memory leak. I also didn't see the "wifi evt: 7" events in your log, but I'm sitting right next to my wifi router, so the wifi connection is rock solid. I tried to simulate a bad connection by blocking the esp with a firewall rule, but still no memory leak. Sorry I couldn't confirm your findings. I was hoping a memory leak would explain the intermittent MQTT reboots I'm seeing with my application.

Manchicor, 2018-03-05:
I have the same issue and it is triggered by the wifi evt: 7 (ProbeRequest). It happens when working as an AP, of course, and can be replicated by having several computers, phones o tablets refreshing the list of available wifi networks.

The module is working alright handling the WS2812 pixels until after one of the "wifi evt: 7" goes wrong and the memory claimed by the wifi connection handler is not returned. (Previously, all the calls claim some memory from the heap and then it is returned). After that point, the wifi connection is interrupted and with every wifi event the heap gets smaller until resetting.

To avoid that I´m using Arduino IDE 1.6.4 and SDK 1.2.0. Have you been able to get it to work with newer SDK and/or IDE?

If I remove the interrupt-lock while sending the WS2812 data, it doesn´t have any problem, but of course, the pixels get all f´d up...

bsfconception, 2021-06-10:
Hi
Did you find out a solution if the this issu (I do encounter exactly the same problem with same features).
For ESP32 there is another WS2812 lib that looks good, but I'm unable to solve it on ESP8266
Thanks

Manchicor, 2021-06-10:
Hey! I am not using libraries anymore, but using my own direct-port-manipulation code to drive the WS2812 pixels. Still, if the #of pixels is too high (over 2000) the wifi starts breaking up sometimes. My solution was to split the output in a couple different gpio´s, that way there´s a little time for the ESP8266 to take care of the wifi tasks between sending the pixel stream to the different outputs.  Send out1 --> delay(0) --> send out2 --> delay(0) send out3 --> etc.  (The output bit has to stay inside interrupt blocking statements ets_intr_lock() / ets_intr_unlock();). That´s so far what I have been using to prevent the issue.

bsfconception, 2021-06-11:
Hi @Manchicor, thanks a lot for the detailed answer. I guess that I I'll do the same.
Sometime re-doing from scratch using the device's specifications is the better way ;-) 
Regards