esp8266/Arduino #4008: Wifi Not connecting on Wemos mini pro

martin072, 2017-12-21:
I Have a couple of wemos devices (Wemos D1 R2 and a Wemos Mini) and the all run a piece of code fine. 

I recently got some wemos mini pro's, but when loading the same s/w on to it, it won't connect to the wifi network. 

I tried moving closer to the router, no joy.
Flashed it with a wifi scanner, and noticed my network doesn't pop up in the list.. not even when I move close.. 

Any idea's?

martin072, 2017-12-23:
Other Update.. Running the module in Soft AP does broadcast the SSID, but unable to connect to it..
Could this be related to non-original wemos modules (ie. this one is from AliExpress)

5chufti, 2017-12-27:
Does the WiFi scanner show any other networks?
Is your network on 2.4GHz?  On a channel <=12 ?
Is your AP configured to show SSID?

aslake, 2017-12-30:
I have the same problem with the Wemos D1 mini pro.
Only shows the AP when without password, but cannot connect.
Code works well on D1 mini.

martin072, 2018-01-02:
Yes it sometimes shows up in the scanner, it appears and disappears. It is running on 2.4Ghz indeed. 
When I try a D1 R2 board it works fine instantly. I am thinking this could be caused by a Wemos Mini Pro clone? I noticed they don't carry the wemos logo on it... I've ordered an official wemos mini pro, see if it makes a difference..

d-a-v, 2018-01-02:
If you have a magnifier or a multimeter, you could check the antenna connection ?

martin072, 2018-01-02:
It’s on the internal antenna.

d-a-v, 2018-01-02:
I meant the soldering points of the internal antenna. With esp powered off and with a digital multimeter, you can check if the wire is continuous (if things are [accessible](https://cdn.instructables.com/FJV/FBRR/IRLQIJ1O/FJVFBRRIRLQIJ1O.MEDIUM.jpg))

beastgrim, 2018-02-28:
I bought 4 pcs Wemos D1 Mini Pro clone and have the same trouble ((
Does anyone know at least some way to make them work?
Thanks!

pieman64, 2018-03-01:
@beastgrim quite a few bugs in the latest Core release and we are eagerly awaiting the next release that has the fixes. Try rolling back to the very buggy, but usable 2.3.0 or use Git with the master branch.

ThomasQvist, 2018-03-09:
I experience the same problems with running the Wemos D1 Mini pro as an AP.
(No problems with the same software on a "non-pro" Wemos)
The signal fluctuates (signal measured with the excellent vistumbler)
[image]
Sometimes I am lucky and can connect to the access point (I expect that I need a connection for a couple of seconds to make the actual handshake).
I see no change in the measured signal whether I setup the Wemos AP to run with or without password - - but I might be that the handshake is faster without password - and therefore it works a little better.

I have full connection to the wemos (over serial) and I get no error code or any other diagnostics.

Also note that I have not seen any problems connecting the Wemos D1 Mini Pro to a wifi-network - it is only the AP-functionality that is giving me problems

igrr, 2018-03-09:
@ThomasQvist make sure you are running it in AP mode only. Check that you have `WiFi.mode` before calling `WiFi.softAP`:

```
  WiFi.mode(WIFI_AP);
  WiFi.softAP(ssid, password);
```

It the chip is in STA+AP mode, then the station will be trying to scan for a network to connect to, and since scan covers multiple channel, the AP will not be on the desired channel for some percentage of total time. This might explain the behavior you are observing.

martin072, 2018-03-14:
Small Update, I ordered an **official** Wemos Mini Pro, and unfortunately the same results. I am not able to connect to my Local home WiFi, where as the regular wemos don't have any issues at all..

5chufti, 2018-03-14:
please check that you are on internal antenna according to this picture:
https://arduino-projekte.info/unterschied-wemos-d1-mini-vs-d1-mini-pro/

and also check that the crystal (samll metall square close to D1/Rx pins) is 26MHz

ThomasQvist, 2018-03-14:
Thanks for the update - makes me happy - since I have been trying "everything" (including the advice from @igrr ) and checked the antenna without any luck.
...and also makes me sad - I am waiting for a new batch of Wemos d1 mini pros - hoped that I just had a faulty one.
Are there anyone that can get a stable signal from a Wemos Mini d1 Pro running in AP-mode?
(And again - I only have problems in AP-mode - not connecting via Wifi to other AP's)

martin072, 2018-03-14:
@ThomasQvist,

For me it’s the WiFi connections that doesn’t work. I am not using the AP function

fabian727, 2018-05-06:
I also bought my WEMOS from Ali...
I am connecting to a router, but only have problems if I add the UpdateServer.
If I recompile and flash with other data, then recompile with the correct one, I can connect. If I uncomment the lines for the UpdateServer I cannot connect.
There is also running: mDNS and NeoPixel. That's it

martin072, 2018-08-06:
Any update on this subject?

michaelradhuber, 2019-01-23:
Try this: Before setting up the softAP, erase the flash storage and every else that might be saved there with this command
`Wifi.disconnect();`

zeldor, 2019-09-08:
Have the same issue as already described above with a WEMOS D1 mini pro board.
Tried different WIFI-modes, from unprotected to WEP and WPA. Different static Channels, short passwords without any special characters etc.
I can't establish a connection :(
WIFI module gets recognised, also is able to see the WIFI-channels around me.

Need some advise or a handy trick to get a connection to work.

[image]
[image]

Thanks!

devyte, 2019-09-08:
Short password means >= 8 chars
Try reducing Tx power

zeldor, 2019-09-09:
@devyte, thank you for your reply! Password length depends on the used security: 5 chars, 8 chars.

TX power on the AP cant be reduced, its kind of blackbox.  
I will try to test this point on my second DDWRT AP which has an additional 2,4 GHz channel.

Changing the antenna is not so easy: 
the resistor has to be desoldered and changed the position to make the external antenna working.
Currently I am able to scan for near Channels from the WEMOS, this mens antenna is woking pretty fine and the range is for the testing mode acceptable. 
[image]

Has someone a good piece of code to test the WIFI connection in a good way?

devyte, 2019-09-09:
Check return of WiFi.begin() (i think it returns whether it succeeded).
By reduce Tx I meant on the ESP, not your AP. 
And try pwd with 8chars min regardless of security.

srinpraveen, 2019-09-09:
> 
> 
> I have been playing around with this for hours, everything seemed fine but i could not connect to my home WiFi, my router saw it but didn't assign an IP.
> I tried setting a static IP and still the same, i then tried a hotspot from my mobile and it connected as soon as I uploaded the script, so I set up my router identical to my mobile hotspot, but still nothing. until i pressed reset on my Wemos!! straight away it connected to my router with an IP. everytime i upload a script it needs a reset before it will connect hopefully this helps people!

seanmcd1405 , I have the exact same problem on my Wemos D1 mini board bought from Aliexpress. I have done everything except press the reset button. Maybe I will do that tonight and report back to you if it works on mine. Thanks for the tip.

mhightower83, 2019-09-10:
Try positioning the WEMOS board such that there is nothing near the antenna. If your kit came with the connectors with the long leads, plug that onto your headers to give more space between the antenna and the protoboard. Try and keep your jumper wires clear of the antenna space as much as possible.

srinpraveen, 2019-09-11:
@mhightower83 ,  sorry turned out that it was the simplest of mistake. The ssid of my router when first configured many many years ago had a whitespace at the end of the ssid string. I was not aware of this till now. I would like to kick myself for my years of ignorance. Now it seems to work very well.

jandrassy, 2019-09-11:
> Try positioning the WEMOS board such that there is nothing near the antenna. If your kit came with the connectors with the long leads, plug that onto your headers to give more space between the [...]

did you try reduce the RF TX power with WiFi.setOutputPower??

srinpraveen, 2019-09-11:
@jandrassy , I actually did not have to change my wifi power settings. The root cause of my problem was a trailing whitespace character on my ssid string when the router was configured. I had missed catching this for a long period of time. After fixing it, the wifi connectivity so far has been very stable on the d1 mini.

zeldor, 2019-09-11:
@srinpraveen, sorry for the dumb question, is it one original wemos or a copy from aliexpress or so? 
I'am asking just to understand the root issues for other problems.

Thanks!

srinpraveen, 2019-09-11:
Hi zeldor,
I edited my original comment. By mistake, I had mentioned that I had bought the board from aliexpress. Actually when I checked back my emails, turns out that I had actually got it from Amazon. And this was a original wemos d1 mini board. The reviews for the product were good. And thankfully it's been working out very well so far. Wifi connection is pretty stable.

srinpraveen, 2019-09-11:
@zeldor , here is a link to the one that I purchased from amazon back in 2017. Hope that helps. https://www.amazon.com/gp/product/B01MDRVUQU/ref=ppx_od_dt_b_asin_title_s00?ie=UTF8&psc=1
The price is not super cheap like in Aliexpress but all goods are of pretty good quality from this seller.

jandrassy, 2019-09-11:
> @jandrassy , I actually did not have to change my wifi power settings. The root cause of my problem was a trailing whitespace character on my ssid string when the router was configured. I had missed [...]

then delete your comments from this issue. I commented the quote, not your problem

romansapia, 2019-10-09:
Hi, not sure if it's related. But I struggled 3 days, with exactly the same problem on 50 Wemos Boards. Finally I solved that removing one inductor in the antenna matching network. Not sure why it worked (I can guess it has an erroneous value) but it did!. Not just on one, but all. Complete details and photos:

[https://geekenargentina.wordpress.com/2019/10/08/esp8266-connectivity-troubleshooting/](url)

I hope it helps others

Román

devyte, 2019-10-10:
@romansapia excelente trabajo! There are way too many copies out there with hw mistakes, and some manufacturers don't even bother to test these days.

I'm closing this, because there is no singular reproducible issue or solution described.

closed 2019-10-10 by devyte

woodlist, 2020-04-23:
> @ThomasQvist make sure you are running it in AP mode only. Check that you have `WiFi.mode` before calling `WiFi.softAP`:
> 
> ```
>   WiFi.mode(WIFI_AP);
>   WiFi.softAP(ssid, password);
> ```
> 
> It the chip is in STA+AP mode, then the station will be trying to scan for a network to connect to, and since scan covers multiple channels, the AP will not be on the desired channel for some percentage of total time. This might explain the behavior you are observing.

The Wemos D1 Pro brodcasts its own SSID and custom user's code unable to be overwritten. What's a mistery? May it lack of some compile options that I leave everything by default on Arduino IDE?